esphome/issues #1909: Docker image is missing black formatter

madron, 2021-03-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

None

**ESPHome version (latest production, beta, dev branch)**

dev branch

**Affected component:**

https://esphome.io/guides/contributing.html#codebase-standards

**Description of problem:**
```
$ docker run --rm -v "${PWD}/":/esphome -it esphome/esphome-lint script/fulltest
+ script/ci-custom.py
+ script/lint-python
Running black...
Traceback (most recent call last):
  File "script/lint-python", line 97, in <module>
    main()
  File "script/lint-python", line 59, in main
    log = get_err(*cmd)
  File "/esphome/script/helpers.py", line 108, in get_err
    proc = subprocess.Popen(args, stdout=subprocess.PIPE, stderr=subprocess.PIPE)
  File "/usr/lib/python3.6/subprocess.py", line 729, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.6/subprocess.py", line 1364, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'black': 'black'
```

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

madron, 2021-03-13:
Here's an attempt to fix it:  https://github.com/esphome/esphome-docker-base/pull/13

OttoWinter, 2021-04-10:
Included in latest lint image.

closed 2021-04-10 by OttoWinter