esphome/issues #1911: Template switch turns off instantly

TheLastGimbus, 2021-03-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Installed with normal system-wide `pip3` on Ubuntu 20

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 wemos d1 mini

**ESPHome version (latest production, beta, dev branch)**

1.16.2

**Affected component:**

[Template switch](https://esphome.io/components/switch/template.html)

**Description of problem:**
Hi there!
I have a board which controlls lights in the hallway, with pir motion sensor. I want motion sensor to enable the light for 3 minutes, and then turn it off. *But*, I also want to be able to *disable* this behavior so it doesn't annoy me when I don't want it to do so

So I created a template switch:
```yaml
switch:
  - platform: template
    name: "Hall 2 auto pir"
    id: hall_2_auto_pir
```
And then in pir sensor:
```yaml
binary_sensor:
  - platform: gpio
    name: "Hall 2 PIR"
    pin: 5  # D1
    device_class: motion
    id: hall_2_pir
    on_press:
      - if:
          condition:
            switch.is_on: hall_2_auto_pir
          then:
            - script.execute: some_script
```
But the `script.execute: some_script` never executes, because `switch.hall_2_auto_pir` is always disabled!
<sub>To be sure - it works when I remove `if` statement...</sub>

When I click on it in Home Assistant dashboard, it clicks, and then turns off after 1 second:

https://user-images.githubusercontent.com/40139196/111090671-d4337a80-8530-11eb-8616-1b8da1665b58.mp4

Meanwhile, logs from esphome just say:
```
[01:51:45][D][switch:021]: 'Hall 2 auto pir' Turning ON.
[01:51:48][D][switch:021]: 'Hall 2 auto pir' Turning ON.
[01:51:51][D][switch:021]: 'Hall 2 auto pir' Turning ON.
[01:51:55][D][switch:021]: 'Hall 2 auto pir' Turning ON.
```

**Additional information and things you've tried:**

I tried adding
```yaml
lambda: |-
  return ! id(hall_2_auto_pir).state;
```
But then it just kept toggling in a loop :disappointed:

nikito7, 2021-03-15:
```yaml
binary_sensor:
  - platform: gpio
    name: "Hall 2 PIR"
    pin: 5  # D1
    device_class: motion
    id: hall_2_pir
    on_press:
    - if:
        condition:
          lambda: return id(hall_2_auto_pir).state;
        then:
        - script.execute: some_script
```

TheLastGimbus, 2021-03-15:
I don't know how is this supposed to help, since we clearly see that switch silently turns off without logging it - but I tried it, and doesn't work :disappointed:

nikito7, 2021-03-15:
You need add a helper in home assistant

TheLastGimbus, 2021-03-15:
Okay so a workaround - of course I can do a workaround, and I will do for now, but this is clearly an issue with ESPHome that needs to be fixed

Or I just don't understand how to make it work and it should be better documented

nikito7, 2021-03-15:
Create a switch with a unused gpio

Maybe an issue, but you use home assistant to control it

Not esphome standalone

If you will use mqtt only maybe will be a problem

TheLastGimbus, 2021-03-15:
> Create a switch with a unused gpio

*So a workaround*

I know how to fix it with workaround. But it's clearly a bug, and (as I understand), template switch was created to be this kind of helper - just as `input_boolean` in home assistant

TheLastGimbus, 2021-03-19:
`optimistic: true` helped! Thanks!

I will make a PR to documentation anyway, to make this clear :wink:

ssieb, 2021-05-22:
This is a common question on Discord.  The documentation is not clear on what that parameter does and the effect of not having it.

closed 2021-05-22 by ssieb