esphome/issues #1913: homeassistant companion (Android) --> esphome ble tracker UUID

alexd321, 2021-03-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant Companion App on Android (and ESPHome Add-on on Home Assistant OS on raspberry pi 4)

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32-wroom-32

**ESPHome version (latest production, beta, dev branch)**

1.16.2 (latest production)

**Affected component:**

ESP32 Bluetooth Low Energy Device
https://esphome.io/components/binary_sensor/ble_presence.html

**Description of problem:**
Home Assistant companion App BLE Transmitter Sensor (iBeacon) packets are not detected using ESPHome BLE Presence sensor. BLE Packets broadcast by the companion app are not detected by the BLE sensor.
 I used an example UUID of a00eb31c-85c8-11eb-8dcd-0242ac130003 (major and minor numbers don't seem to matter, and no way to define these in ESPHome)

I think the format of the packets from the companion app are different to what ESPHome is expecting?

**Problem-relevant YAML-configuration entries:**
```yaml
esp32_ble_tracker:

binary_sensor:
  # Presence based on BLE Service UUID
  - platform: ble_presence
    service_uuid: 'a00eb31c-85c8-11eb-8dcd-0242ac130003'
    name: "ESP32 BLE Tracker Test"

```

**Logs (if applicable):**

```
Using 'COM10' as serial port.
Showing logs:
[20:13:16] 8 2016 00:2[I][logger:166]: Log initialized
[20:13:16][C][status_led:014]: Setting up Status LED...
[20:13:16][C][ota:366]: There have been 5 suspected unsuccessful boot attempts.
[20:13:16][I][app:029]: Running through setup()...
[20:13:16][D][esp32_ble_tracker:148]: Starting scan...
[20:13:16][C][wifi:033]: Setting up WiFi...
[20:13:16][I][wifi:194]: WiFi Connecting to '*****'...
[20:13:17][I][wifi:457]: WiFi Connected!
[20:13:17][C][wifi:303]:   SSID: '*****'
[20:13:17][C][wifi:304]:   IP Address: 192.168.0.47
[20:13:17][C][wifi:306]:   BSSID: ********
[20:13:17][C][wifi:307]:   Hostname: 'livingroom'
[20:13:17][C][wifi:311]:   Signal strength: -63 dB ▂▄▆█
[20:13:17][C][wifi:315]:   Channel: 1
[20:13:17][C][wifi:316]:   Subnet: 255.255.255.0
[20:13:17][C][wifi:317]:   Gateway: 192.168.0.1
[20:13:17][C][wifi:318]:   DNS1: 192.168.0.1
[20:13:17][C][wifi:319]:   DNS2: 0.0.0.0
[20:13:17][D][wifi:466]: Disabling AP...
[20:13:17][C][ota:029]: Over-The-Air Updates:
[20:13:17][C][ota:030]:   Address: livingroom.local:3232
[20:13:17][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 5 restarts
[20:13:17][C][api:022]: Setting up Home Assistant API server...
[20:13:17][I][app:059]: setup() finished successfully!
[20:13:17][I][app:105]: ESPHome version 1.16.2 compiled on Mar 15 2021, 20:09:43
[20:13:17][C][status_led:019]: Status LED:
[20:13:17][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT)
[20:13:17][C][wifi:443]: WiFi:
[20:13:17][C][wifi:303]:   SSID: '*******'
[20:13:17][C][wifi:304]:   IP Address: 192.168.0.47
[... 85 lines cut ...]
[20:18:32][D][esp32_ble_tracker:567]:   Name: 'LYWSD03MMC'
[20:18:41][D][esp32_ble_tracker:544]: Found device ****** RSSI=-74
[20:18:41][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[20:18:41][D][esp32_ble_tracker:567]:   Name: 'LYWSD03MMC'
[20:18:42][D][esp32_ble_tracker:544]: Found device 33:6A:28:1B:9A:0E RSSI=-92
[20:18:42][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[20:18:46][D][esp32_ble_tracker:544]: Found device CA:12:CC:B2:BA:BE RSSI=-81
[20:18:46][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[20:18:49][D][esp32_ble_tracker:544]: Found device 35:34:85:19:F4:4E RSSI=-81
[20:18:49][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[20:18:49][D][esp32_ble_tracker:567]:   Name: 'Smart Tag'
[20:19:18][D][esp32_ble_tracker:544]: Found device ****** RSSI=-90
[20:19:18][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[20:19:30][D][esp32_ble_tracker:544]: Found device 4C:6C:66:FB:7C:67 RSSI=-72
[20:19:30][D][esp32_ble_tracker:565]:   Address Type: RANDOM
[20:19:48][D][esp32_ble_tracker:544]: Found device ******* RSSI=-74
[20:19:48][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[20:19:48][D][esp32_ble_tracker:567]:   Name: 'ATC_E9BD6C'
[20:20:13][D][esp32_ble_tracker:544]: Found device 2B:2C:E7:16:E4:BF RSSI=-99
[20:20:13][D][esp32_ble_tracker:565]:   Address Type: RANDOM
```

**Additional information and things you've tried:**

Using nRF Connect app to advertise the same UUID works fine

alexd321, 2021-03-15:
using
```yaml
on_ble_service_data_advertise:
    - service_uuid: a00eb31c-85c8-11eb-8dcd-0242ac130003
      then:
        - lambda: 'ESP_LOGD("ble_adv", "Service data: Test Phone found");'
````
doesn't seem to work either

marcinbauer85, 2021-03-17:
I have the same problem in #1912  It's like ble on ESPHome doesn't really support iBecons

ignisf, 2021-07-15:
This issue is still present and a fix is under review in esphome/esphome#1627

closed 2021-09-01 by jesserockz