esphome/issues #1915: Acceptable characters in naming of yaml.

troubleblob, 2021-03-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
My problem ended up being with _ in my yaml names. It uses the name of the yaml for dhcp lookup and my lynksys router didn't handle those request quiet right. My router would randomly change the node ip. Digging deeper into my dhcp client tables some clients would have other clienst names and not the name with _. It's a fairly new router and once I reserved ips for the mac addresses with affected names I'm having no issues and reboots are much faster. I'd recommened removing the _ as accetpable.
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ssieb, 2021-03-20:
It's actually the `name:` parameter in the `esphome:` section that sets the hostname, but it's true, `_` is not valid for hostnames, so should not be accepted.

randybb, 2021-03-21:
I don't have such problem, but after that "patch" I have to rename all my devices? Hmm...not good

troubleblob, 2021-03-21:
That's how my issues started. I started with rename but that was leading to
lots of unconnected devices. I then started openning logs then getting the
ip then remove the device from esphome integration and immediately add it
again with the ip. I was finding my router would shuffle ip's around
because of my names.

randybb, 2021-03-22:
But DNS name can contain "_" so it is just an issue related to some strict (or crappy?) setups. Ubiquity doesn't care and as I said, never had any issue with "_" in hostname. So I think it should just show a warning about not allowed character that might cause problems.

closed 2021-04-08 by OttoWinter