esphome/issues #1918: esphome/log.h not found

aladin2000, 2021-03-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
rasbian buster  ( - debian 10 )

**ESP (ESP32/ESP8266, Board/Sonoff):** LCTECH 

**ESPHome version (latest production, beta, dev branch)**

Current version: 1.16.2 ( Changelog )

**Affected component:**

**Description of problem:**
one file is missing in compile time
esphome/log.h not found in compile time 

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
substitutions:
  devicename: "lctech"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_ssid_passwd

  manual_ip:
    static_ip: 192.168.1.38
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
  domain: .local

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lctech Fallback Hotspot"
    password: !secret lctech_failback_passwd

# Enable Home Assistant API
api:
  password: !secret lctech_passwd

ota:
  safe_mode: True
  password: !secret lctech_passwd

esphome:
  name: ${devicename}
  platform: ESP8266
  board: esp01_1m
  build_path: build/gatecontrol
  includes:
    - lcrelay.h
    - lccomponent.h

logger:
  level: DEBUG
  baud_rate: 0

web_server:
  port: 80

time:
- platform: homeassistant

uart:
  id: uart0
  baud_rate: 19200
  tx_pin: GPIO1
  rx_pin: GPIO3

sensor:
  - platform: wifi_signal
    name: "${devicename}: WiFi Signal"
    update_interval: 60s

switch:

- platform: custom
  lambda: |-
    auto r1 = new LCRelay(id(uart0), 1);
    auto r2 = new LCRelay(id(uart0), 2);
    App.register_component(r1);
    App.register_component(r2);
    return {r1, r2};
  switches:
    - name: "${devicename}: Relay 1"
      id: relay1
    - name: "${devicename}: Relay 2"
      id: relay2

captive_portal:

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/lctech.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/build/gatecontrol
Processing lctech (board: esp01_1m; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/lctech/.pioenvs/lctech/src/main.cpp.o
In file included from src/main.cpp:24:0:
src/lccomponent.h:3:25: fatal error: esphome/log.h: No such file or directory
 #include "esphome/log.h"
                         ^
compilation terminated.
Archiving /data/lctech/.pioenvs/lctech/libd67/libESPAsyncTCP-esphome.a
Compiling /data/lctech/.pioenvs/lctech/lib0d3/ESP8266mDNS/ESP8266mDNS_Legacy.cpp.o
*** [/data/lctech/.pioenvs/lctech/src/main.cpp.o] Error 1
Compiling /data/lctech/.pioenvs/lctech/lib0d3/ESP8266mDNS/LEAmDNS.cpp.o
========================= [FAILED] Took 18.08 seconds =========================

```

**Additional information and things you've tried:**

try to patch but not possible

randybb, 2021-03-17:
This is not related to ESPHome. It is just a problem in your _lccomponent.h_ file that is including header file that is somewhere else. Just take a look into _/data/lctech/.pioenvs/lctech/_ and you will figure it out.

Correct include def is, but you don't need it as it is already included in _esphome.h_.
```cpp
#include "esphome/core/log.h"
```

aladin2000, 2021-03-18:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAD8AAAA/CAYAAABXXxDfAAAAAXNSR0IArs4c6QAAAARzQklUCAgICHwIZIgAAAAmSURBVGiB7cEBDQAAAMKg909tDwcUAAAAAAAAAAAAAAAAAAAAJwY+QwABivJx1AAAAABJRU5ErkJggg==###24x24:true###][Image de pistage][https://github.com/notifications/beacon/AD7K4I7ZRXOIWGYHYDPRZSTTEE4FNA5CNFSM4ZLNITGKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOF7DBCYQ.gif]

Hi
I changed with

> #include "esphome/core/log.h

But got others compilation errors
Seems that  missing file does not have same defs as in standard one

Regards
Aladin

aladin2000, 2021-03-18:
Hi
I had old version of code

I modified my source code...

Everythings is good now...

closed 2021-03-18 by aladin2000