esphome/issues #1921: LoadProhibited crash in esphome/components/api/api_server.cpp

mmakaay, 2021-03-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I use ESPHome in a docker container, latest version.

**ESP (ESP32/ESP8266, Board/Sonoff):**

The board I am using is ESP32-WROOM-32D, a single core ESP32 version.

**ESPHome version (latest production, beta, dev branch)**

v1.17.0-dev

**Affected component:**

https://esphome.io/components/api.html

**Description of problem:**
Sometimes, my device reboots when clients are being disconnected. When this happens, the console logs a core panic related to a LoadProhibited exception case.

[A race condition issue within AsyncTCP](https://github.com/me-no-dev/AsyncTCP/issues/116) caused a lot of random disconnects for me, which in turn made my device hit this bug quite often. I have fixed these disconnects, so this issue isn't really bothering for me anymore. It felt like something that would need fixing though. Spontaneous reboots aren't fun.

**Problem-relevant YAML-configuration entries:**
```yaml
api:
  password: !secret api_password
  reboot_timeout: 0s
```

**Logs (if applicable):**

```
[15:02:43]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[15:02:43]Core 0 register dump:
[15:02:43]PC      : 0x400d6cff  PS      : 0x00060130  A0      : 0x80163618  A1      : 0x3ffb1f00  
[15:02:43]A2      : 0x3ffb8b90  A3      : 0x3ffd3658  A4      : 0x3ffd365c  A5      : 0x00008cc0  
[15:02:43]A6      : 0x00050b23  A7      : 0x3ffb8058  A8      : 0xbaad5678  A9      : 0x3ffb1ea0  
[15:02:43]A10     : 0x00000005  A11     : 0x3f40305c  A12     : 0x00000043  A13     : 0x3f4013a5  
[15:02:43]A14     : 0x3ffd3364  A15     : 0x00060b23  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[15:02:43]EXCVADDR: 0xbaad569c  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[15:02:43]
[15:02:43]Backtrace: 0x400d6cff:0x3ffb1f00 0x40163615:0x3ffb1f20 0x401636a9:0x3ffb1f40 0x400de3a5:0x3ffb1f60 0x400e0eb2:0x3ffb1f90 0x400eda41:0x3ffb1fb0 0x4008926d:0x3ffb1fd0
[15:02:44]
[15:02:44]Rebooting...
```

The stack trace analysis shows:
```
0x400d6cff: esphome::api::APIServer::loop() at src/esphome/components/api/api_server.cpp line 67
0x40163615: esphome::Component::call_loop() at src/esphome/core/component.cpp line 61
0x401636a9: esphome::Component::call() at src/esphome/core/component.cpp line 82
0x400de3a5: esphome::Application::loop() at src/esphome/core/application.cpp line 72
0x400e0eb2: loop() at src/main.cpp line 265
0x400eda41: loopTask(void*) at /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp line 19
0x4008926d: vPortTaskWrapper at /home/paul/src/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

**Additional information and things you've tried:**

I have looked at the code to see what could be the problem. I find a comment in there, about first logging the disconnect operations, before actually deleting the memory, to prevent access to freed memory. Looking at that code, it seems like that possible route for a LoadProhibited exception has been closed down.

My suspicion  about a possible cause (but note: I'm no expert on the `api_server.cpp` code) is that things might go haywire when the code is handling the disconnection of one or more clients around line 67, while at the same time a new client connection is accepted.
The lwIP thread accepts that connection and does the onClient callback to `api_server.cpp`, which adds the new client to the end of the client connection vector. So there we have a vector that is being partitioned and cleared in the `APIServer::loop()` code, while at the same time `a_this->clients_.push_back(new APIConnection(client, a_this));` might get called in the onClient callback.

I haven't tracked the issue any further, but this is what felt suspicous to me in relation to the exception.

ssieb, 2021-03-20:
You have a lot of clients talking to the device over the API connection?

mmakaay, 2021-03-20:
No, Just Home Assistant and the web console logger or even just home assistant.

The issue might be most prevalent when the server on the device disconnects the Home Assistant client, causing a reconnect from HA right away. I don't think this would fail when the HA client disconnects on its own.

Maybe a way to reproduce the issue, would be to forcibly trigger disconnects from a hacked up AsyncTCP-esphome lib. My frequent disconnects by a bug in that library is what made my device hit this exception.

Maybe it's an idea to have a mutex lock around the bits of code that modify the list of connected clients? (in case my assumption about the root cause is correct)

mmakaay, 2021-03-27:
I ran into another LoadProhibited exception reboot, which this time pointed me to a different bit of code.
In this case, the crash happened within the AsyncTCP library code

```
PC: 0x400e474a: AsyncClient::_sent(tcp_pcb*, unsigned short) at /home/mmakaay1/Code/esphome/config/libs/AsyncTCP/src/AsyncTCP.cpp line 917
EXCVADDR: 0x000000a8

Decoding stack results
0x400e474a: AsyncClient::_sent(tcp_pcb*, unsigned short) at /home/mmakaay1/Code/esphome/config/libs/AsyncTCP/src/AsyncTCP.cpp line 917
0x400e4908: AsyncClient::_s_sent(void*, tcp_pcb*, unsigned short) at /home/mmakaay1/Code/esphome/config/libs/AsyncTCP/src/AsyncTCP.cpp line 1224
0x400e4fa9: _async_service_task(void*) at /home/mmakaay1/Code/esphome/config/libs/AsyncTCP/src/AsyncTCP.cpp line 165
0x40089271: vPortTaskWrapper at /home/paul/src/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

On line 917, I find the `AsyncClient::_sent(tcp_pcb* pcb, uint16_t len)` method. It crashes on setting a boolean object field to false, so this is not something that goes wrong inside the AsyncTCP code. The most likely issue route here is that the client connection is deleted from `api_server.cpp`, while there was still traffic being handled by the AsyncTCP code.
This is backed up by the logs around this issue. These showed:

```
[00:20:31][VV][api.service:032]: send_ping_response: PingResponse {}
[00:20:36][W][AsyncTCP.cpp:964] _poll(): ack timeout 4
[00:20:36][V][api.connection:708]: Error: Disconnecting ESPHome v1.18.0-dev (192.168.100.17)
[00:20:36]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
```

My theory for this crash is that `api_server.cpp` destroys the client connection because it thinks the connection is dead, while actually an ACK is coming in at the same time, triggering the `_sent()` method on an object that has just been destroyed.
So your typical "use after free" scenario.

There were no multiple connections active at the same time here. Only home-assistant and my serial debugger were connected.

mmakaay, 2021-04-06:
Yet another crash, which looked like this:
```
[16:56:39][V][api.connection:708]: Error: Disconnecting ESPHome v1.18.0-dev (192.168.100.17)
[16:56:39]Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
[16:56:39]Core 0 register dump:
[16:56:39]PC      : 0x400e4ba1  PS      : 0x00060d30  A0      : 0x800e4c8c  A1      : 0x3ffd3150  
[16:56:39]A2      : 0x00000000  A3      : 0x3ffd3808  A4      : 0x3ffd3618  A5      : 0x00000001  
[16:56:39]A6      : 0x00000000  A7      : 0x3ffcf280  A8      : 0x00000000  A9      : 0x3ffd3130  
[16:56:39]A10     : 0x00016346  A11     : 0x00060f23  A12     : 0x00060f20  A13     : 0x0000000c  
[16:56:39]A14     : 0x3ffcee84  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001d  
[16:56:39]EXCVADDR: 0x000000b8  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
[16:56:39]
[16:56:39]Backtrace: 0x400e4ba1:0x3ffd3150 0x400e4c89:0x3ffd3190 0x400e537e:0x3ffd31b0 0x40089271:0x3ffd31e0
```

The backtrace is:
```
0x400e4ba1: AsyncClient::_close() at ...../esphome/config/libs/AsyncTCP/src/AsyncTCP.cpp line 843
0x40089271: vPortTaskWrapper at /home/paul/src/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

Line 843 is in the following code.
```c++
int8_t AsyncClient::_close(){
    //ets_printf("X: 0x%08x\n", (uint32_t)this);
    int8_t err = ERR_OK;
    if(_pcb) {
        //log_i("");
        tcp_arg(_pcb, NULL);
        tcp_sent(_pcb, NULL);
        tcp_recv(_pcb, NULL);
        tcp_err(_pcb, NULL);
        tcp_poll(_pcb, NULL, 0);
        _tcp_clear_events(this);
        err = _tcp_close(_pcb, _closed_slot);
        if(err != ERR_OK) {
            err = abort();
        }
        _pcb = NULL;
        if(_discard_cb) {
            _discard_cb(_discard_cb_arg, this); // <------- this line is 843
        }
    }
    return err;
}                                                                                                                                                     
```

glmnet, 2021-04-06:
Are you aware ESPHome is using a forked library? https://github.com/OttoWinter/AsyncTCP

mmakaay, 2021-04-06:
Yes, I am aware of that. And I am using a fork of that fork to be able to track and fix issues. Therefore, I also included pointers to what code is failing, instead of just pointing to a line number. Due to my own changes, the line numbering might be off.
I did find a race condition in that library and I reported it to the upstream repository. 
https://github.com/me-no-dev/AsyncTCP/issues/116
In the discussion about that issue, you'll find a link to my personal fork with some fixes that I did to prevent random disconnects. Since the issue and possible fix are the same for the upstream and the fork, I reported it upstream. 

So I guess the return question is: does the forked library still follow the upstream version, or is the fork fully detached from upstream by now? In that case, the above report ought to be in the esphome/issues too.

The crashes that I described in this issue report, might be seen as AsyncTCP issues. However, these all seem related to the api_server code sometimes destroying the AsyncTCP client objects, while the AsyncTCP library still is performing actions on them. So it's s crash in AsyncTCP, which might be triggered by tcp_server's behavior.

The relation between my AsyncTCP issue report and this one, is that by fixing the random disconnects in the AsyncTCP code, the tcp_server will disconnect and destroy the client connections a lot less, and will therefore run into the abovementioned exceptions a lot less too. The more disconnects happen, the bigger the chance is that a spontaneous reboot occurs.

It's all a bit hard to explain, since I'm looking at interrelated issues between multiple libraries here, but I hope this makes things a bit more clear.

glmnet, 2021-04-06:
> does the forked library still follow the upstream version

I guess it is, it should. But I noticed it's behind so just wanted to be sure you're aware as You're not mentioning it before. 

Otto is around so we can ping him to update the fork, and if your fixes proves to be good you can submit pr there and get it merged quickly. 

How can I test your branch? 
Is it posible by just `platformio_options` config?

mmakaay, 2021-04-06:
Yes, that can be done. I use my patched library that way on my local build.
I [described the procedure ](https://github.com/mmakaay/esphome-yeelight_bs2/blob/main/README.md#issue-the-device-keeps-losing-its-connection-to-home-assistant)in a project that I'm working on right now.

OttoWinter, 2021-04-10:
Wow @mmakaay thanks for debugging! The patch you posted looks good to me, could you create a pull request against https://github.com/OttoWinter/AsyncTCP ?

mmakaay, 2021-04-10:
Yes, no problem at all, @OttoWinter 
I will cook up a pull request against the fork.

mmakaay, 2021-04-12:
A quick update on this: I am now using the recently updated framework packages from:
https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.6
And with those, I can now use framework espressif32@3.2.0
I don't want to cheer too early about this, but I must say that I didn't notice any spontaneous reboots so far.

My fixes for AsyncTCP have been merged into AsyncTCP-esphome by @OttoWinter, so possibly this will prove to be a winning combination to not hit these pesky exceptions anymore. Fingers crossed!

mmakaay, 2021-04-12:
Unfortunately, I'm still seeing spontaneous reboots of the device.

From the connected log client, I see that the last log message before the reboot was "[00:16:03][V][api.connection:708]: Error: Disconnecting Home Assistant 2021.4.1 (192.168.100.135)", so that's very likely one of the above exception causes.
Possibly, my device is at a spot from where its WiFi connection is a bit dodgy (I moved downstairs, while upstairs it was very stable), because I see several of these "Disconnecting" messages recently. Only the last one hit a killer race condition.

This means that there's still things to fix in here.
I will give debugging this a go, once I finished my current component project.

mmakaay, 2021-08-11:
Agreed on closing this ticket.
With the changes that we did in the AsyncTCP department, things have been running very stable for me.

closed 2021-08-11 by ssieb