esphome/issues #1923: Logger timezone issue

chemelli74, 2021-03-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 2021.3.4

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 dev board

**ESPHome version (latest production, beta, dev branch)**

v1.16.2

**Affected component:**

logger

**Description of problem:**
Time is synchronized with HA but is off by 1 hour

**Problem-relevant YAML-configuration entries:**
```yaml
# Sync time via Home Assistant API
time:
  - platform: homeassistant
    id: homeassistant_time
    timezone: "Europe/Rome"
```

**Logs (if applicable):**

esphome config:
```
time:
- platform: homeassistant
  id: homeassistant_time
  timezone: CET-1CEST-2,M3.4.0/2,M10.5.0/3
  update_interval: 15min
```

esphome run:
```
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00
```

esphome logs:
```
[08:13:17][C][homeassistant.time:010]: Home Assistant Time:
[08:13:17][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
```

**Additional information and things you've tried:**

Logs shows **8**:13 but here it's **9**:13 
Tried removing `timezone` from yaml, but I still get **8**:13 :-S

randybb, 2021-03-19:
Do you have a correct time on HA server? 

I have the same timezone (Europe/Bratislava, that is GMT+1) and time on my side is correct.

chemelli74, 2021-03-19:
You, time in HA is fine as well as time zone.

Do you have the same values on if you do a `config` check ?
What I don’t get is why it shows “CET-1” and not CET

Thx,

Simone

chemelli74, 2021-03-24:
@randybb can you please check your config and let me know if it's like mine or not please ?

Thank you,

Simone

randybb, 2021-03-24:
In my case it is
```yaml
  - platform: homeassistant
    id: ha_time
    timezone: Europe/Bratislava
```
```syslog
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00
[14:44:35][C][homeassistant.time:010]: Home Assistant Time:
[14:44:35][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
```
and time in the log is matching the current time.

OttoWinter, 2021-04-10:
The timezone used when the python app logs is the system timezone (so what your libc localtime function outputs). This is in line with common software (including how HA prints log timestamps)

The time component is separate from that and only used for telling the firmware what timezone it is in.

Also note: the time you see when running esphome ... logs is NOT generated by the ESP, but prepended by the python software (in most configurations, the esp doesn't know the time. But the python software does, so it inserts it).

The fix here would be to configure your host system to have the correct timezone configured, the you also don't need to manually set the timezone string.

chemelli74, 2021-04-10:
Hi @OttoWinter, thx for your reply.

I'm running HA in docker, and I see the correct system time:

```
root@HomeNAS:~# docker exec -it homeassistant bash
bash-5.0# date
Sat Apr 10 14:08:48 CEST 2021
```

So what should be changed in your opinion to fix the esphome time ?

Simone

chemelli74, 2021-05-23:
Latest esphome docker container honors "`TZ`".

closed 2021-05-23 by chemelli74