esphome/issues #1924: MCP23017 random outputs at startup

comm-lab, 2021-03-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 2021.1.4
**ESP (ESP32/ESP8266, Board/Sonoff):**

Olimex ESP32-EVB
**ESPHome version (latest production, beta, dev branch)**

1.17.0-dev
**Affected component:**

MCP23017

**Description of problem:**
During startup of the firmware the output pins of the MCP23017 are set to random values. These should be due to the fact that in the component setup function the code that was setting all pins as inputs it's been removed (in commit b52f7cfe86a167bcd3f41baf2194243699a3e354 MCP230xx open drain interrupt pins (#1243)
):
```cpp
  // all pins input
  this->write_reg_(MCP23017_IODIRA, 0xFF);
  this->write_reg_(MCP23017_IODIRB, 0xFF);
```
They should be all set as inputs at setup, then, after OLATA and OLATB are set, they can be set as configured in yaml at main.cpp:
```cpp
 out_puntoluce1->set_pin(new mcp23xxx_base::MCP23XXXGPIOPin(MCP23017_0, 8, mcp23xxx_base::MCP23XXX_OUTPUT, true, mcp23xxx_base::MCP23XXX_NO_INTERRUPT));
```
**Problem-relevant YAML-configuration entries:**
```yaml
#------------------------MCP23017_0-----------------------------------------          
mcp23017:
  - id: 'MCP23017_0'
    address: 0x27

#------------------------MCP23017_1-----------------------------------------          
  - id: 'MCP23017_1'
    address: 0x26

#------------------------MCP23017_2-----------------------------------------          

  - id: 'MCP23017_2'
    address: 0x25

switch:             
#------------------------MCP23017_0 Port B (outputs)----------
  - platform: gpio
    id: out_puntoluce1
    name: "MCP23017_0 PIN #8"
    pin:
      mcp23xxx: MCP23017_0
      number: 8
      mode: OUTPUT
      inverted: True
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

I've tried adding those two lines back to the setup fuction, (after open drain interrupts setting) and everything works fine.
My component setup now looks like this:

```cpp
void MCP23017::setup() {
  ESP_LOGCONFIG(TAG, "Setting up MCP23017...");
  uint8_t iocon;
  if (!this->read_reg(mcp23x17_base::MCP23X17_IOCONA, &iocon)) {
    this->mark_failed();
    return;
  }

  if (this->open_drain_ints_) {
    // enable open-drain interrupt pins, 3.3V-safe
    this->write_reg(mcp23x17_base::MCP23X17_IOCONA, 0x04);
    this->write_reg(mcp23x17_base::MCP23X17_IOCONB, 0x04);
  }

  // all pins input
  this->write_reg(mcp23x17_base::MCP23X17_IODIRA, 0xFF);
  this->write_reg(mcp23x17_base::MCP23X17_IODIRB, 0xFF);
}
```

kbx81, 2021-03-23:
> They should all be set as inputs at setup

This is not _always_ desirable behavior -- it depends on the application.

One of the reasons that such an I/O expander may be used in a hardware design is because its I/O pins can retain their states across resets/reboots of the processor (ESP32 or '8266 in this case).

For example, I built a [thermostat](https://github.com/kbx81/ClimateSprinklerController). If the ESP resets and the setup/init code forces all of the pins on the I/O expander to a fixed state, this may cause the outputs to glitch which is actually _bad_ for the HVAC system(s) it is controlling. This is often true for other types of control systems, too. In a worst-case scenario, the ESP ends up in a boot loop and such setup/init code causes the outputs to continually glitch which would indeed be _very bad_.

With respect to I/O devices such as this, I would argue that it's best for us to not assume anything and force states that "we think are best" in setup code unless there is a specific reason to do so. I think flexibility is key. Perhaps the right solution is to add an option that can be set in the config to control the behavior of the setup/init code. As an alternative, it's also possible to add a function/lambda called by `on_boot:` that sets all the pins to your preferred state(s) at power-on.

comm-lab, 2021-03-23:
> For example, I built a thermostat. If the ESP resets and the setup/init code forces all of the pins on the I/O expander to a fixed state, this may cause the outputs to glitch which is actually bad for the HVAC system(s) it is controlling.
Yes indeed, that's exactly what I was trying to avoid, and that's the situation I run into in the first place.

> With respect to I/O devices such as this, I would argue that it's best for us to not assume anything and force states that "we think are best" in setup code unless there is a specific reason to do so
That makes perfectly sense, I'm just wondering why those two lines of code was removed just one month ago and nobody ever complained about it, the removal could have caused unpredictable behaviors like the one I had, with the light in my house going on and off as hell everytime ESP restarted :)

> I think flexibility is key. Perhaps the right solution is to add an option that can be set in the config to control the behavior of the setup/init code.
We completely agree here, it would be great to have such an option.

>  As an alternative, it's also possible to add a function/lambda called by on_boot: that sets all the pins to your preferred state(s) at power-on.
Fine, thanks for the on_boot suggestion I'll try to do so.

kbx81, 2021-03-23:
> going on and off as hell everytime ESP restarted

I would expect the outputs to be in an undefined state at power-on only. If this is happening _every time your ESP resets,_ you might have a different problem.

> Perhaps the right solution is to add an option that can be set in the config to control the behavior of the setup/init code.

I slept on this and had an idea for implementation...
We could add a setup option for each port -- I was thinking `initial_pin_state_porta` and `initial_pin_state_portb`. It's an optional config parameter with possible values of `INPUT`, `INPUT_PULLUP`, `OUTPUT_HIGH`, `OUTPUT_LOW`, and `UNDEFINED`. The default is `INPUT` which I believe is what we were doing before I removed that init code. The respective port's pin states are set based on this parameter, with `UNDEFINED` causing **no** port/pin initialization during `setup()`.

@jesserockz I know you worked on this recently and are using it, what do you think? 😄

jesserockz, 2021-03-23:
Sounds good. Allows people to restore previous functionality that was working for them

kbx81, 2021-07-22:
I'm still planning to work on this Mr. Stalebot 😄

andriej, 2021-07-22:
I have issues with warnings regarding 4xMCP's and ESP32 - 0x2*, with short wire in quite noisy environment (electrical noise from relays).

I guess it might be connected that unused pins (i have 128 GPIOs total) may be inputs and are flooded.
Any way to prevent/set all unused GPIOs as OUTPUT so they don't "spam" bus? As disabling is probably not possible. ;-)

Fix would be welcome too!

MajorBreakfast, 2022-10-04:
Running into this as well. Getting weird noises from my relay boards during startup.

To workaround this I forked the esphome repo, added the "all input pins" lines mentioned by @comm-lab above and used the [external_components](https://esphome.io/components/external_components.html) feature to use the patched code. Now my relays stay all off during startup.