esphome/issues #1926: Battery level not available in xiaomi_hhccjcy01 (device fw 3.2.4)

chemelli74, 2021-03-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant core 2021.3.4

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 dev board

**ESPHome version (latest production, beta, dev branch)**

v1.16.2

**Affected component:**

xiaomi_hhccjcy01

**Description of problem:**
Starting again the discussion from [Readds the battery level for xiaomi_hhccjcy01](https://github.com/esphome/esphome/pull/1288)

I have 3 sensors with firmware 3.2.4 and mobile app shows battery level; so I think it's advertised.
But unfortunately VERY_VERBOSE doesn't compile.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

chemelli74, 2021-04-05:
Anyone that can help debugging what's coming in from the sensor ?

Thx in advance,

Simone

chemelli74, 2021-05-01:
Anyone that can give me an advise ?

Simone

javicalle, 2021-05-11:
Since v3.2.1 no battery level is reported, so this integration can't get it:
* https://github.com/esphome/esphome/pull/1288#issuecomment-695809481

An active conection to device will be required to get battery level, but not trivial:
* https://github.com/custom-components/ble_monitor/issues/239

Cheers.