esphome/issues #1927: no i2c devicees found. Wemos D1 mini + SSD1306 64x48 OLED

nakermann1973, 2021-03-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

esphome dashboard running on docker (image esphome/esphome)

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini

**ESPHome version (latest production, beta, dev branch)**

1.16.2

**Affected component:**

Not sure if the is i2c or display component
https://esphome.io/components/i2c.html
https://esphome.io/components/display/ssd1306.html

**Description of problem:**
I am unable to get esphome to connect to my display over i2c.  The most relevant log entries are

```
[19:55:39][I][i2c:033]: Scanning i2c bus for active devices...
[19:55:39][I][i2c:049]: Found no i2c devices!
[19:55:39][C][ssd1306_i2c:023]: I2C SSD1306
[19:55:39][C][ssd1306_i2c:023]:   Rotations: 0 °
[19:55:39][C][ssd1306_i2c:023]:   Dimensions: 64px x 48px
[19:55:39][C][ssd1306_i2c:024]:   Address: 0x3C
[19:55:39][C][ssd1306_i2c:025]:   Model: SSD1306 64x48
[19:55:39][C][ssd1306_i2c:027]:   External VCC: NO
[19:55:39][C][ssd1306_i2c:028]:   Update Interval: 1.0s
[19:55:39][E][ssd1306_i2c:031]: Communication with SSD1306 failed!
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: co2
  platform: ESP8266
  board: d1_mini

i2c:
  sda: D1
  scl: D2  
  scan: True
  
font:
  - file: "comic.ttf"
    id: comic_sans
    size: 20

display:
  - platform: ssd1306_i2c
    model: "SSD1306 64x48"
    address: 0x3C
    lambda: |-
      it.print(0, 0, id(comic_sans), "Hello World!");

wifi:
  ssid: "IOT_2"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Co2 Fallback Hotspot"
    password: "xxx"

captive_portal:

# Enable logging
[... 357 lines cut ...]
20:55:49.986 -> [0;38m[VV][api.service:032]: send_ping_response: PingResponse {}[0m
20:55:55.011 -> [0;38m[VV][api.service:220]: on_ping_request: PingRequest {}[0m
20:55:55.011 -> [0;38m[VV][api.service:032]: send_ping_response: PingResponse {}[0m
20:56:00.038 -> [0;38m[VV][api.service:220]: on_ping_request: PingRequest {}[0m
20:56:00.038 -> [0;38m[VV][api.service:032]: send_ping_response: PingResponse {}[0m
20:56:05.038 -> [0;38m[VV][api.service:220]: on_ping_request: PingRequest {}[0m
20:56:05.038 -> [0;38m[VV][api.service:032]: send_ping_response: PingResponse {}[0m
20:56:10.072 -> [0;38m[VV][api.service:220]: on_ping_request: PingRequest {}[0m
20:56:10.072 -> [0;38m[VV][api.service:032]: send_ping_response: PingResponse {}[0m
20:56:15.093 -> [0;38m[VV][api.service:220]: on_ping_request: PingRequest {}[0m
20:56:15.093 -> [0;38m[VV][api.service:032]: send_ping_response: PingResponse {}[0m
20:56:20.107 -> [0;38m[VV][api.service:220]: on_ping_request: PingRequest {}[0m
20:56:20.107 -> [0;38m[VV][api.service:032]: send_ping_response: PingResponse {}[0m

```

**Additional information and things you've tried:**

I have tried using a extremely basic i2c port scanner (https://github.com/jainrk/i2c_port_address_scanner/blob/master/i2c_port_address_scanner/i2c_port_address_scanner.ino).  It does detect the i2c device at adddress 0x3C.

Tasmota works correctly (tasmota-display.bin). I configured it the template and commands here https://templates.blakadder.com/oled.html

Based on this, I assume my hardware and connections are working correctly, and there is either an error in my esphome configuration, or a bug in the code.

How can I further debug this in esphome?

nakermann1973, 2021-03-20:
Since this actually turned out to be an error in my configuration, I am closing the bug.

Working config (specifying GPIO4/GPIO5 instead of D1/D2 for the i2c sda and scl settings):

```
esphome:
  name: co2
  platform: ESP8266
  board: d1_mini

i2c:
  sda: GPIO4
  scl: GPIO5
  scan: True
  
font:
  - file: "comic.ttf"
    id: comic_sans
    size: 12

display:
  - platform: ssd1306_i2c
    model: "SSD1306 64x48"
    address: 0x3C
    lambda: |-
      it.print(0, 0, id(comic_sans), "Hello World!");

wifi:
  ssid: "IOT_2"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Co2 Fallback Hotspot"
    password: "xxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxx"

ota:
  password: "xxx"
```

closed 2021-03-20 by nakermann1973

Habbie, 2021-07-05:
> Working config (specifying GPIO4/GPIO5 instead of D1/D2 for the i2c sda and scl settings):

A note for whoever finds this and tries to use it as documentation: the problem is not `D` versus `GPIO` syntax, but that the original config swapped `D1` and `D2` around.

This works:
```
i2c:
  sda: D2
  scl: D1
  scan: true
```

and causes this output:
```
[20:59:52][C][i2c:028]: I2C Bus:
[20:59:52][C][i2c:029]:   SDA Pin: GPIO4
[20:59:52][C][i2c:030]:   SCL Pin: GPIO5
[20:59:52][C][i2c:031]:   Frequency: 50000 Hz
[20:59:52][I][i2c:033]: Scanning i2c bus for active devices...
[20:59:52][I][i2c:040]: Found i2c device at address 0x02
...
```

which matches your working version, but both syntaxes work, as long as you get the numbers right.

Habbie, 2021-07-05:
In fact, and this is also mentioned at https://esphome.io/components/i2c.html#i2c, those values are -the defaults-. So the following config is identical to the config I posted above, and indeed works fine:

```
i2c:
```