esphome/issues #1932: Sonoff RF Bridge with ESPHome - Send commands

nmsoares, 2021-03-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff RF Bridge
**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.16.2
**Affected component:**

Send RF command

**Description of problem:**
Unable to send RF commands.
I have 2 Sonoff RF Bridges with ESPHome 1.16.2 installed. When sending the command in Hass.io :

service: esphome.sonoff_rf_bridge2_send_rf_code
data: {
sync: 0x303E,
low: 0x01A4,
high: 0x0500,
code: 0x41C401
}
the command is send and seen in the logs of RF Bridge:
[10:23:43][D][rf_bridge:147]: Sending code: sync=0x303E low=0x01A4 high=0x0500 code=0x41C401
[10:23:43][D][rf_bridge:036]: Action OK

But the device never responds or this not seen in my second RF Bridge using the same ESPHome version 1.16.2
I've also send a raw command :
service: esphome.sonoff_rf_bridge2_send_rf_raw
data:
  raw: "010000011100010000000001"

the command is send and seen in the logs:
[10:32:19][D][rf_bridge:205]: Sending Raw Code: 010000011100010000000001

but also never catched/seen in the second RF Bridge.

Both RF Bridges are 2 /3 meters apart from each other.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

lordzid, 2021-03-26:
the same situation.

jant90, 2021-04-03:
@nmsoares @lordzid initially I couldn't get it to work either but after converting the HEX values to decimal everything is working fine. So for your example try this:
```yaml
service: esphome.sonoff_rf_bridge2_send_rf_code
data:
  sync: 12350
  low: 420
  high: 1280
  code: 4310017
```

lordzid, 2021-04-03:
> @nmsoares @lordzid изначально я не мог заставить его работать, но после преобразования значений HEX в десятичные все работает нормально. Итак, для вашего примера попробуйте следующее:
> 
> ```yaml
> сервис : esphome.sonoff_rf_bridge2_send_rf_code 
> данные :
>    синхронизация : 12350 
>   низкий : 420 
>   высокий : 1280 
>   код : 4310017
> ```

can your yaml file for example?

jant90, 2021-04-03:
My RF Bridge yaml looks like this:
```yaml
esphome:
  name: sonoff_rf_bridge
  platform: ESP8266
  board: esp01_1m

status_led:
  pin:
    number: 13
    inverted: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  fast_connect: true
  power_save_mode: none

ota:
  password: !secret ota_pass

api:
  services:
    - service: send_rf_code
      variables:
        sync: int
        low: int
        high: int
        code: int
      then:
        - rf_bridge.send_code:
            sync: !lambda 'return sync;'
            low: !lambda 'return low;'
            high: !lambda 'return high;'
            code: !lambda 'return code;'
    - service: learn
      then:
        - rf_bridge.learn

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 19200

logger:
  baud_rate: 0

rf_bridge:
  on_code_received:
    then:
      - homeassistant.event:
          event: esphome.rf_code_received
          data:
            sync: !lambda 'char buffer [10];return itoa(data.sync,buffer,16);'
            low: !lambda 'char buffer [10];return itoa(data.low,buffer,16);'
            high: !lambda 'char buffer [10];return itoa(data.high,buffer,16);'
            code: !lambda 'char buffer [10];return itoa(data.code,buffer,16);'
```

closed 2021-08-15 by stale[bot]