esphome/issues #1938: ESPHome uses hyphens from its device name in user defined services for HA which causes errors

jant90, 2021-03-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff RF Bridge

**ESPHome version (latest production, beta, dev branch)**

1.16.2

**Affected component:**

API component, user-defined services
https://esphome.io/components/api.html#user-defined-services

**Description of problem:**
I set up a Sonoff RF Bridge with ESPHome using [the instructions from the ESPHome docs](https://esphome.io/components/rf_bridge.html).

I frequently use hyphens for device names out of habbit because underscores are forbidden in hostnames. In my case I set up the RF Bridge with the yaml found below.

This results in a service in HA named `esphome.sonoff-rf-bridge_send_rf_code`. While HA sees the service and from the Developer Tools menu I can even call it, as soon as I try to add the required data variables to it I get formatting errors. I tried adding the service to my configuration.yaml but again, all kinds of errors.

It took me a while to figure out what was wrong but as soon as I replaced the hyphens with underscores in the name of my ESPHome device and re-added it to HA all my issues went away. Now the service shows up as `esphome.sonoff_rf_bridge_send_rf_code` in HA, I can call it with data variables from the Developer Tools and add it to my configuration.yaml for template switches and automations.

I don't know if hyphens in service names in HA are actually forbidden but if they are then ESPHome should do something about it.

**Problem-relevant YAML-configuration entries:**
```yaml
# Initial yaml (with hyphens)
esphome:
  name: sonoff-rf-bridge
  platform: ESP8266
  board: esp01_1m

api:
  services:
    - service: send_rf_code
      variables:
        sync: int
        low: int
        high: int
        code: int

# Fixed yaml (with underscores)
esphome:
  name: sonoff_rf_bridge
```

**Logs (if applicable):**

```

```

**Additional information and things you've tried:**

glmnet, 2021-03-24:
I cannot reproduce this 

[image]

jant90, 2021-03-25:
Here is an example error from my HA log when using hyphens in the service name when configuring a switch:
```
2021-03-25 16:19:30 ERROR (MainThread) [homeassistant.config] Invalid config for [switch.template]: Service esphome.sonoff-rf-bridge_send_rf_code does not match format <domain>.<name> for dictionary value @ data['switches']['rf_switch_d']['turn_off'][0]['service']. Got None
Service esphome.sonoff-rf-bridge_send_rf_code does not match format <domain>.<name> for dictionary value @ data['switches']['rf_switch_d']['turn_on'][0]['service']. Got None. (See /config/configuration.yaml, line 85). Please check the docs at https://www.home-assistant.io/integrations/template
```

Here is the switch I configured in my HA configuration.yaml:
```
switch:
  - platform: template
    switches:
      rf_switch_d:
        turn_on:
          service: esphome.sonoff-rf-bridge_send_rf_code
          data:
            sync: 4650
            low: 140
            high: 450
            code: 8213588
        turn_off:
          service: esphome.sonoff-rf-bridge_send_rf_code
          data:
            sync: 4650
            low: 160
            high: 460
            code: 8213585
``` 
When I change the hyphens with underscores and rename my RF Bridge in ESPHome the switch is working perfectly without errors.

Full yaml for my RF Bridge:
```
esphome:
  name: sonoff-rf-bridge
  platform: ESP8266
  board: esp01_1m

status_led:
  pin:
    number: GPIO13
    inverted: True

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  fast_connect: true

ota:
  password: !secret ota_pass

api:
  services:
    - service: send_rf_code
      variables:
        sync: int
        low: int
        high: int
        code: int
      then:
        - rf_bridge.send_code:
            sync: !lambda 'return sync;'
            low: !lambda 'return low;'
            high: !lambda 'return high;'
            code: !lambda 'return code;'
    - service: learn
      then:
        - rf_bridge.learn

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 19200

logger:
  baud_rate: 0

rf_bridge:
  on_code_received:
    then:
      - homeassistant.event:
          event: esphome.rf_code_received
          data:
            sync: !lambda 'char buffer [10];return itoa(data.sync,buffer,16);'
            low: !lambda 'char buffer [10];return itoa(data.low,buffer,16);'
            high: !lambda 'char buffer [10];return itoa(data.high,buffer,16);'
            code: !lambda 'char buffer [10];return itoa(data.code,buffer,16);'
```

glmnet, 2021-03-29:
Alright, this happened to me now, but **it does work** when using the _Developers tools_, and does not work when using the _automation editor_, can you confirm this?

jant90, 2021-03-29:
Yes I can confirm this. Any idea what's causing the error?

glmnet, 2021-03-29:
Just another question, do you see this as a regression (i.e. it did worked before) or this never worked for you and this is the first time you are trying?

jant90, 2021-03-29:
This is the first time trying so I don't know if it's a regression, sorry.

glmnet, 2021-04-19:
Found the issue, will be fixed in next ha release most likely.

closed 2021-08-31 by stale[bot]