esphome/issues #1943: Value for "white" color ignored inside lambda

riker09, 2021-03-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

- ESP8266

**ESPHome version (latest production, beta, dev branch)**

- latest production

**Affected component:**

https://esphome.io/components/light/neopixelbus.html

**Description of problem:**

I have an RGBW LED strip that I control with NeoPixelBus. Inside a script I want to set the white value like this:

```yaml
## SCRIPT
script:
- id: strip_script
  mode: restart
  then:
  - light.turn_on:
      id: strip
      white: !lambda |-
        return id(gwhite);
      brightness: !lambda |-
        return id(gwhite);
```

The value comes from a brightness sensor (TEMT6000) that stores its readings every 5 seconds in a global variable:

```yaml
## SENSOR
sensor:
- platform: adc
  pin: A0
  id: brightness
  name: "${devicename_upper} Lux"
  unit_of_measurement: lx
  update_interval: 5s
  filters:
  - multiply: 200
  on_raw_value:
    ## Store current brightness value in global variable
  - globals.set:
      id: gwhite
      value: !lambda |-
        if (x < 0.1) {
          return (double) 0.1;
        } else if (x > 1) {
          return (double) 1;
        }
        return (double) (x);

## GLOBAL VALUE
globals:
- id: gwhite
  type: double
  restore_value: no
  initial_value: '0.4'

```

This is the strip config:

```yaml
## LED Strip
light:
- platform: neopixelbus
  id: strip
  type: GRBW
  pin: GPIO3
  num_leds: "${num_leds}"
  name: "${devicename_upper} LED Strip"
  restore_mode: ALWAYS_OFF
```

When I read the logs I can see that the brightness is set correctly to whatever value is stored in `gwhite`. 10% minimum when the raw_value is below `0.1` or `53%` when the value is `0.53`. The value for `white` is always at `100`.

**Additional information and things you've tried:**

I'm not sure wether to ask here first or on the Discord server. Apologies, if this is the wrong channel. A friendly nudge in the right direction would be much appreciated in that case.

oxan, 2021-07-23:
Can you try v1.20? This should be resolved now.

riker09, 2021-08-02:
Currently I'm lacking the time to check this. But judging from the code changes I take your word for it and consider this fixed. Nice job! 👍🏻

closed 2021-08-02 by riker09