esphome/issues #1946: Issue with multiple MCP23017

Patrryk, 2021-03-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant on QNAP Virtualization Station

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 D1 mini

**ESPHome version (latest production, beta, dev branch)**

Checked on 1.16.2 and latest dev

**Affected component:**

I2C Bus/MCP23017

**Description of problem:**
Issues with uploading OTA and flashing via ESP Home Flasher code presented below. This issue starts in case of trying to use more than three MCP23017 in setup. Setup for two MCP23017 works without any issue. MCP23017 are used as port expanders (outputs) for handling multiple relays. Code compiles properly according to logs, but uploading to device leads to failure. I cannot identify source of this problem therefore I am raising this topic as bug report.

**Problem-relevant YAML-configuration entries:**
```
esphome:
  name: przekazniki1
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 
  password: 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Przekazniki1 Fallback Hotspot"
    password: "bk0BRkatukF7"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 

ota:
  password: 

web_server:
  port: 80

i2c:
  sda: D1
  scl: D2
  scan: True
 
mcp23017:
  - id: 'mcp23017_hub1'
    address: 0x20
  - id: 'mcp23017_hub2'
    address: 0x21
  - id: 'mcp23017_hub3'
    address: 0x22

switch:
#mcp23017_hub1
- platform: gpio
  pin:
    mcp23017: mcp23017_hub1
    number: 8
    mode: OUTPUT
[... 1408 lines cut ...]
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 1 seconds
WARNING Error resolving IP address of przekazniki1.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 2 seconds
WARNING Error resolving IP address of przekazniki1.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 3 seconds
WARNING Error resolving IP address of przekazniki1.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 5 seconds
WARNING Error resolving IP address of przekazniki1.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 7 seconds
WARNING Error resolving IP address of przekazniki1.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 11 seconds
WARNING Error resolving IP address of przekazniki1.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -5] No address associated with hostname). Re-Trying in 17 seconds
```

**Additional information and things you've tried:**

Using ESPHome in ver. 1.16 and dev. Flashing via OTA and ESPHome Flasher. Flashing for two MCP's is working properly and I am able to use with this solution 32 realys but I need much more and then this issue occurs.

Patrryk, 2021-03-29:
This issue was also risen on HA community. One of users "Mikefila" has done separate tests with similar resault:

_It looks like there is too much and is throwing an actual stack overflow haha. Mine sits in a reboot loop right after flashing. Anyway a quick search for_

```
Stack canary watchpoint triggered
```

_Suggests increasing the stack size._

_Full boot loop output:_

```
[06:33:44]Rebooting...
[06:33:44]ets Jun  8 2016 00:22:57
[06:33:44]
[06:33:44]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[06:33:44]configsip: 0, SPIWP:0xee
[06:33:44]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[06:33:44]mode:DIO, clock div:2
[06:33:44]load:0x3fff0018,len:4
[06:33:44]load:0x3fff001c,len:1044
[06:33:44]load:0x40078000,len:8896
[06:33:44]load:0x40080400,len:5828
[06:33:44]entry 0x400806ac
[06:33:44][I][logger:166]: Log initialized
[06:33:44][C][ota:366]: There have been 5 suspected unsuccessful boot attempts.
[06:33:44][I][app:029]: Running through setup()...
[06:33:44][C][mcp23017:010]: Setting up MCP23017...
[06:33:44][W][i2c:070]: Received NACK on transmit of address 0x20
[06:33:44][E][component:092]: Component was marked as failed.
[06:33:44][C][mcp23017:010]: Setting up MCP23017...
[06:33:44][W][i2c:070]: Received NACK on transmit of address 0x21
[06:33:44][E][component:092]: Component was marked as failed.
[06:33:44][C][mcp23017:010]: Setting up MCP23017...
[06:33:44][W][i2c:070]: Received NACK on transmit of address 0x22
[06:33:44][E][component:092]: Component was marked as failed.
[06:33:44][C][mcp23017:010]: Setting up MCP23017...
[06:33:44][W][i2c:070]: Received NACK on transmit of address 0x25
[06:33:44][E][component:092]: Component was marked as failed.
[06:33:44][C][mcp23017:010]: Setting up MCP23017...
[06:33:44][W][i2c:070]: Received NACK on transmit of address 0x24
[06:33:44][E][component:092]: Component was marked as failed.
[06:33:44][C][switch.gpio:011]: Setting up GPIO Switch 'relay1'...
[06:33:44][D][switch:025]: 'relay1' Turning OFF.
[06:33:44]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
[06:33:44]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[06:33:44]Core 1 register dump:
[06:33:44]PC      : 0x4008be27  PS      : 0x00060336  A0      : 0x40085228  A1      : 0x3ffb0020  
[06:33:44]A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000004  A5      : 0x0000002d  
[06:33:44]A6      : 0x00000000  A7      : 0x00000004  A8      : 0x3ffb00f0  A9      : 0x3ffb00f0  
[06:33:44]A10     : 0x3ffb00f0  A11     : 0x0000002a  A12     : 0xffffffff  A13     : 0x3f4005cc  
[06:33:45]A14     : 0x3ffb00f0  A15     : 0x3ffb0314  SAR     : 0x00000012  EXCCAUSE: 0x00000001  
[06:33:45]EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  
[06:33:45]
[06:33:45]Backtrace: 0x4008be27:0x3ffb0020 0x40085225:0x3ffb03f0 0x40136d7a:0x3ffb0480 0x400d8621:0x3ffb04c0 0x400d8655:0x3ffb04f0 0x400d8689:0x3ffb0540 0x400d883c:0x3ffb0560 0x400df51d:0x3ffb0580 0x400df54a:0x3ffb05b0 0x400d9a79:0x3ffb0610 0x400d7761:0x3ffb0640 0x400d9957:0x3ffb0690 0x400d74c9:0x3ffb06b0 0x400d74f2:0x3ffb06e0 0x4016daa1:0x3ffb0700 0x4016db25:0x3ffb0720 0x400deafd:0x3ffb0740 0x400e924e:0x3ffb0790 0x400f772b:0x3ffb1fb0 0x40089699:0x3ffb1fd0
[06:33:45]
[06:33:45]Rebooting...
```

The stack size can/should be able to be increased. Esp’s look like they run on freertos 1 and the config is in a text file when it is compiled. This is more involved than I can help you with.

Esphome’s discord channel would probably be the best place to ask how to increase stack size.

The only other thing I could think to reduce the size of the code. I looked for a global automation, turn off action could be set for all the relays, but I couldn’t find anything. 

I started this post and as I wrote the last paragraph figured let me try without the wifi component. I removed wifi ota and api, sure enough it loads and runs!

I don’t know how much space the ethernet component adds. I still think discord might be the best spot to figure out if you can increase the stack size.

glmnet, 2021-03-29:
https://discord.com/channels/429907082951524364/524177279270780928/822457121941225473

copying for convenience:

~/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp
```
extern "C" void app_main()
{
    loopTaskWDTEnabled = false;
    initArduino();
    xTaskCreateUniversal(loopTask, "loopTask", 8192 * 4, NULL, 1, &loopTaskHandle, CONFIG_ARDUINO_RUNNING_CORE);
}
```
that *4 it

Patrryk, 2021-03-30:
Partial solution which allowed to use 48pcs of relays (not more, then the same issue occurred) was to disable unnecessary/unused part of code - web server, captive portal and ap. It is suspected that RAM value in this board might be to small to handle this.

closed 2021-05-13 by ssieb