esphome/issues #1948: PCF8575 Floating pins / INPUT_PULLUP

Tofandel, 2021-03-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP 32 (POE Olimex version)

**ESPHome version (latest production, beta, dev branch)**

1.16.2

**Affected component:**

https://esphome.io/components/pcf8574.html

**Description of problem:**
Does the pcf8575 needs external pullup resistors? I haven't found anything explicit saying it does or doesn't, but hinting more to the fact that there is indeed internal pullups

Right now I have a board setup with some floating pins and I'm seeing a few random on/off events
[image]

**Problem-relevant YAML-configuration entries:**
```yaml

i2c:
  sda: GPIO13
  scl: GPIO14
  scan: True
pcf8574:
  - id: 'pcf8575_hub'
    address: 0x21
    pcf8575: true

binary_sensor:
  - platform: gpio
    device_class: motion
    name: "Motion First Floor Room 1"
    pin:
      pcf8574: pcf8575_hub
      number: 1
      mode: INPUT
      inverted: false

```

**Logs (if applicable):**

I sometimes see this in the logs: maybe I have an underlying issue?
```
[12:46:05][W][i2c:086]: Requesting 2 bytes from 0x21 failed!
```

glmnet, 2021-03-29:
I'm not expert but ...
PCF8574, has one read register and one write register, so in order for it to work as Input and Output, the following happens:
If you write a Zero, the pin output is activated, like a NPN output the pins sink current (i.e. inverted logic pin).
If you write a One, the pin is high, but it's current is about a few micro amps, so not enough to actually turn on (or off) something, so you might need a pull up even as output to turn on e.g. a transistor.

Finally the port supports inputs, this just read the pin state, however you have to write a 1 on the pin you want to read (otherwise a zero will always be read)
So actually writing a 1 to enable the pin as input, enables a pull up on the pin.

Is the pin really not connected to anything? I believe is quite possible that if the pin has attached a long wire which is not connected to anything, then EMI can be picked up, specially if it runs close to high power lines.

Tofandel, 2021-03-29:
Interesting that you brought EMI up as indeed I may have a cable attached on this one (A few without cables and one with cable)
[image]

I'll check a bit latter to make sure it's indeed the pins that are doing random stuff that are connected to floating wires, in which case EMI will be the culprit, anything I can do to reduce it? Or will it be an issue once the cables are connected to ground and only open occasionally? (I'm guessing it could still trigger false positives instead of false negatives)

glmnet, 2021-03-29:
A stronger pull up and in both ends of the wire will help. Also shielded cable will help

Also a higher voltage, are you using just 3.3v signal? You can use 5V pull up but I'm not sure it will back feed vcc. 

Also make sure your connections aren't loose

Tofandel, 2021-03-29:
No risk of loose connections, using really good connectors,

So I'm actually feeding ground to the pin and using the internal pullup the rest of the time, so ground is going trough the wire normally and when motion is detected it opens the circuit (the long cable) and the internal pullup takes the relay and feeds 3v, I can try switching the power supply of the board to 5v to see if that helps, if still no luck I'll try switching to internal pulldown rather than pullup and feed 5v in the wire, not sure the board supports it though

Tofandel, 2021-03-29:
Okay I think I figured out my issue, basically when the circuit is closed EMI is minimized because the cables are tangled

But by default the circuit is open so one part of the cable is acting as a big antenna and the input pin receives it, so I could put a high resistor in between the relay so it's always closed

Tofandel, 2021-04-03:
So it seems I solved the issue, my first intuition was right, it wasn't EMI but I2C communication issue, I didn't use a pullup resistor on SDA and SCL which was causing the occasional `[13:04:35][W][i2c:086]: Requesting 2 bytes from 0x21 failed!` which in turn caused the pins to randomly turn on/off

After adding a 10K resistor between SDA and 3.3v and SCL and 3.3v it stopped showing up in the logs and so did the random on/off

Will monitor for a few days just in case

closed 2021-04-03 by Tofandel

andriej, 2021-07-22:
@Tofandel 10k on SDA and SCL pins declared in sketch both going to 3.3V solved all issues? Since april it's been some time, do you still keep the 'lesson learnt'?

It's shame it's not written anywhere, I've just found this issue and it seems that it will help...

Tofandel, 2021-07-22:
@andriej Yep, not a single issue since then

andriej, 2021-07-22:
Jesus, more than one year of issues can be solved with two resistors and not a single word anywhere near Olimex / ESP32 PoE / esphome etc. ;-) 

Will test it then and make PR in the docs just for all the others having the same 'floating' issue (I'm glad I came here, since my problem is with MCP23017 i2c and not PCF).

Tofandel, 2021-07-22:
Yeah technically the same issue would be there with any I2C device so it might be a good idea to warn about it in the I2C section

andriej, 2021-07-22:
Sorry to totally OT the issue (glad it's closed), but it may be the reason why some people tend NOT to have issues on esp8266 and issue persists on esp32.
They may be differently built and 'internal' pullup of esp8266 is enough and on esp32 it's not. 
I can't remember when I was so excited last time to test something worth few cents... ;-)

andriej, 2021-07-22:
I'm after first test with 2x10k to 3.3V, seems to work so far but still seeing:
`[W][i2c:086]: Requesting 1 bytes from 0x21 failed!`

No "MCP dropout" yet tho.

Tofandel, 2021-07-22:
I guess you can experiment with different resistor values see if it changes anything

andriej, 2021-07-22:
I wonder if my GPIOs (being):
```
  sda: GPIO14
  scl: GPIO15
```
could make a difference.

Tofandel, 2021-07-22:
You can always try, I put mine on which are the ones designed by default on ESP32 for I2C communication
  sda: GPIO13
  scl: GPIO14

andriej, 2021-07-24:
Still persists, I guess it's too much EMF on MCP. 

```
[W][i2c:086]: Requesting 1 bytes from 0x21 failed!
[W][i2c:073]: Received NACK on transmit of data for address 0x21
[W][i2c:076]: Unknown transmit error 4 for address 0x21
```

GPIO13 and 14 with 10k's.
I will switch over to SSRs to less the noise from relays and see if it helps.

SeByDocKy, 2021-11-13:
Hi,

Replying here to describe my problem with the PCF8575/L298N... 
Currently I am trying to control up to 4 x L298N with a PCF8575. My problem it seems the output voltage at HIGH state is too low (0.80V) to trig the L298N. Any tips to have higher value ? (except to have 16 pullup resistor, not easy to do with dupont wire. On the L298N break board there are R1,R2,R3 bridge indications. I don't know their role. Pullup resistor emplacement ?

Tofandel, 2021-11-14:
R1, R2, R3 are to give a unique address to your board since it's an I2c device, but that's not the scope of this issue and your problem is very different than what I described, so please, no highjacking of a closed issue

This repo is also not really a good place for this issue as it's mostly an electrical issue so i' d recommend https://electronics.stackexchange.com/ instead

SeByDocKy, 2021-11-14:
> R1, R2, R3 are to give a unique address to your board since it's an I2c device, but that's not the scope of this issue and your problem is very different than what I described, so please, no [...]

You mixed with A0,A1,A2 (for I2C adressing)....  I wrote R1,R2,R3 on this specific breakboard. Thanks for your nice comment...

Tofandel, 2021-11-15:
I did indeed mix them and I will reiterate that you need to ask for help somewhere else.

This is a repo for esphome which is software and you are only talking about an electrical issue between PCF8575/L298N, second you are commenting this on my closed issue so you are pretty much targeting your question only at me and I can't help you. So I directed you politely to where you should ask for help. I was not being bitter