esphome/issues #1949: Ultrasonic sensor echo_pin on i2c invalid input

meijerwynand, 2021-03-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** 

pip

**ESP8266 (NodeMCU):** 

EPS8266 NodeMCU

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.15.2

**Affected component:**

https://esphome.io/components/sensor/ultrasonic.html

**Description of problem:**
When attempting to add an ultrasonic distance sensor on a PCF8574, I am unable to specify the `echo_pin` as an "i2c" pin. The compiler stops with an error message `[pcf8574] is an invalid option for [echo_pin].`

**Problem-relevant YAML-configuration entries:**
```yaml
...
i2c:
  sda: D2
  scl: D1
  scan: True
  id: bus_a

pcf8574:
  - id: 'pcf8574_hub'
    address: 0x38
    pcf8575: False
...
```

### Expected config to work:
```yaml
  - platform: ultrasonic
    name: "Ultrasonic Sensor"
    trigger_pin:
      number: 3
      mode: INPUT
      pcf8574: pcf8574_hub
    echo_pin:
      number: 2
      mode: INPUT
      pcf8574: pcf8574_hub
```
### Expected config to work log:
```
> esphome  i2c.yaml compile
INFO Reading configuration i2c.yaml...
INFO Detected timezone 'SAST' with UTC offset 2
Failed config

sensor.ultrasonic: [source i2c.yaml:49]
  platform: ultrasonic
  name: Ultrasonic Sensor
  trigger_pin:
    number: 3
    mode: INPUT
    pcf8574: pcf8574_hub
  echo_pin:  [source i2c.yaml:56]
    number: 2
    mode: INPUT

    [pcf8574] is an invalid option for [echo_pin]. Please check the indentation.
    pcf8574: pcf8574_hub [source i2c.yaml:58]
```
### Expected config to work note:
The `trigger_pin` option does not appear to be subjected to this issue / limitation

**Additional information and things you've tried:**
### Alternative test
Here I set the `echo_pin` to a "normal pin" (non-i2c), keep `trigger_pin` as an i2c pin and I am able to compile 
```yaml
  - platform: ultrasonic
    name: "Ultrasonic Sensor"
    echo_pin:
      number: 2
      mode: INPUT
    trigger_pin:
      number: 3
      mode: INPUT
      pcf8574: pcf8574_hub
```
 
### Alternative test log
```
> esphome  i2c.yaml compile
INFO Reading configuration i2c.yaml...
INFO Detected timezone 'SAST' with UTC offset 2
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d i2c
Processing i2c (board: nodemcuv2; framework: arduino; platform: espressif8266@2.6.2)
...
Building .pioenvs/i2c/firmware.bin
RAM:   [=====     ]  45.5% (used 37296 bytes from 81920 bytes)
Flash: [====      ]  38.5% (used 401732 bytes from 1044464 bytes)
Creating BIN file ".pioenvs/i2c/firmware.bin" using "/home/xxx/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and ".pioenvs/i2c/firmware.elf"
====== [SUCCESS] Took 9.99 seconds ======
INFO Successfully compiled program.
```

### Observation:
It appears the `echo_pin` does not honor the Pin Schema as in the documentation.

Thanks

glmnet, 2021-04-03:
Well, did not test this but ESPHome measures time between sending a trigger and receiving an echo, while this could be done via a port expansor, getting reliable results will be a nightmare, so I recommend you to not use port expansion pins in this case.

meijerwynand, 2021-04-03:
> Well, did not test this but ESPHome measures time between sending a trigger and receiving an echo, while this could be done via a port expansor, getting reliable results will be a nightmare, so I recommend you to not use port expansion pins in this case.

That is what I was testing with, a port expander. The chip is supported as per [documentation](https://esphome.io/components/pcf8574.html).

As you mentioned results may not be as reliable, however, for the use case I have in mind should not be an issue. Using a distance sensor to check whether a vehicle is present in the garage. Thus nano second response time is not high on the requirements list. 

[image]

Thank you for taking the time to look into the issue and giving feedback. It is much appreciated.

ssieb, 2021-05-13:
The echo_pin is defined in the config to only accept a gpio pin, not a general pin config.  As already mentioned, this is due to the timing requirements and it most likely also uses an interrupt.  If your timing requirements are that flexible, you could implement it yourself with a `switch:` and `binary_sensor:` using a global variable to measure the time.

closed 2021-09-10 by meijerwynand