esphome/issues #1950: Higher power consumption on power plug than other firmwares

hanzoh, 2021-03-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

platform: ESP8266
board: esp8285
**ESPHome version (latest production, beta, dev branch)**

1.16.2
**Affected component:**

n/a

**Description of problem:**
I have flashed a Gosund SP1 plug with esphome and found that it then uses a constant 1.75W by itself, when the relay is on.
Then I flashed Tasmota and after that ESPurna on the same plug to compare power usage.
Both firmwares use a maximum power of ~1.15W, but they also lower it in between sending updates.
The esphome power usage was independent of yaml config, I even reduced it to the bare minimum (see below).
The best power consumption was on the original Tuya firmware, they seem to turn stuff off when it is not currently needed.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp01
  platform: ESP8266
  board: esp8285

wifi:
  ssid: "myssid"
  password: "mypassword"

ota:
```

**Logs (if applicable):**

```
n/a
```

**Additional information and things you've tried:**
n/a

rradar, 2021-05-12:
Did you try to "play" with `wifi` `power save mode` or `output power`?
> **power_save_mode** (Optional, string): The power save mode for the WiFi interface. See Power Save Mode
> 
> **output_power** (Optional, string): The amount of TX power for the WiFi interface from 10dB to 20.5dB. Default for ESP8266 is 20dB, 20.5dB might cause unexpected restarts.

https://esphome.io/components/wifi.html

hanzoh, 2021-05-12:
Yes, setting power save mode to LIGHT lowers the power consumption in both cases (normal and too-high power consumption) by about 0.3W.

I can now reproduce the too-high state like this:
- use tuya-convert when new or OTA flash when already converted to flash Tasmota
--> Tasmota: power consumption is about 0.6W when relay is off and power save mode has not kicked in
- do a power cycle by removing the plug from the outlet and plugging it back in
--> Tasmota: power consumption is still about 0.6W when relay is off and power save mode has not kicked in
- flash gzipped esphome.bin via Tasmota OTA
--> esphome: power consumption is about 0.6W when relay is off and power save mode has not kicked in
- do a power cycle by removing the plug from the outlet and plugging it back in
--> esphome: power consumption is about 1.0W when relay is off and power save mode has not kicked in
- flash gzipped esphome.bin via esphome OTA
--> esphome: power consumption is still about 1.0W when relay is off and power save mode has not kicked in

hanzoh, 2021-05-17:
I finally found the issue. I have added switches for every GPIO in yaml and set them to ```restore_mode: ALWAYS_OFF```.
After a power cycle the power consumption stays at 0.6W, just as expected and what is also happening with Tasmota et al.
I then turned the switches on, one by one, and GPIO15 was the culprit: Once I switched that on, power consumption went up to 1.0W.
The odd thing is: there is nothing connected to that.

So why does esphome enable this pin after a cold reboot, while Tasmota and ESPurna do not?

rradar, 2021-05-18:
> The odd thing is: there is nothing connected to that.

Would be interesting if this behavior could be reproduced on another device :thinking:

hanzoh, 2021-09-21:
I can still reproduce this on all four of my Gosund SP1 plugs.

JamesSwift, 2022-05-14:
Could you share your config for this?

Fma965, 2022-06-06:
FWIW i can replicate the same thing on a UP111, this should be added to the config as forcing GPIO15 to off works great, from 1w to 0.7w and less heat

twasilczyk, 2024-02-08:
I can confirm setting all GPIOs as pull-up/down inputs saves about 0.45W for Gosund WP5.

On the other hand, setting `power_save_mode` and `output_power` for Wi-Fi makes no measureable difference.