esphome/issues #1951: Sonoff S26: OTA timeout

TheNetStriker, 2021-03-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff S26

**ESPHome version (latest production, beta, dev branch)**

latest production

**Affected component:**

https://esphome.io/devices/sonoff_s20.html

**Description of problem:**
I flashed the ESPHome firmware to a Sonoff S26 using UART and everything worked fine besides the OTA update. The OTA update always fails after transferring about 30% of the firmware. (I’m using the esphome Docker image to compile and transfer the firmware and the esphomeflasher for the UART update.)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: sonoff01
  platform: ESP8266
  board: esp01_1m

wifi:
  domain: .mydomain
  networks:
    - ssid: "MYWIFI"
      bssid: "XX:XX:XX:XX:XX:XX"
      password: !secret MYWIFIPASSWORD

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sonoff01 Fallback Hotspot"
    password: !secret CAPTIVE_PORTAL_PASSWORD

mqtt:
  broker: 172.17.2.21
  client_id: sonoff01
  discovery: True
  birth_message:
    topic: sonoff01/availability
    payload: online
  will_message:
    topic: sonoff01/availability
    payload: offline

api:

logger:

ota:
  password: !secret OTA_PASSWORD

binary_sensor:
- platform: gpio
  pin:
    number: GPIO0
    mode: INPUT_PULLUP
    inverted: True
  name: "Sonoff S20 Button"
  on_press:
    - switch.toggle: relay
- platform: status
  name: "Sonoff S20 Status"

switch:
- platform: gpio
  name: "Sonoff S20 Relay"
  pin: GPIO12
  id: relay

output:
- platform: esp8266_pwm
  id: s20_green_led
  pin: GPIO13
  inverted: True

light:
- platform: monochromatic
  name: "Sonoff S20 Green LED"
  output: s20_green_led
```

**Logs (if applicable):**

OTA update logs:
```
INFO Uploading /config/sonoff01/.pioenvs/sonoff01/firmware.bin (424416 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error receiving acknowledge receive OK: timed out
```
ESP logs:
```
[12:53:02][D][ota:072]: Starting OTA Update from 172.17.128.204...
[12:53:02][D][ota:243]: OTA in progress: 0.2%
[12:53:03][D][ota:243]: OTA in progress: 5.0%
[12:53:04][D][ota:243]: OTA in progress: 9.9%
[12:53:06][D][ota:243]: OTA in progress: 14.7%
[12:53:07][D][ota:243]: OTA in progress: 19.5%
WARNING Disconnected from API: Timeout while waiting for message response!
```

**Additional information and things you've tried:**
I’ve tried to power cycle the device, setting a static ip and I also tried the **dev** Docker image, but the OTA update still fails. I’ve also tried setting `board: esp8285` and `arduino_version: 2.4.2`, but that does also not work.

I noticed that during the update the whole transfer seems to happen in under one second, but the **OTA in progress** logs are coming in very slowly. My guess was that there is a hard coded timeout somewhere that is set to low and breaks the connection while the device is still under 50% of the update. So I did take a look at the esphome code and found a socket timeout in the file **/usr/src/app/esphome/espota2.py**. After I've modified this file and set both socket timeouts to 120 seconds the OTA update worked.

I've also written a Dockerfile that modifies the existing image to increase this timeout:
```
FROM esphome/esphome
RUN   sed -i "s|sock.settimeout(20.0)|sock.settimeout(120.0)|g" /usr/src/app/esphome/espota2.py
RUN   sed -i "s|sock.settimeout(10.0)|sock.settimeout(120.0)|g" /usr/src/app/esphome/espota2.py
```
I'm now using this as a workarround, but I don't know if this is the correct solution for this problem. Maybe there is an underlying problem that the OTA updater thinks that the transfer is already finished while it is still transferring and this causes the **Uploading: [=======** progress bar to be completed in under one second.

Could you please take a look at this?

TTR6214, 2021-05-06:
Do you have tried to add `board_flash_mode: dout` as explain here? https://www.reddit.com/r/Esphome/comments/ey5y16/flashing_a_sonoff_s26_with_esphome/

TheNetStriker, 2021-05-06:
@TTR6214 Yes, I've also tried that, but this doesn't fix the problem.

TheNetStriker, 2021-05-25:
It seems that the problem is fixed with the latest version of the Docker container. (Could also be the latest ESPHome version) The progress bar for uploading the firmware now works again and is not completed in under a second like before. The issue can be closed.

closed 2021-05-25 by TheNetStriker