esphome/issues #1952: High current on deep sleep with interrupt?

jbrepogmailcom, 2021-03-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 FireBeetle

**ESPHome version (latest production, beta, dev branch)**

latest, dont know how to get it, ```esphome --version``` does not work

**Affected component:**

deep sleep

**Description of problem:**
When I use deep sleep, the current is about 11μA. As soon as I add wakeup pin, the current during deep sleep rises to 1500μA. Is the deep sleep with interrupt wakeup configured correctly?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```
deep_sleep:
  id: deep_sleep_1
  run_duration: 3s
  sleep_duration: 20s
  wakeup_pin:
    number: 27
```

**Logs (if applicable):**
<!--
Please copy the debug log here. If possible, also connect to the ESP over USB and copy those logs into the backticks.
-->
```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

Papadoma, 2021-04-09:
I've experienced the same behaviour and already raised it as an issue here https://github.com/esphome/issues/issues/1668#issue-756214233

OttoWinter, 2021-04-10:
Closing as duplicate

closed 2021-04-10 by OttoWinter