esphome/issues #1955: Cannot compile: Error: Detected unknown package 'tool-scons'

TotallyInformation, 2021-04-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10, PowerShell 7, ESPHome v1.14.5, Python 3.8.8, pip 21.0.1
Note that I am not using HA.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Wemos D1 Mini Pro

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.14.5

**Affected component:**

N/A, none added

**Description of problem:**

Cannot compile.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: d1m03
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: "<my ssid>"
  password: "<my password>"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "D1M03 Fallback Hotspot"
    password: "jfNNMauyiZ0B"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
> esphome d1m03.yaml compile
INFO Reading configuration d1m03.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d d1m03
Processing d1m03 (board: d1_mini_pro; framework: arduino; platform: espressif8266@2.2.3)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
CorePackageManager: Installing tool-scons @ ~3.30102.0
Error: Detected unknown package 'tool-scons'
```

**Additional information and things you've tried:**

Google search to see if anyone else has had this issue. It appears to be a Platform.io error of some kind but I don't use Platform.io

TotallyInformation, 2021-04-02:
In addition, if I try to compile for an ESP32 device I get the error:

```
> esphome ttgo.yaml compile
INFO Reading configuration ttgo.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
INFO Running:  platformio run -d ttgo
Processing ttgo (board: ttgo-t1; framework: arduino; platform: espressif32@1.12.1)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
PlatformManager: Installing espressif32 @ 1.12.1
Error: Detected unknown package 'espressif32'
```

TotallyInformation, 2021-04-03:
And yet more issues - when trying to compile for a ttgo t-display (as board type featheresp):

```
Failed config

text_sensor.wifi_info: [source ttgo.yaml:84]
...

  [mac_address] is an invalid option for [text_sensor.wifi_info]. Did you mean [ip_address]?
  mac_address:  [source ttgo.yaml:95]
    name: ESP Mac Wifi Address
    state_topic: ESP/ttgo01/info/mac
color:

  Component not found: color.

  ...

display.st7789v: [source ttgo.yaml:214]

  Platform not found: 'display.st7789v'.
  ...
```

OttoWinter, 2021-04-10:
> And yet more issues - when trying to compile for a ttgo t-display (as board type featheresp):
> 
> ```
> Failed config
> 
> text_sensor.wifi_info: [source ttgo.yaml:84]
> ...
> 
>   [mac_address] is an invalid option for [text_sensor.wifi_info]. Did you mean [ip_address]?
>   mac_address:  [source ttgo.yaml:95]
>     name: ESP Mac Wifi Address
>     state_topic: ESP/ttgo01/info/mac
> color:
> 
>   Component not found: color.
> 
>   ...
> 
> display.st7789v: [source ttgo.yaml:214]
> 
>   Platform not found: 'display.st7789v'.
>   ...
> ```
> 
> Noting that using WSL works fine. BUT, trying to edit the yaml files with VScode and the ESPHome extension gives these same errors which is less than helpful.

That system appears to have an old version of esphome installed. please check you're using the latest version.

closed 2021-04-10 by OttoWinter

TotallyInformation, 2021-04-14:
Thanks Otto, I did eventually discover that at the weekend. I really don't know why though since I installed it recently using the instructions on the website.

Some faffing and forcing eventually resulted in the esphome app being brought up to the current version and it now works.