esphome/issues #1960: Excessive stack usage in setup() due to font setup

numo68, 2021-04-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip under Debian Linux

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 Devkit

**ESPHome version (latest production, beta, dev branch)**

dev branch

**Affected component:**

code generation

**Description of problem:**
Larger setups tend to crash, see e.g. #855 for the symptoms. The culprit seem to be an excessive stack usage of the generated setup() function.

I am including the output for `-fno-inline -Wstack-usage=256`. The `setup()` is indeed a clear winner with a 3120 bytes of stack (3248 if inlining is enabled so inluned methods are not the problem here). That did not make much sense to me as pretty much everything put on stack there are the negligible colors and `wifi::WiFiAP` (with sizeof 112). 

After some more digging my opinion is that the stack usage is connected to the font creation generated as

```
font1 = new display::Font({display::Glyph(" ", uint8_t_2, 0, 0, 15, 5, 0), display::Glyph("!", uint8_t_2, 0, 0, 3, 6, 12), ...
```
I am not an expert on C++ compiler details but it is the _only_ place that smells of stack usage. I changed the font to include just one glyph which immediately brought the stack usage to 1424 bytes. So in my opinion this should be better done somehow iteratively.

Also note the `void esphome::ota::OTAComponent::handle_()` with 1248 bytes that indeed puts a char[1024] on stack; that should be probably fixed but unlikely to kick in here. Quite a few 3rd party methods are flagged with a potentially unbounded usage mostly doing `char foo[strlen(something)...]` but neither seems to likely ask for too much in the common scenarios.

**Problem-relevant YAML-configuration entries:**
```yaml
font:
  - file: "fonts/DejaVuSans.ttf"
    id: font1
    size: 16
    # glyphs: [ '0' ]
```

**Logs (if applicable):**

```
src/esphome/components/display/display_buffer.cpp: In member function 'void esphome::display::DisplayBuffer::vprintf_(int, int, esphome::display::Font*, esphome::Color, esphome::display::TextAlign, const char*, va_list)':
src/esphome/components/display/display_buffer.cpp:201:6: warning: stack usage is 336 bytes [-Wstack-usage=]
 void DisplayBuffer::vprintf_(int x, int y, Font *font, Color color, TextAlign align, const char *format, va_list arg) {
      ^
src/esphome/components/ota/ota_component.cpp: In member function 'void esphome::ota::OTAComponent::handle_()':
src/esphome/components/ota/ota_component.cpp:51:6: warning: stack usage is 1248 bytes [-Wstack-usage=]
 void OTAComponent::handle_() {
      ^
src/esphome/components/wifi/wifi_component_esp32.cpp: In member function 'bool esphome::wifi::WiFiComponent::wifi_sta_connect_(esphome::wifi::WiFiAP)':
src/esphome/components/wifi/wifi_component_esp32.cpp:152:6: warning: stack usage is 336 bytes [-Wstack-usage=]
 bool WiFiComponent::wifi_sta_connect_(WiFiAP ap) {
      ^
src/esphome/components/wifi/wifi_component.cpp: In member function 'void esphome::wifi::WiFiComponent::check_scanning_finished()':
src/esphome/components/wifi/wifi_component.cpp:329:6: warning: stack usage is 288 bytes [-Wstack-usage=]
 void WiFiComponent::check_scanning_finished() {
      ^
src/main.cpp: In function 'void setup()':
src/main.cpp:92:6: warning: stack usage is 3120 bytes [-Wstack-usage=]
 void setup() {
      ^
/home/stano/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/WiFi/src/WiFiAP.cpp: In member function 'bool WiFiAPClass::softAP(const char*, const char*, int, int, int)':
/home/stano/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/WiFi/src/WiFiAP.cpp:92:6: warning: stack usage is 288 bytes [-Wstack-usage=]
 bool WiFiAPClass::softAP(const char* ssid, const char* passphrase, int channel, int ssid_hidden, int max_connection)
      ^
/home/stano/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/WiFi/src/WiFiSTA.cpp: In member function 'wl_status_t WiFiSTAClass::begin(const char*, const char*, int32_t, const uint8_t*, bool)':
/home/stano/.platformio/packages/framework-arduinoespressif32@3.10004.210126/libraries/WiFi/src/WiFiSTA.cpp:115:13: warning: stack usage is 288 bytes [-Wstack-usage=]
 wl_status_t WiFiSTAClass::begin(const char* ssid, const char *passphrase, int32_t channel, const uint8_t* bssid, bool connect)
             ^
[... 22 lines cut ...]
 void AsyncStaticWebHandler::handleRequest(AsyncWebServerRequest *request)
      ^
.piolibdeps/gui/ESPAsyncWebServer-esphome/src/WebRequest.cpp: In member function 'bool AsyncWebServerRequest::_parseReqHeader()':
.piolibdeps/gui/ESPAsyncWebServer-esphome/src/WebRequest.cpp:312:6: warning: stack usage is 320 bytes [-Wstack-usage=]
 bool AsyncWebServerRequest::_parseReqHeader(){
      ^
.piolibdeps/gui/ESPAsyncWebServer-esphome/src/WebResponses.cpp: In member function 'virtual String AsyncWebServerResponse::_assembleHead(uint8_t)':
.piolibdeps/gui/ESPAsyncWebServer-esphome/src/WebResponses.cpp:126:8: warning: stack usage might be 400 bytes [-Wstack-usage=]
 String AsyncWebServerResponse::_assembleHead(uint8_t version){
        ^
.piolibdeps/gui/ESPAsyncWebServer-esphome/src/WebResponses.cpp: In constructor 'AsyncFileResponse::AsyncFileResponse(AsyncFileResponse::FS&, const String&, const String&, bool, AwsTemplateProcessor)':
.piolibdeps/gui/ESPAsyncWebServer-esphome/src/WebResponses.cpp:507:1: warning: stack usage might be unbounded [-Wstack-usage=]
 AsyncFileResponse::AsyncFileResponse(FS &fs, const String& path, const String& contentType, bool download, AwsTemplateProcessor callback): AsyncAbstractResponse(callback){
 ^
.piolibdeps/gui/ESPAsyncWebServer-esphome/src/WebResponses.cpp: In constructor 'AsyncFileResponse::AsyncFileResponse(AsyncFileResponse::File, const String&, const String&, bool, AwsTemplateProcessor)':
.piolibdeps/gui/ESPAsyncWebServer-esphome/src/WebResponses.cpp:541:1: warning: stack usage might be unbounded [-Wstack-usage=]
 AsyncFileResponse::AsyncFileResponse(File content, const String& path, const String& contentType, bool download, AwsTemplateProcessor callback): AsyncAbstractResponse(callback){
 ^
/home/stano/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Esp.cpp: In function 'uint32_t sketchSize(sketchSize_t)':
/home/stano/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/Esp.cpp:144:17: warning: stack usage is 304 bytes [-Wstack-usage=]
 static uint32_t sketchSize(sketchSize_t response) {

```

**Additional information and things you've tried:**

numo68, 2021-04-04:
Hmm. Interestlingly the rising number of binary sensor enlarges the stack usage as well by about 48 bytes per sensor. The increments seem to be connected with the lambdas and initializer lists. A `foo->set_bar({ baz })` takes 32 bytes, an `on_state` definition takes 16. No playing with `{` `}` around the blocks seems to help. The C++ probably keeps the lambda captures until the whole thing gets out of the function scope or something like that. Not sure whether this is the way it should be but it is...

I attempted to switch off generating the `main.cpp` and split the `setup()` into `setup1()`, `setup2()` etc. As with the exception of `wifi::WiFiAP` and the fonts the `setup()` only accesses globals this does not create significant problems and allows bringing the stack space down. I have however no idea how complicated or even desirable it would be to do such workaround as a "normal" way. Bumping the stack size is probably a lot easier...

numo68, 2021-04-19:
Well, looks like this is a compiler issue. All the initializer lists indeed consume the stack. I did not find any way to stop it inside one function so there does not seem to be any other way than to try to cut the `setup()` to smaller pieces...

```cpp
#include <vector>

static std::vector<void *> dummyVector;

void copyToVector(const std::vector<void *> &v) {
    dummyVector = v;
}

void method1() {
    copyToVector({nullptr, nullptr, nullptr, nullptr});
}

void method2() {
    copyToVector({nullptr, nullptr, nullptr, nullptr});
    copyToVector({nullptr, nullptr, nullptr, nullptr});
}
```

```
$ g++ -Wstack-usage=32 -S stack_size.cpp 
stack_size.cpp: In function ‘void method1()’:
stack_size.cpp:9:6: warning: stack usage is 128 bytes [-Wstack-usage=]
    9 | void method1() {
      |      ^~~~~~~
stack_size.cpp: In function ‘void method2()’:
stack_size.cpp:13:6: warning: stack usage is 224 bytes [-Wstack-usage=]
   13 | void method2() {
      |      ^~~~~~~
```

```asm
_Z7method2v:
	...
	movq	$0, -176(%rbp)
	movq	$0, -168(%rbp)
	movq	$0, -160(%rbp)
	movq	$0, -152(%rbp)
	...
	call	_ZNSt6vectorIPvSaIS0_EEC1ESt16initializer_listIS0_ERKS1_
	...
	movq	$0, -96(%rbp)
	movq	$0, -88(%rbp)
	movq	$0, -80(%rbp)
	movq	$0, -72(%rbp)
	...
	call	_ZNSt6vectorIPvSaIS0_EEC1ESt16initializer_listIS0_ERKS1_
```

numo68, 2021-04-19:
Erm... Looks like the following works to force a stack frame but man is this ugly...

```cpp
void method2() {
    [&]() {
        copyToVector({nullptr, nullptr, nullptr, nullptr});
    }();
    [&]() {
        copyToVector({nullptr, nullptr, nullptr, nullptr});
    }();
}
```

numo68, 2021-04-20:
> Do we really need all those lambdas?

Nope, I just searched for the way to give back the claimed stack as soon as possible and this was the only thing I found without needing to understand the black magic of the code generator ;) The lambdas probably even take some stack to themselves, just less.

Frankly this seems like a compiler deficiency to me (gcc was never exactly embedded-friendly) and I am considering logging an issue to gcc folks as well to see what they think but I guess that the gcc just works with allocating the stack once per function, moving it further on `alloca()` and `new foo[variable]` but never reclaiming it nor trying to reuse the allocated place for different things.

> The setup split doesn’t seem too bad to me, all components are globals so shouldn’t be too hard.
> 
> Will this issue be solved by creating a setup function per component?

Absolutely (probably with the `noinline` attribute as well). Everything is indeed global so it should be doable. I just did not see any way to do it quickly due all the coroutine stuff because of the dependency solver.

The fonts are still extreme as the initializer list is huge. One font definition can easily eat 1/4 of the default stack, maybe more if I need a "full" font including all the national characters. Here probably more could be moved to progmem as a fixed structure and let the font code handle that. I am seeing the code split and the font refactoring as two orthogonal tasks.

numo68, 2021-04-20:
The above PR does the following

```cpp
  //   raw_data_id: uint8_t_2
  //   raw_glyph_id: display_glyphdata
  static const uint8_t uint8_t_2[] PROGMEM = {0x18, 0x18, 0x18, 0x18, 0x18, 0x18, ...
  static const display::GlyphData display_glyphdata[] = {display::GlyphData{
    .a_char = " ",
    .data = uint8_t_2 + 0,
    .offset_x = 0,
    .offset_y = 19,
    .width = 6,
    .height = 0,
  }, display::GlyphData{
    .a_char = "!",
    .data = uint8_t_2 + 0,
    .offset_x = 0,
    .offset_y = 4,
    .width = 8,
    .height = 15,
  }, display::GlyphData{
    .a_char = "\042",
    .data = uint8_t_2 + 15,
  ...
  }};
  sans20 = new display::Font(display_glyphdata, 75, 19, 24);
```

alleviating the single most stack usage caused by just a few lines.

skandalfo, 2021-04-22:
Why wouldn't we move display_glyphdata to PROGMEM/flash too?

The GlyphData struct is trivial and I'd be surprised it's not placed into .rodata even without the PROGMEM annotation.

numo68, 2021-04-22:
> Why wouldn't we move display_glyphdata to PROGMEM/flash too?

The only reason is that I don't have any ESP8266 to test. As far as I understand the PROGMEM is a noop on ESP32 but needs to be moved to RAM before usage on the 8266, so I am reluctant to commit something I cannot test.

skandalfo, 2021-04-22:
I think this should be the relevant reference for Arduinio-style PROGMEM in ESP8266: https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html#guide-to-progmem-on-esp8266-and-arduino-ide

I don't think you need a real ESP8266 to verify what this does. You might be able to just run "esphome compile" on a test configuration and inspect the resulting firmware.elf with readelf to check symbols and sections. You will likely need to remove the "static" qualifier for symbols to be generated though.

numo68, 2021-04-22:
> I don't think you need a real ESP8266 to verify what this does. You might be able to just run "esphome compile" on a test configuration and inspect the resulting firmware.elf with readelf to check [...]

This is not what I am afraid of. As far as I understand any *access* needs to go through an indirection layer, similarly to what `Glyph::get_pixel` does with `pgm_read_byte(this->data_ + ...`. There are enough accesses to this data to trust myself not to miss anything without a way to test. The compiler is not going to warn me.

skandalfo, 2021-04-22:
Oh, right... the ESP8266's flash isn't byte-addressable for data access. You'd need to either use accessors for all field accesses or ensure all fields are 32-bit wide and 32-bit aligned, which is quite a low-level thing to guarantee/enforce.

You could pepper a lot of assertions checking sizeof/offsetof for every field and maybe get away with it but, yes, I understand this is something that'd need testing.

I'll try to check whether this is possible to do if I have the time next weekend (I do have 8266s around).

closed 2021-09-04 by stale[bot]