esphome/issues #1966: SIM800L and NodeMCU / ESP32 DevKit

baerni, 2021-04-07:
Has anyone successfully connected a NodeMCU or ESP32 DevKit to a SIM800L?
I read [this](https://community.home-assistant.io/t/esp32-sim800l-ttgo-t-call-not-working/149781), [this](https://github.com/esphome/issues/issues/1015) and [this](https://github.com/esphome/issues/issues/1326), tried different solutions (gpio change to rx and tx, gpio change to D1 and D2, baund_rate to 9600 and 115200, etc.) but nothing helps.
I have RSSI: 0dB all the time.
I do not have an ESP32 board with a built-in SIM800L like those who write there, only standalone NodeMCU and ESP32 DevKit boards.
The LED on the SIM800L flashes once every 3 seconds, which means that it is properly connected to the network.
Please help.

closed 2021-04-07 by close-issue-app[bot]

glmnet, 2021-04-07:
please fill an issue properly, logs needed

Tylersuard, 2021-04-12:
@baerni Did you find a solution to this?  Looking for the same answer myself.

baerni, 2021-04-13:
Yes. I opened a new issue according to the template and everything became clear here. 
[Here is the link](https://github.com/esphome/issues/issues/1969).