esphome/issues #1968: How used LittleFS?

SCratORS, 2021-04-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2021-04-08:
```
esphome:
  platformio_options:
    board_build.filesystem: littlefs
```
note the `:` - this is in a YAML file, so needs to be in YAML syntax.

SCratORS, 2021-04-08:
I know that.
But the YAML editor reports that "error: expected a dictionary"

SCratORS, 2021-04-08:
Ok.
board_build.filesystem: littlefs
custom component code:
`include <LittleFS.h>
...
LittleFS.format(); //return false
LittleFS.begin(); //return false
`
Does not work.

SCratORS, 2021-04-08:
SOLVED. Thanks.
```
esphome: 
  platformio_options:
    board_build.filesystem: littlefs
    board_build.ldscript: eagle.flash.4m3m.ld
```
**I wish it was written somewhere...**

closed 2021-04-08 by SCratORS