esphome/issues #1969: SIM800L and NodeMCU / ESP32 DevKit

baerni, 2021-04-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32/ESP8266

**ESPHome version (latest production, beta, dev branch)**

1.16

**Affected component:**

Sim800L

**Description of problem:**
Has anyone successfully connected a NodeMCU or ESP32 DevKit to a SIM800L?
I read [this](https://community.home-assistant.io/t/esp32-sim800l-ttgo-t-call-not-working/149781), [this ](https://github.com/esphome/issues/issues/1015)and [this](https://github.com/esphome/issues/issues/1326), tried different solutions (gpio change to rx and tx, gpio change to D1 and D2, baund_rate to 9600 and 115200, etc.) but nothing helps.
I have RSSI: 0dB all the time.
I do not have an ESP32 board with a built-in SIM800L like those who write there, only standalone NodeMCU and ESP32 DevKit boards.
The LED on the SIM800L flashes once every 3 seconds, which means that it is properly connected to the network.
Please help.

**Problem-relevant YAML-configuration entries:**
```yaml

uart:
  baud_rate: 115200
  tx_pin: TX
  rx_pin: RX

sim800l:
  on_sms_received:
    - logger.log:
        format: "Received '%s' from %s"
        args: [ 'message.c_str()', 'sender.c_str()' ]

logger:
  baud_rate: 0 # disable uart logger on esp 8266

binary_sensor:
  - platform: gpio
    pin: GPIO5
    name: "sensor"
    on_press:
      then:
        - sim800l.send_sms:
            recipient: '+XXXXXXXXXXX'
            message: 'Hello there'

```

**Logs (if applicable):**

```
[13:05:36][C][sim800l:259]: SIM800L:
[13:05:36][C][sim800l:260]:   RSSI: 0 dB

```

**Additional information and things you've tried:**

glmnet, 2021-04-08:
Please increase log level:

```yaml

# Enable logging
logger:
  baud_rate: 0 # disable log by hardware UART
  level: VERBOSE
  logs:
    i2c: DEBUG
    uart: DEBUG
```

you should get something like this:

```
INFO Connecting to eh_planta_alta.local:6053 (10.0.1.173)
INFO Successfully connected to eh_planta_alta.local
[11:51:00][I][app:105]: ESPHome version 1.18.0-dev compiled on Apr  8 2021, 11:50:36
[11:51:00][C][wifi:443]: WiFi:
[11:51:00][C][wifi:303]:   SSID: 'xxxx'
[11:51:00][C][wifi:304]:   IP Address: 10.0.1.173
[11:51:00][C][wifi:306]:   BSSID: 74:4D:28:60:F1:A6
[11:51:00][C][wifi:307]:   Hostname: 'eh_planta_alta'
[11:51:00][C][wifi:311]:   Signal strength: -36 dB ▂▄▆█
[11:51:00][C][wifi:315]:   Channel: 1
[11:51:00][C][wifi:316]:   Subnet: 255.255.255.0
[11:51:00][C][wifi:317]:   Gateway: 10.0.1.1
[11:51:00][C][wifi:318]:   DNS1: 10.0.1.1
[11:51:00][C][wifi:319]:   DNS2: 190.57.228.27
[11:51:00][C][i2c:028]: I2C Bus:
[11:51:00][C][i2c:029]:   SDA Pin: GPIO4
[11:51:00][C][i2c:030]:   SCL Pin: GPIO5
[11:51:00][C][i2c:031]:   Frequency: 50000 Hz
[11:51:00][I][i2c:033]: Scanning i2c bus for active devices...
[11:51:00][I][i2c:040]: Found i2c device at address 0x08
[11:51:00][I][i2c:040]: Found i2c device at address 0x09
[11:51:01][C][uart_esp8266:075]: UART Bus:
[11:51:01][C][uart_esp8266:077]:   TX Pin: GPIO1
[11:51:01][C][uart_esp8266:080]:   RX Pin: GPIO3
[11:51:01][C][uart_esp8266:081]:   RX Buffer Size: 256
[11:51:01][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[11:51:01][C][uart_esp8266:084]:   Data Bits: 8
[11:51:01][C][uart_esp8266:085]:   Parity: NONE
[11:51:01][C][uart_esp8266:086]:   Stop bits: 1
[11:51:01][C][uart_esp8266:088]:   Using hardware serial interface.
[11:51:01][C][template.text_sensor:020]: Template Sensor 'Sms Sender'
[11:51:01][C][template.text_sensor:020]: Template Sensor 'Sms Message'
[11:51:01][C][output.switch:009]: Output Switch 'Bomba Presurizadora'
[11:51:01][V][sim800l:044]: S: AT - 1
[11:51:01][C][logger:185]: Logger:
[11:51:01][C][logger:186]:   Level: VERBOSE
[11:51:01][C][logger:187]:   Log Baud Rate: 0
[11:51:01][C][logger:188]:   Hardware UART: UART0
[11:51:01][C][logger:190]:   Level for 'i2c': DEBUG
[11:51:01][C][logger:190]:   Level for 'uart': DEBUG
[11:51:01][C][light:178]: Light 'Luz baño Planta alta'
[11:51:01][C][light:178]: Light 'Luz Escalera'
[11:51:01][C][light:178]: Light 'Luz Ante baño Planta alta'
[11:51:01][C][light:178]: Light 'Luz Hall Planta alta'
[11:51:01][C][light:178]: Light 'Luz Patio Este'
[11:51:01][C][light:178]: Light 'Luz Patio Oeste'
[11:51:01][C][sim800l:279]: SIM800L:
[11:51:01][C][sim800l:280]:   RSSI: 0 dB
[11:51:01][C][status:034]: Status Binary Sensor 'Planta Alta Status'
[11:51:01][C][status:034]:   Device Class: 'connectivity'
[11:51:01][C][restart:021]: Restart Switch 'Planta Alta Restart'
[11:51:01][C][restart:021]:   Icon: 'mdi:restart'
[11:51:01][C][ota:029]: Over-The-Air Updates:
[11:51:01][C][ota:030]:   Address: eh_planta_alta.local:8266
[11:51:01][C][api:095]: API Server:
[11:51:01][C][api:096]:   Address: eh_planta_alta.local:6053
[11:51:01][C][homeassistant.binary_sensor:032]: Homeassistant Binary Sensor 'Switch Escalera 1 from HA'
[11:51:01][C][homeassistant.binary_sensor:033]:   Entity ID: 'binary_sensor.switch_escalera_1'
[11:51:02][V][api.connection:601]: Hello from client: 'Home Assistant 2021.4.0 (10.0.1.206)'
[11:51:02][D][api.connection:617]: Client 'Home Assistant 2021.4.0 (10.0.1.206)' connected successfully!
[11:51:03][D][homeassistant.binary_sensor:021]: 'binary_sensor.switch_escalera_1': Got state OFF
[11:51:03][D][binary_sensor:034]: 'Switch Escalera 1 from HA': Sending initial state OFF
[11:51:08][D][binary_sensor:034]: 'Tanque Agua Arriba Lleno': Sending initial state OFF
[11:51:16][V][sim800l:044]: S: AT - 1
[11:51:16][V][sim800l:051]: R: ? - 2
[11:51:16][W][sim800l:065]: Not ack. 2 ?
[11:51:16][V][sim800l:051]: R: OK - 0
[11:51:16][D][sim800l:236]: Unhandled: OK - 0
[11:51:31][V][sim800l:044]: S: AT - 1
[11:51:31][V][sim800l:051]: R:  - 2
[11:51:31][V][sim800l:051]: R: OK - 2
[11:51:31][V][sim800l:044]: S: AT+CMGF=1 - 2
[11:51:31][V][sim800l:051]: R:  - 3
[11:51:31][V][sim800l:051]: R: OK - 3
[11:51:31][V][sim800l:044]: S: AT+CREG? - 3
[11:51:31][V][sim800l:051]: R:  - 4
[11:51:31][V][sim800l:051]: R: +CREG: 0,1 - 4
[11:51:31][D][sim800l:106]: Registered OK
[11:51:31][V][sim800l:051]: R:  - 5
[11:51:31][V][sim800l:051]: R: OK - 5
[11:51:31][V][sim800l:044]: S: AT+CSQ - 5
[11:51:31][V][sim800l:051]: R:  - 6
[11:51:31][V][sim800l:051]: R: +CSQ: 24,0 - 6
[11:51:31][D][sim800l:132]: RSSI: 24
[11:51:31][V][sim800l:051]: R:  - 11
[11:51:31][V][sim800l:051]: R: OK - 11
[11:51:31][V][sim800l:044]: S: AT+CMGL="ALL" - 11
[11:51:31][V][sim800l:051]: R:  - 12
[11:51:31][V][sim800l:051]: R: +CMGL: 1,"REC READ","74@62634861636@6","","21/04/05,10:10:00-12" - 12
[11:51:31][V][sim800l:051]: R: Para cuidarnos entre todos enviamos autoevaluacion de sintomas COVID19 covid.chaco.gob.ar Tu ayuda es importante para la deteccion temprana. - 13
[11:51:31][V][sim800l:051]: R:  - 13
[11:51:31][V][sim800l:051]: R: OK - 13
[11:51:36][V][sim800l:044]: S: AT - 1
[11:51:36][V][sim800l:051]: R:  - 2
[11:51:36][V][sim800l:051]: R: OK - 2
[11:51:36][V][sim800l:044]: S: AT+CMGF=1 - 2
[11:51:36][V][sim800l:051]: R:  - 3
[11:51:36][V][sim800l:051]: R: OK - 3
[11:51:36][V][sim800l:044]: S: AT+CREG? - 3
[11:51:36][V][sim800l:051]: R:  - 4
[11:51:36][V][sim800l:051]: R: +CREG: 0,1 - 4
```

Please share complete log (you can remove any sensitive data, but keep it as complete as possible)

baerni, 2021-04-08:
```
18:43:07][C][wifi:443]: WiFi:
[18:43:07][C][wifi:303]:   SSID: 'XXX'
[18:43:07][C][wifi:304]:   IP Address: 192.168.0.39
[18:43:07][C][wifi:306]:   BSSID: XXXXXXXX
[18:43:07][C][wifi:307]:   Hostname: 'esp32test'
[18:43:07][C][wifi:311]:   Signal strength: -55 dB ▂▄▆█
[18:43:07][V][wifi:313]:   Priority: 0.0
[18:43:07][C][wifi:315]:   Channel: 3
[18:43:07][C][wifi:316]:   Subnet: 255.255.255.0
[18:43:07][C][wifi:317]:   Gateway: 192.168.0.1
[18:43:07][C][wifi:318]:   DNS1: 192.168.0.102
[18:43:07][C][wifi:319]:   DNS2: 192.168.0.102
[18:43:07][C][uart_esp32:088]: UART Bus:
[18:43:07][C][uart_esp32:090]:   TX Pin: GPIO19
[18:43:07][C][uart_esp32:093]:   RX Pin: GPIO18
[18:43:07][C][uart_esp32:094]:   RX Buffer Size: 256
[18:43:07][C][uart_esp32:096]:   Baud Rate: 9600 baud
[18:43:07][C][uart_esp32:097]:   Data Bits: 8
[18:43:07][C][uart_esp32:098]:   Parity: NONE
[18:43:07][C][uart_esp32:099]:   Stop bits: 1
[18:43:07][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'sensor'
[18:43:07][C][gpio.binary_sensor:016]:   Pin: GPIO5 (Mode: INPUT)
[18:43:07][C][logger:185]: Logger:
[18:43:07][C][logger:186]:   Level: VERBOSE
[18:43:07][C][logger:187]:   Log Baud Rate: 0
[18:43:07][C][logger:188]:   Hardware UART: UART0
[18:43:07][C][logger:190]:   Level for 'i2c': DEBUG
[18:43:07][C][logger:190]:   Level for 'uart': DEBUG
[18:43:07][C][sim800l:259]: SIM800L:
[18:43:07][C][sim800l:260]:   RSSI: 0 dB
[18:43:07][C][captive_portal:169]: Captive Portal:
[18:43:08][C][ota:029]: Over-The-Air Updates:
[18:43:08][C][ota:030]:   Address: esp32test.local:3232
[18:43:08][C][ota:032]:   Using Password.
[18:43:08][C][api:095]: API Server:
[18:43:08][C][api:096]:   Address: esp32test.local:6053
[18:43:09][V][sim800l:041]: S: AT - 1
[18:43:24][V][sim800l:041]: S: AT - 1
[18:43:39][V][sim800l:041]: S: AT - 1
```

baerni, 2021-04-08:
I connected RX to TX and TX to RX and it started working. I'm sorry for the trouble. I thought I should connect RX to RX and TX to TX.

```yaml
[18:49:50][C][api:096]:   Address: esp32test.local:6053
[18:49:54][V][sim800l:041]: S: AT - 1
[18:49:54][V][sim800l:048]: R: AT - 2
[18:49:55][V][sim800l:041]: S: ATE0 - 18
[18:49:55][V][sim800l:048]: R: OK - 2
[18:49:55][V][sim800l:041]: S: AT+CMGF=1 - 2
[18:49:55][V][sim800l:048]: R: ATE0AT+CMGF=1 - 3
[18:49:55][W][sim800l:062]: Not ack. 3 ATE0AT+CMGF=1
[18:49:55][V][sim800l:048]: R: OK - 0
[18:49:55][D][sim800l:216]: Unhandled: OK - 0
[18:49:55][V][sim800l:048]: R:  - 0
[18:49:55][V][sim800l:048]: R: OK - 0
[18:49:55][D][sim800l:216]: Unhandled: OK - 0
[18:50:09][V][sim800l:041]: S: AT - 1
[18:50:09][V][sim800l:048]: R:  - 2
[18:50:09][V][sim800l:048]: R: OK - 2
[18:50:09][V][sim800l:041]: S: AT+CMGF=1 - 2
[18:50:09][V][sim800l:048]: R:  - 3
[18:50:10][V][sim800l:048]: R: OK - 3
[18:50:10][V][sim800l:041]: S: AT+CREG? - 3
[18:50:10][V][sim800l:048]: R:  - 4
[18:50:10][V][sim800l:048]: R: +CREG: 0,1 - 4
[18:50:10][D][sim800l:103]: Registered OK
[18:50:10][V][sim800l:048]: R:  - 5
[18:50:10][V][sim800l:048]: R: OK - 5
[18:50:10][V][sim800l:041]: S: AT+CSQ - 5
[18:50:10][V][sim800l:048]: R:  - 6
[18:50:10][V][sim800l:048]: R: +CSQ: 17,0 - 6
[18:50:10][D][sim800l:129]: RSSI: 17
[18:50:10][V][sim800l:048]: R:  - 11
[18:50:10][V][sim800l:048]: R: OK - 11
[18:50:10][V][sim800l:041]: S: AT+CMGL="ALL" - 11
[18:50:10][V][sim800l:048]: R:  - 12
[18:50:10][V][sim800l:048]: R: +CMGL: 1,"REC UNREAD","+ XXXXXXXXXXX","","21/04/07,18:05:53+08" - 12
[18:50:10][V][sim800l:048]: R: abc - 13
[18:50:10][V][sim800l:048]: R:  - 13
[18:50:10][V][sim800l:048]: R: +CMGL: 2,"REC UNREAD","+ XXXXXXXXXXX","","21/04/07,17:54:10+08" - 13
[18:50:10][V][sim800l:048]: R: Aaaaa - 14
[18:50:10][D][sim800l:174]: Received SMS from: +XXXXXXXXXXX
[18:50:10][D][sim800l:175]: Aaaaa

```

glmnet, 2021-04-08:
Awesome! Closing this now

glmnet, 2021-04-08:
Hmm, may be a warning that comms not working can be useful here

closed 2021-04-08 by glmnet