esphome/issues #1971: Multiple addressable RGB strips all output to the pin of the last RGB strip.

Akisame-AI, 2021-04-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32: nodemcu-32S

**ESPHome version 1.16.2**

**Affected component:** addressable RGB strips using NeoPixelBus@2.6.0

**Description of problem:**
When using 2 addressable RGB strips the setup will use only the pin of the last RGB strip.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: livingroom
  platform: ESP32
  board: nodemcu-32s
  # a workaround for the latest bug in the neopixel lib bug
  platformio_options:
    lib_deps: NeoPixelBus@2.6.0
    
wifi:
  ssid: !secret wifi
  password: !secret wifi_pass
  manual_ip:
    static_ip: 192.168.2.144
    gateway: 192.168.2.254
    subnet: 255.255.255.0
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Livingroom Fallback Hotspot"
    password: "6txO2H1U1coK"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_pass

ota:
  password: !secret ota_pass

light:
  - platform: neopixelbus
    variant: WS2812X
    type: brg
    pin: 18
    num_leds: 100
    name: "Desk RGB light"
    effects:
      - addressable_scan:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_color_wipe:
          name: Toilet wipe
          colors:
            - red: 0%
              green: 20%
              blue: 0%
              num_leds: 5
            - red: 20%
              green: 0%
              blue: 0%
              num_leds: 30
          add_led_interval: 100ms
          reverse: True
  - platform: neopixelbus
    variant: WS2812X
    type: brg
    pin: 19
    num_leds: 100
    name: "Livingroom RGB light"
    effects:
      - addressable_scan:
      - addressable_rainbow:
      - addressable_color_wipe:
      - addressable_color_wipe:
          name: Toilet wipe
          colors:
            - red: 0%
              green: 20%
              blue: 0%
              num_leds: 5
            - red: 20%
              green: 0%
              blue: 0%
              num_leds: 30
          add_led_interval: 100ms
          reverse: True

```

**Logs (if applicable):**

```
[14:37:00][D][api.connection:604]: Client 'Home Assistant 2021.3.4 (192.168.2.111)' connected successfully!             [14:37:03][D][light:265]: 'Livingroom RGB light' Setting:                                                               [14:37:03][D][light:274]:   State: OFF                                                                                  [14:37:03][D][light:304]:   Transition Length: 1.0s                                                                     [14:37:05][D][light:265]: 'Desk RGB light' Setting:                                                                     [14:37:05][D][light:274]:   State: OFF                                                                                  [14:37:05][D][light:304]:   Transition Length: 1.0s                                                                     [14:37:07][D][light:265]: 'Desk RGB light' Setting:                                                                     [14:37:07][D][light:274]:   State: ON                                                                                   [14:37:07][D][light:278]:   Brightness: 58%                                                                             [14:37:07][D][light:304]:   Transition Length: 1.0s                                                                     [14:37:09][D][light:265]: 'Desk RGB light' Setting:                                                                     [14:37:09][D][light:278]:   Brightness: 22%                                                                             [14:37:09][D][light:304]:   Transition Length: 1.0s                                                                     [14:37:10][D][light:265]: 'Desk RGB light' Setting:                                                                     [14:37:10][D][light:274]:   State: OFF                                                                                  [14:37:10][D][light:304]:   Transition Length: 1.0s                                                                     [14:39:27][D][light:265]: 'Livingroom RGB light' Setting:                                                               [14:39:27][D][light:274]:   State: ON                                                                                   [14:39:27][D][light:304]:   Transition Length: 1.0s                                                                     [14:39:29][D][light:265]: 'Livingroom RGB light' Setting:                                                               [14:39:29][D][light:274]:   State: OFF                                                                                  [14:39:29][D][light:304]:   Transition Length: 1.0s                                                                     [14:39:30][D][light:265]: 'Desk RGB light' Setting:                                                                     [14:39:30][D][light:274]:   State: ON                                                                                   [14:39:30][D][light:304]:   Transition Length: 1.0s                                                                     [14:39:33][D][light:265]: 'Desk RGB light' Setting:                                                                     [14:39:33][D][light:278]:   Brightness: 55%                                                                             [14:39:33][D][light:304]:   Transition Length: 1.0s                                                                     [14:39:34][D][light:265]: 'Desk RGB light' Setting:                                                                     [14:39:34][D][light:274]:   State: OFF                                                                                  [14:39:34][D][light:304]:   Transition Length: 1.0s     

```

**Additional information and things you've tried:**
I currently have 2 rgb strips hooked up to 1 esp32 on pin 18 and 19. After compiling and flashing when I change the settings for either strip it will output both to the last pin in the config. 
So if I define the light that uses pin 18 first and then the light that uses pin 19 it will only control the rgb strip connected to pin 19 when adjusting either light.

OttoWinter, 2021-04-10:
Correct, when using the i2s methods from neopixelbus you only have 2 channels (that's a hardware limitation).

It would be better if esphome would warn you about that directly when validating, but unfortunately the validation system doesn't support validation across units.

Akisame-AI, 2021-04-11:
Aah okay. I see. Thanks. Maybe a short warning on the addressable lights page of esphome would be in order?

jamien82, 2021-04-18:
Have you had a look through this component. I have only split a strip, but it can join strips.
https://esphome.io/components/light/partition.html

closed 2021-08-30 by stale[bot]