esphome/issues #1976: ac_dimmer / dallas interrupt issue

fraxinas, 2021-04-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
`homeassistant/home-assistant:stable` Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32-WROOM32D custom board

**ESPHome version (latest production, beta, dev branch)**
`esphome 1.16.2`

**Affected component:**
`ac_dimmer` in conjunction with `dallas`

**Description of problem:**
When polling a dallas sensor, the ac_dimmer output stays `high` for a couple of mains cycles, therefore the light flashes for 15 ms for each dallas device read.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: geckocontrol
  platform: ESP32
  board: nodemcu-32s

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: 16
    min_power: 15%
    zero_cross_pin:
      number: 17
      mode: INPUT
      inverted: no

light:
  - platform: monochromatic
    output: dimmer1
    name: "Dimmer Daylight"
    gamma_correct: 1.0
    default_transition_length: 0s

dallas:
  - pin: 22

  - platform: dallas
    address: 0x0305473B405A0028
    name: "Temperature 1"
    unit_of_measurement: "°C"

  - platform: dallas
    address: 0x5505473B82670028
    name: "Temperature 2"
    unit_of_measurement: "°C"
```

**Logs (if applicable):**
Whenever the Dallas sensors are polled (in my case everyy 60s) as indicated by the log, the light flashes and the triac output reads as following:
[image]
[image]
it's visible that there are 2 moments with missing triac output around each poll of the dallas sensor, taking 760ms in total, exactly the time which a dallas poll takes to complete.

**Additional information and things you've tried:**
We found out, that the dallas sensor is read blockingly and therefore a few interrupts for ac_dimmer are missed because dallas integration disables the interrupts.

**Possible solution:**
unblock/re-enable interrupts during waiting time in the dallas polling routine.

fraxinas, 2021-04-12:
btw. this is my hardware: https://github.com/fraxinas/Geckocontroller/blob/main/hardware/Geckocontroller.pdf

and the complete esphome config yaml: https://github.com/fraxinas/Geckocontroller/blob/main/software/gecko.yaml

this issue might also be interesting for @OttoWinter

glmnet, 2021-04-12:
Can you add the GPIO22 to that logic trace? I'm not familiar with dallas, so that might help.

Also, in the logic is clear the gpio interrupt got delayed, should have raised before, then the timer interrupt (which falls the gate signal) either did not trigger or calculations went wrong due to gpio interrupt got delayed.

However trying to get those two working together is not gonna be easy

fraxinas, 2021-04-13:
> Can you add the GPIO22 to that logic trace? I'm not familiar with dallas, so that might help.

[image]

glmnet, 2021-04-13:
Interesting, dallas seems pretty busy there.
may be ac_dimmer can do better, if Dallas is done with the InterruptLock when the IO ends, looks like the timer interrupt was delayed for a bit more than a mains cycle, then a series of calculations goes wrong.

May be the artifact will be less noticeable

LeoDJ, 2021-08-11:
The issue still persists.

fraxinas, 2021-12-30:
Still a problem

Th3M1k3y, 2022-01-23:
I wonder if this was a part of the problem I had too.

After adding the ac_dimmer component, my dallas sensor kept giving me errors.

fraxinas, 2022-02-05:
it's the other way around for me: my light flickers when reading out the sensors. my current "workaround" is to poll the sensors only once every 5 minutes which is quite unsatisfying

OttoWinter, 2022-02-19:
tbh there's not much we can do here. Dallas needs to have interrupts disabled during the reading of the sensors, otherwise any interrupt that fires during it will mess up the timing and you get read errors (and because the reading takes ~11ms, that's almost guaranteed to happen).

And for `ac_dimmer` we really need to use the interrupts. One solution would be to move the `ac_dimmer` timer interrupt to the other core in the case of esp32 dual core, but I don't have the time to research/implement that.

alnavasa, 2022-06-15:
I am having this issue too, How could I disable dallas sensor while lights are on?

elgatho, 2022-09-26:
yep still an issue, maybe there is some sensor alternatives?

alnavasa, 2022-09-27:
I fitted an extra esp8266 just for the dimmer, not the best solution but a quick and easy one for my case scenario

jehy, 2022-12-13:
Still an issue. Dallas sensors stop working after adding ac_dimmer component to yaml.

By the way, that's really fun that I also use dallas and AC dimmer for my gecko, same as @fraxinas :)

jehy, 2022-12-13:
As a workaround, i used [this library](https://github.com/nrandell/dallasng) which is a drop in replacement. And now my sensors work flawlessly even with dimmer enabled and working. May be ESPHome should adopt it?