esphome/issues #1978: Pulse Width Sesnsor never returns zero

koeberlue, 2021-04-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**ESPHome version (latest production, beta, dev branch)**

v1.16.2
**Affected component:**

https://esphome.io/components/sensor/pulse_width.html

**Description of problem:**

I have connected a [5V PWM Fan](https://noctua.at/de/products/fan/nf-f12-5v-pwm) and I'm trying to read out its current current speed via PWM. This works perfectly fine while it is turning. But when it is stopped, I never receive a pulse width of zero. It seems like i always get the last measured value.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: pulse_width
    pin: D0
    name: Pulse Width Sensor
    filters:
      - lambda: return 60 / (4*x);
```

**Logs (if applicable):**

```
(Will paste logs when I'm at home)
```

**Additional information and things you've tried:**

I looked into [`pulse_width.cpp`](https://esphome.io/api/pulse__width_8cpp_source.html) and it seems like `arg->last_width_` never gets reset to zero. Is this the expected behavior?

RoganDawes, 2021-04-14:
Thinking about it, would it ever make sense to return a pulse width of zero? What would a zero-width pulse even look like?
As I understand it, when the fan has stopped, the pulse has effectively infinite width, until the fan starts again, and the rotation sensor triggers the next pulse.
So, IMO, detecting that the fan has stopped moving is more about time since the last pulse being reported, than it would be about the actual value reported by the sensor.

koeberlue, 2021-04-15:
> As I understand it, when the fan has stopped, the pulse has effectively infinite width, until the fan starts again, and the rotation sensor triggers the next pulse.

Oh, yes. You are absolutely right. Turns out I actually need the [Duty Cycle Sensor](https://esphome.io/components/sensor/duty_cycle.html), that I kind of replicated with my lamdba above.

What probably confused me was the `updateInterval` which is used for calculations when using the duty cycle, but does not have any effect on the measured output of the pulse width sensor. But following your explanation this behavior absolutely makes sense.

Thank you very much!

closed 2021-04-15 by koeberlue