esphome/issues #1981: MQTT Error resolving MQTT broker IP address: -6

frspp, 2021-04-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

Nodemcu ESP32

**ESPHome version (latest production, beta, dev branch)**

1.16.2.

**Affected component:**

MQTT https://esphome.io/components/mqtt.html

**Description of problem:**
Logger reads: 
"Couldn't resolve IP address for 'http://mqtt.thingspeak.com:1883'!
Error resolving MQTT broker IP address: -6"

What is error -6? Why I cannot connect?

**Problem-relevant YAML-configuration entries:**
```yaml
# MQTT connection
mqtt:
  broker: http://mqtt.thingspeak.com:1883 
  username: whatever
  password: PASSWORDREMOVED

```

**Logs (if applicable):**

```
[16:47:09][W][mqtt:127]: Couldn't resolve IP address for 'http://mqtt.thingspeak.com:1883'!
[16:47:09][W][mqtt:112]: Error resolving MQTT broker IP address: -6

```

**Additional information and things you've tried:**
Tried without http://. Tried without port.

frspp, 2021-04-13:
Aha, I see in the logs "dns: 0.0.0.0.0". 

Tried settings dns1 manually. Now I did OTA flash and still get "ERROR Cannot connect to MQTT broker: [Errno 8] nodename nor servname provided, or not known"

frspp, 2021-04-13:
updated the OP

koenvervloesem, 2021-04-13:
When using DHCP, your DHCP server sends your client the IP addresses of DNS servers. You have set up your IP address manually, so you need to [add the DNS servers manually](https://esphome.io/components/wifi.html#configuration-variables):

```yaml
wifi:
  ssid: "CUT"
  password: "CUT"
  fast_connect: true
  manual_ip:
    static_ip: 192.168.8.42
    subnet: 255.255.255.0
    gateway: 192.168.8.1
    dns1: YOURDNS1
    dns2: YOURDNS2
```

The `http://` before the domain of the MQTT broker is wrong too: it just needs a hostname or IP address, no protocol (which isn't http anyway).

frspp, 2021-04-13:
@koenvervloesem thanks! I have another device (Shelly) for what I also configured a static IP but it did not ask for DNS and works fine so I supposed that's the way how it works...

closed 2021-05-13 by ssieb