esphome/issues #1983: Compilation project problem !!!

lluis4l, 2021-04-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I'm working wih hass.io ( RPI4 4gb ram, and ssd ) latest version
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266(NODEMCU)
**ESPHome version (latest production, beta, dev branch)**

1.16.2
**Affected component:**

https://github.com/esphome/esphome/blob/dev/platformio.ini

**Description of problem:**

I’ve been working with esphome in several projects but now I’ve tried to compile a new nodemcu basic project but with only compiling It gets stuck compiling… and HA restarts.

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: level
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "MiFibra-0B4A"
  password: "xxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Level Fallback Hotspot"
    password: "aPm9IaOFCFXM"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
```

**Logs (if applicable):**

```
> INFO Reading configuration /config/esphome/lev.yaml...
> INFO Generating C++ source...
> INFO Core config or version changed, cleaning build files...
> INFO Compiling app...
> INFO Running:  platformio run -d /config/esphome/lev
> Processing lev (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266@2.6.2)
> --------------------------------------------------------------------------------
> HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
> PACKAGES: 
>  - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
>  - tool-esptool 1.413.0 (4.13) 
>  - tool-esptoolpy 1.20800.0 (2.8.0) 
>  - toolchain-xtensa 2.40802.200502 (4.8.2)
> Library Manager: Installing ESPAsyncTCP-esphome @ 1.2.3
> Library Manager: ESPAsyncTCP-esphome @ 1.2.3 has been installed!
> Library Manager: Installing ESPAsyncWebServer-esphome @ 1.2.7
> Library Manager: ESPAsyncWebServer-esphome @ 1.2.7 has been installed!
> Library Manager: Installing dependencies...
> Library Manager: Installing AsyncTCP-esphome
> Library Manager: AsyncTCP-esphome @ 1.2.1 has been installed!
> Library Manager: Installing Hash
> Library Manager: Already installed, built-in library
> Library Manager: Installing Update
> Library Manager: Already installed, built-in library
> Dependency Graph
> |-- <ESPAsyncTCP-esphome> 1.2.3
> |   |-- <ESP8266WiFi> 1.0
> |-- <ESP8266WiFi> 1.0
> |-- <ESP8266mDNS> 1.2
> |   |-- <ESP8266WiFi> 1.0
> |-- <ESPAsyncWebServer-esphome> 1.2.7
> |   |-- <ESPAsyncTCP-esphome> 1.2.3
> |   |   |-- <ESP8266WiFi> 1.0
> |   |-- <Hash> 1.0
> |   |-- <ESP8266WiFi> 1.0
> |-- <DNSServer> 1.1.1
> |   |-- <ESP8266WiFi> 1.0
> Compiling /data/lev/.pioenvs/lev/src/esphome/components/api/api_connection.cpp.o
> Compiling /data/lev/.pioenvs/lev/src/esphome/components/api/api_pb2.cpp.o
> Compiling /data/lev/.pioenvs/lev/src/esphome/components/api/api_pb2_service.cpp.o
> Compiling /data/lev/.pioenvs/lev/src/esphome/components/api/api_server.cpp.o

```

**Additional information and things you've tried:**

slyons, 2021-04-17:
I'm getting the exact same issue. I'm getting kernel warnings that tasks have been running for too long and eventually HA just stops responding.

randybb, 2021-04-17:
Compilation is CPU and memory consuming. If your computer is already overloaded and/or has not enough ram, it will just kill this job. I guess this is your problem...

closed 2021-08-29 by stale[bot]