esphome/issues #1987: ESP32 / SCD30 / problem with "temperature_offset"

LeDistordu, 2021-04-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker / Homeassistant

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 NodeMCU

**ESPHome version (latest production, beta, dev branch)**

ESPHome version v1.16.2

**Affected component:**

ESP32

**Description of problem:**

i follow this documentation
https://esphome.io/components/sensor/scd30.html

I have ESP32 NodeMCU and sensor SCD30

When i add "temperature_offset" like documentation, i have an error : [scd30:089]: Measurement Initialization failed!

```
- platform: scd30
    automatic_self_calibration: true
    co2:
      name: "co2"
      accuracy_decimals: 1
    temperature:
      name: "temperature"
      accuracy_decimals: 2
    humidity:
      name: "humidity"
      accuracy_decimals: 1
#     temperature_offset: 1.5 °C
    update_interval: 300s
```

Without it's ok !

I try : 

```
temperature_offset: 1.5 °C
temperature_offset: "1.5 °C"
temperature_offset: '1.5 °C'
temperature_offset: 1.5
```

comatose-tortoise, 2021-04-26:
Noticing this too.

agners, 2021-05-24:
This is likely related to clock stretching. The SCD30 component sets clocks stretching to 150ms for ESP8266 devices:
https://github.com/esphome/esphome/blob/v1.18.0/esphome/components/scd30/scd30.cpp#L27

This is also what the "Interface Description Sensirion SCD30Sensor Module" says:

> Clock stretching period in write-and read-frames is 30ms, however, due to internal calibration processes a maximal clock stretching of 150ms may occur once per day. 

However, the ESP32 only supports clock stretching up to 13ms, unfortunately:
https://github.com/espressif/esp-idf/issues/4173

closed 2021-10-06 by stale[bot]