esphome/issues #1989: ssd1327_i2c appears to break wifi

tvwerkhoven, 2021-04-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip3 esphome 1.16.2 / esptool 2.8

**ESP (ESP32/ESP8266, Board/Sonoff):** 

Wemos D1 mini pro (ESP8266)

**ESPHome version (latest production, beta, dev branch)**

esphome 1.16.2 / esptool 2.8
**Affected component:**

https://esphome.io/components/display/ssd1327.html

**Description of problem:**
When I enable the OLED in the YAML file and upload it, the board does not connect to wifi anymore (no ping, no MQTT, no HA). It appears the rest is working (i.e. sensor values can be displayed on the OLED).

**Problem-relevant YAML-configuration entries:**

This works:
```yaml
esphome:
  name: esp_mobile
  platform: ESP8266
  board: d1_mini_pro

wifi:
  networks:
  - ssid: "SSID1" # not IoT because this one needs WAN access for MQTT
    password: "password"
    hidden: true
  - ssid: "SSID2"
    password: "password"
  - ssid: "SSID3"
    password: "password"
  - ssid: "SSID4"
    password: "password"
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esphome.io"
    password: "password"

  use_address: "IPv4"
  #fast_connect: True # Required to connect to hidden SSIDs -- only works with one network
  domain: ".lan"
  # Do not reboot if no wifi
  reboot_timeout: 0s

logger:
  # Disable bc we use serial port for fine dust sensor already
  baud_rate: 0

# Enable Home Assistant API - disable for MQTT? And for reboot loop?
api:
  reboot_timeout: 0s

ota:

mqtt:
  broker: IPV4
  port: 1883
  username: username
  password: password
  id: mqtt_client

i2c:
  sda: GPIO4 # = D2 = 4
  scl: GPIO5 # = D1 = 5
  scan: True

```

When I append the following the wifi breaks:

```yaml
font:
  - file: "slkscr.ttf"
    id: my_font1
    size: 8

display:
  - platform: ssd1327_i2c
    model: "SSD1327 128x128"
    # reset_pin: D0
    address: 0x3C
    # id: oled_128x128
    lambda: |-
      it.print(0, 0, id(my_font1), "Hello World!");
```

**Logs (if applicable):**

```
Not available.

```

**Additional information and things you've tried:**
There are multiple sensors connected to the board (1 Dallas, 1 bme280, 1 mh-z19b, 1 SDS011), but these all work fine as long as the oled is disabled (and possible even with the oled enabled). The wifi only breaks when I enable the oled.

tvwerkhoven, 2021-04-18:
I found a partial mitigation. If I simplify the wifi section to:

```yaml
wifi:
  networks:
  - ssid: "SSID1"
    password: "password"
    hidden: true

  fast_connect: True # Required to connect to hidden SSIDs
  domain: ".lan"
```

the board does connect, but I get a I2C failure:
```
[09:34:36][C][i2c:028]: I2C Bus:
[09:34:36][C][i2c:029]:   SDA Pin: GPIO4
[09:34:36][C][i2c:030]:   SCL Pin: GPIO5
[09:34:36][C][i2c:031]:   Frequency: 50000 Hz
[09:34:36][I][i2c:033]: Scanning i2c bus for active devices...
[09:34:36][I][i2c:043]: Unknown error at address 0x01
[...]
[09:34:37][I][i2c:043]: Unknown error at address 0x77
[09:34:37][I][i2c:049]: Found no i2c devices!
[09:34:37][C][logger:188]:   Hardware UART: UART0
[09:34:37][C][ssd1327_i2c:023]: I2C SSD1327
[09:34:37][C][ssd1327_i2c:023]:   Rotations: 0 °
[09:34:37][C][ssd1327_i2c:023]:   Dimensions: 128px x 128px
[09:34:37][C][ssd1327_i2c:024]:   Address: 0x3C
[09:34:37][C][ssd1327_i2c:025]:   Model: SSD1327 128x128
[09:34:37][C][ssd1327_i2c:027]:   Update Interval: 1.0s
[09:34:37][E][ssd1327_i2c:030]: Communication with SSD1327 failed!
```

mmakaay, 2021-04-19:
Does the problem occur when you omit the `display` section, but keep the `font` section?

mmakaay, 2021-04-19:
It's a bit of a long shot, but I recently ran into quite an obscure issue when using font/display, causing the ESP8266 to crash with reboots: what version of the pillow python library do you use? Could you try upgrading pillow (using pip3)?

tvwerkhoven, 2021-04-19:
I [found a solution](https://www.gitmemory.com/issue/esphome/issues/1920/803429566): increase the update_interval to e.g. 10s.

Problem (from source): "The display used the I2C protokoll. Maximum rate for the Wemos D1 seem to bee 100khz, which seem to be too slow to keep the default display refresh rate at 5s! As a result, it looks like this breaks the hard requirement of the ESPHome "scheduler" to not spend more than 10ms. In the end, the other components don't get a chance to handle their loops."

@mmakaay for the record (I don't think this was it), I have Pillow version 8.1.0.

mmakaay, 2021-04-19:
For the record: you are right. That version of pillow has no issue.

closed 2021-05-25 by tvwerkhoven