esphome/issues #1992: slow_pwm doesn't work as intended (it is always on)

qqgg231, 2021-04-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 and ESP8266

**ESPHome version (latest production, beta, dev branch)**

1.16.2
**Affected component:**

https://esphome.io/components/output/slow_pwm.html

**Description of problem:**

https://community.home-assistant.io/t/how-to-use-slow-pwm-and-define-the-duty-cycle/287822

**Problem-relevant YAML-configuration entries:**
```yaml
in link above

```

**Logs (if applicable):**

```
-

```

**Additional information and things you've tried:**

ssieb, 2021-05-13:
It looks like you found the answer.

closed 2021-05-13 by ssieb

qqgg231, 2021-05-13:
@ssieb 
yes, but I think the way slow_pwm works is unintuitive

ssieb, 2021-05-13:
In what way?  It works exactly the same as all the other pwm outputs.