esphome/issues #1993: homeassistant.service - light.turn_on sending "rgb_color" fails - does not accept "list"?

tubalainen, 2021-04-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 / NodeMCU V2

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.16.1

**Affected component:**

**Description of problem:**
https://esphome.io/components/api.html#homeassistant-service-action

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: gpio
    pin:
      number: D3
      inverted: True
    name: "Displaylabb knapp 1"
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms
    on_press:
      then:
        - homeassistant.service:
            service: light.turn_on
            data:
              entity_id: light.lampa_ikea_rgb
              rgb_color: [0,255,255] # <<- Not allowed by the editor "must be a string", making it into a string will not be accepted by HA
```

**Logs (if applicable):**

```
N/A

```

**Additional information and things you've tried:**
https://www.home-assistant.io/integrations/light/
https://github.com/esphome/issues/issues/784
https://community.home-assistant.io/t/passing-list-to-service-call/266214

[image]

meijerwynand, 2021-04-20:
Greetings, 

I have not attempted this on one of my lights, however i suspect you may get around this issue by changing the structure of your RBG values. 

I have tested the following in HA -> Dev tools -> Services. Both gave the same result.

Sent to HA as a list (array)

[image]

Send to HA as separate values

[image]

Try and see if you get the same in esphome if you change it FROM
```yaml
...
rgb_color: [0,255,255]
...
```

and change it TO

```yaml
...
rgb_color: 
  - 0
  - 255
  - 255
...
```

Hope it helps

tubalainen, 2021-04-20:
@meijerwynand thanks for trying to help. However, the example you are trying to show is from a template and script solution, you found it here, right?: https://esphome.io/components/api.html#homeassistant-service-action

Sorry, it does not work from ESPHOME. I have tried it. Works great in HA / dev-tools.
[image]

meijerwynand, 2021-04-20:
> @meijerwynand thanks for trying to help. However, the example you are trying to show is from a template and script solution, you found it here, right?: [...]

I recalled the separate RBG values instead of a list from some I did a while back and thought it looked familiar. 

I just tried dummy code in my editor and got the same results as your original post indicated.
I then tried my suggested code and also received the same error.

I am using a different editor, but could not get the command `esphome my.yaml config` to run successfully

However, it does run not give any error when encapsulating it and then running `esphome my.yaml config` from my side and I was able to compile it.

[image]

You did highlight in the original post as a comment:

```yaml 
rgb_color: [0,255,255] # <<- Not allowed by the editor "must be a string", making it into a string will not be accepted by HA
```

Did you actually test that is the result you experience on HA if you "make it a string" in ESP home?

tubalainen, 2021-04-21:
@meijerwynand yep, I believe I have tested all possibilities. This bug is related to the fact that ESPHOME doesn't seems to accept lists according to the YAML standard.

One cannot just send random syntax to HA and expect HA to accept it :) e.i. sending a string when that is not expected.

Making it into a string makes ESPHOME´s editor "happy" and it also compiles.

Home Assistant gives this error when the API command is being sent.
```
2021-04-20 19:13:42 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/core.py", line 1434, in async_call
    processed_data = handler.schema(service_data)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 218, in __call__
    return self._exec((Schema(val) for val in self.validators), v)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 337, in _exec
    v = func(v)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 215, in _run
    return self._exec(self._compiled, value, path)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 341, in _exec
    raise e if self.msg is None else AllInvalid(self.msg, path=path)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/validators.py", line 339, in _exec
    v = func(path, v)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 817, in validate_callable
    return schema(data)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 272, in __call__
    return self._compiled([], data)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict
    return base_validate(path, iteritems(data), out)
  File "/usr/local/lib/python3.8/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping
    raise er.MultipleInvalid(errors)
voluptuous.error.MultipleInvalid: None for dictionary value @ data['rgb_color']

```
As stated in the comment in the org. post code above. `".... , making it into a string will not be accepted by HA"`.

Check out this post, it relates to the same issue/bug. 
https://community.home-assistant.io/t/passing-list-to-service-call/266214

My work-a-round for me at this stage is using `color_name` instead.
https://www.w3.org/TR/css-color-3/#svg-color

Thanks for trying to help - I do believe that this is a true bug and not my pea brain mistake. 😸

meijerwynand, 2021-04-21:
@tubalainen - Thanks for the feedback. I also had a look at the link you supplied again and I noticed it states 

> "...Data structures are not possible..."

[image]

It would appear to be more of a limitation than a bug, or a known (documented) issue causing the limitation... 

Glad to hear you did test the options and you are able to confirm although it is compiling correctly, HA is not happy with the input. 
It is also good to hear you did find a work around in the mean time. 

Cheers.

tubalainen, 2021-04-21:
> @tubalainen - Thanks for the feedback. I also had a look at the link you supplied again and I noticed it states
>
>
> [image]
>
> It would appear to be more of a limitation than a bug, or a known [...]

I read that as well - a bit vague - `color_name` would be a data structure as well, right? 
But yes, I guess you are 100% right on this one. "Lists" (what I call them) are "Data Structures" and not supported.

tubalainen, 2021-04-21:
Its not a bug, or a feature - its a short coming :)

closed 2021-04-21 by tubalainen