esphome/issues #2001: cover.template.publish Action, State only allows open or closed.

MaccaAus, 2021-04-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Supervised install on debian 10, ESPhome from add on store
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini
**ESPHome version (latest production, beta, dev branch)**

ESPhome 16.2
**Affected component:**

https://esphome.io/components/cover/template.html

**Description of problem:**
When using cover.template.publish Action ESPhome only allows open or closed in the state field.

Home Assistant allows the following states of a cover in the value template
open/true or opening/closing/closed/false 
as per 
https://www.home-assistant.io/integrations/cover.template/
It would be nice if ESPhome would allow the same values to be published.

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO12
      mode: INPUT_PULLUP
      inverted: false
    filters:
      - delayed_on: 1000ms  
      - delayed_off: 1000ms
    name: "Garage Closed Sensor"
    id: closed_endstop
    on_press:
      - logger.log: "CLOSED"
      - cover.template.publish:
          id: door_position_sensor 
          state: 'CLOSED'
    on_release:
      - logger.log: "CLOSING" 
      - cover.template.publish:
          id: door_position_sensor 
          state: "CLOSING"

  - platform: gpio
    pin:
      number: GPIO13
      mode: INPUT_PULLUP
      inverted: false
    filters:
      - delayed_on: 1000ms  
      - delayed_off: 1000ms      
    name: "Garage Open Sensor"
    id: open_endstop
    on_press:
      - logger.log: "OPEN"  
      - cover.template.publish:
          id: door_position_sensor 
          state: "OPEN"
    on_release:
      - logger.log: "OPENING" 
      - cover.template.publish:
          id: door_position_sensor 
          state: "OPENING"
     
cover:
  - platform: template
    name: "Door Position Sensor"
    id: door_position_sensor

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/garage_door_sensor.yaml...
Failed config

binary_sensor.gpio: [source /config/esphome/garage_door_sensor.yaml:33]
  platform: gpio
  pin: 
    number: GPIO12
    mode: INPUT_PULLUP
    inverted: False
  filters: 
    - delayed_on: 1000ms
    - delayed_off: 1000ms
  name: Garage Closed Sensor
  id: closed_endstop
  on_press: 
    - logger.log: CLOSED
    - cover.template.publish: 
        id: door_position_sensor
        state: CLOSED
  on_release:  [source /config/esphome/garage_door_sensor.yaml:49]
    - logger.log: CLOSING
    - [source /config/esphome/garage_door_sensor.yaml:50]
      cover.template.publish:  [source /config/esphome/garage_door_sensor.yaml:51]
        id: door_position_sensor
        
        Unknown value 'CLOSING', did you mean 'CLOSED'?.
        state: CLOSING [source /config/esphome/garage_door_sensor.yaml:52]
binary_sensor.gpio: [source /config/esphome/garage_door_sensor.yaml:54]
  platform: gpio
  pin: 
    number: GPIO13
    mode: INPUT_PULLUP
    inverted: False
  filters: 
    - delayed_on: 1000ms
    - delayed_off: 1000ms
  name: Garage Open Sensor
  id: open_endstop
  on_press: 
    - logger.log: OPEN
    - cover.template.publish: 
        id: door_position_sensor
        state: OPEN
  on_release:  [source /config/esphome/garage_door_sensor.yaml:70]
    - logger.log: OPENING
    - [source /config/esphome/garage_door_sensor.yaml:71]
      cover.template.publish:  [source /config/esphome/garage_door_sensor.yaml:72]
        id: door_position_sensor
        
        Unknown value 'OPENING', did you mean 'OPEN'?.
        state: OPENING [source /config/esphome/garage_door_sensor.yaml:73]

```

**Additional information and things you've tried:**

glmnet, 2021-04-26:
Please read ESPHome docs carefully

closed 2021-04-26 by glmnet

MaccaAus, 2021-04-27:
So this is how I got it to do what I wanted it to do in Home Assistant
`binary_sensor:
  - platform: gpio
    pin:
      number: GPIO12
      mode: INPUT
      inverted: true
    filters:
      - delayed_on: 1000ms  
      - delayed_off: 1000ms
    name: "Garage Closed Sensor"
    id: closed_endstop
    on_press:
      - logger.log: "CLOSED"
      - cover.template.publish:
          id: door_position_sensor 
          current_operation: IDLE
          state: 'CLOSED'
    on_release:
      - logger.log: "OPENING" 
      - cover.template.publish:
          id: door_position_sensor 
          current_operation: OPENING`

So IMO the doc's don't explain how both State and Current Operation relate back into the value template in Home Assistant for a cover.