esphome/issues #2004: Set Sleep Duration lambda service command not working

curt7000, 2021-04-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

PIP
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**ESPHome version (latest production, beta, dev branch)**

1.16.2
**Affected component:**

Deep Sleep

**Description of problem:**
I am trying to adjust the deep_sleep time via a MQTT payload. I have the code below, but the set_sleep_duration does not appear to be working. My ESP8266 will go to sleep but never wake. Note: I am ensuring the MQTT payload is OFF prior to re-awaking.

**Problem-relevant YAML-configuration entries:**
```yaml
`deep_sleep:
id: deep_sleep_1

mqtt:
broker: 10.0.1.10
username: homeassistant
password: my password
birth_message:
will_message:
on_message:
- topic: officecontroller/ota_mode
payload: 'ON'
then:
- deep_sleep.prevent: deep_sleep_1
- topic: officecontroller/sleep_mode_day
payload: 'ON'
then:
- logger.log: "Time to sleep ... setting sleep duration"
- lambda: |-
id(deep_sleep_1).set_sleep_duration(6000);
- logger.log: "Delaying sleep by 10s"
- delay: 10s
- logger.log: "Time to sleep !!! Hibernating"
- deep_sleep.enter: deep_sleep_1
- topic: officecontroller/sleep_mode_night
payload: 'ON'
then:
- logger.log: "Time to sleep ... setting sleep duration"
- lambda: |-
id(deep_sleep_1).set_sleep_duration(12000);
- logger.log: "Delaying sleep by 10s"
- delay: 10s
- logger.log: "Time to sleep !!! Hibernating"
- deep_sleep.enter: deep_sleep_1`

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

ssieb, 2021-05-13:
Have you ever had sleep working properly?  Do you have the necessary pins connected?  What does the log show?
Also, you might find it easier to use:
```
- deep_sleep.enter:
     id: deep_sleep_1
     sleep_duration: 6s
```

curt7000, 2021-05-13:
> Have you ever had sleep working properly?  Do you have the necessary pins connected?  What does the log show?
> 
> Also, you might find it easier to use:
> 
> ```
> 
> - deep_sleep.enter:
> 
>      id: deep_sleep_1
> 
>      sleep_duration: 6s
> 
> ```
> 
> 

Yes, DeepSleep works fine when I manually define it.

It is my understanding that it must be in milliseconds when using the set_sleep_duration command.  I tried it with units (s) and as a attribute, as you recommended, but it would not compile.

ssieb, 2021-05-13:
I was suggesting how to not use the `set_sleep_duration` command.  Show me what you tried.

curt7000, 2021-05-14:
I've tried setting the units white the lambda call

`id(deep_sleep_1).set_sleep_duration(600s);`
or
` id(deep_sleep_1).set_sleep_duration(10m);`

But it will not compile.

I've tried setting the using 

`mqtt:
    - topic: officecontroller/sleep_mode_day
      payload: 'ON'
      then:
        - deep_sleep.enter:
                  id: deep_sleep_1
                  sleep_duration: 10m `

But, the editor immediately flags the last line "sleep_duration ..." as an error and will not compile.

ssieb, 2021-05-14:
You can't use units in the lambda call, it's only milliseconds.
And I see that the `sleep_duration:` parameter was only added recently, so you need to use beta or dev for that.

curt7000, 2021-05-19:
> You can't use units in the lambda call, it's only milliseconds.
> 
> And I see that the `sleep_duration:` parameter was only added recently, so you need to use beta or dev for that.

Yea, unfortunately the lambda call didn't work when leaving just as a integer in milliseconds.

I noticed 17.2 is available.  Updated ESPHome, recompiled now successfully using the 'sleep_duration' parameter! But, tested after uploading, still doesn't work.

Are you saying that although the documentation includes the 'sleep_duration' parameter for 17.2, I need to run beta/DEV branch?

ssieb, 2021-05-19:
I don't know for sure which version had it first.  But if it compiles, then it's supported.  It still doesn't sleep for the right time?  What happens?

ssieb, 2021-05-19:
Btw, I see why the editor was flagging your `sleep_duration: 10m` lines.  "m" is not a valid unit.  "ms" or "min"

curt7000, 2021-05-19:
> I don't know for sure which version had it first.  But if it compiles, then it's supported.  It still doesn't sleep for the right time?  What happens?

I have two MQTT topics to listen for a Day Time Sleep and a Night Time Sleep.  During testing, I set 'sleep_duration' to 5min for Day and 10min for Night.  It successfully sees the MQTT payload and goes to sleep, but never comes back.  Only way to recover is power cycle.

curt7000, 2021-05-19:
> Btw, I see why the editor was flagging your `sleep_duration: 10m` lines.  "m" is not a valid unit.  "ms" or "min"

Yea, corrected that, but still doesn't wake up after the defined duration.

ssieb, 2021-05-19:
Ok, I just tested it myself and it works both with the lambda setting and with the sleep_duration parameter.  It goes to sleep and wakes up again at the right time.

closed 2021-05-29 by ssieb