esphome/issues #2005: YAML config with packages break MQTT implementation

marcvl64, 2021-04-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

yaml files compiled on Mac Pro with Big Sur 11.0.1
HA standalone implementation on Raspberry Pi (v2021.2.3)
Mosquitto server 2.05

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 running on Shelly RGBW2

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.16.2

**Affected component:**

MQTT

**Description of problem:**
When using packages to better organize YAML files, the resulting implementation does not create the proper MQTT Config messages, and hence the device is not picked up by Home Assistant.

When running the YAML file through `esphome config` FIRST and using the resulting output as input for `esphome run`, the issue goes away and the device is properly recognized by Home Assistant.

**Problem-relevant YAML-configuration entries:**
See `https://github.com/marcvl64/esphome`

`test_rgbw_o.yaml` is the file that has problems.
`shelly_rgbw.yaml` is the file with the same content, but eliminating packages and substitutions. This file works fine.

`esphome test_rgbw_o.yaml run` completes without errors but no mqtt config messages are sent out. So HA doesn't pick up the device.

However:
```
esphome test_rgbw_o.yaml config > output.yaml
esphome output.yaml run
```
works fine. mqtt config messages are sent out and the device  is picked up by home assistant.

**Logs (if applicable):**

There are no error messages in the logs (i.e. /debug mqtt topic). However, when comparing the two logs (one with and one without packages, the one that works (without packages) shows 4 additional log entries that the broken one doesn't show"
```
[10:52:42][C][mqtt.binary_sensor:018]: MQTT Binary Sensor 'Switch Shelly rgbw':
[10:52:43][C][mqtt.binary_sensor:019]:   State Topic: 'test-light/binary_sensor/switch_shelly_rgbw/state'
[10:52:43][C][mqtt.light:053]: MQTT Light 'shelly_rgbw2':
[10:52:43][C][mqtt.light:054]:   State Topic: 'test-light/light/shelly_rgbw2/state'
[10:52:43][C][mqtt.light:054]:   Command Topic: 'test-light/light/shelly_rgbw2/command'
```

**Additional information and things you've tried:**
Basically, what I found is that when you validate the yaml config through `esphome config` you get a working output, even if the starting point was the yaml with packages. I compared the output of the yaml with packages to the one without packages, and while both work, there was a difference in section ordering and an additional substitution section in the one that started with packages. I manually removed the substitutions section and changed the order of the sections to match, but regardless of what i tried, the output yaml always worked.

So, basically the issue remains. When starting from a yaml that has packages, the mqtt implementation fails (without errors). When starting from the same yaml, and going through the intermediary step of `esphome config` the issue goes away.

koenvervloesem, 2021-04-28:
This is a known issue. See https://github.com/esphome/issues/issues/1487 (which also shows another workaround). Interesting that you found the two-step process works, I hadn't thought of that.

mmakaay, 2021-04-28:
We tried all kinds of things on Discord, and that two step process was a fun one to find :-)
I did try to find an open issue for this, but clearly used the wrong search terms.
So we can consider this one a duplicate.

@marcvl64 : I think that means you can close this one, and do further followups / status checking via https://github.com/esphome/issues/issues/1487

marcvl64, 2021-04-28:
Ok, closing the issue. Duplicate of #1487. Thanks for the feedback

closed 2021-04-28 by marcvl64