esphome/issues #2010: cold_white no longer validates in light: turn_on code

sparkydave1981, 2021-04-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HAOS Add-on
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP01_1m
**ESPHome version (latest production, beta, dev branch)**

Working fine in 1.15.3, fail validation for 1.16.2
**Affected component:**

light: (but only in ESPhome automation)

**Description of problem:**
Code validation fails when checking against new version (1.16.2) due to the `cold_white` entry as an on-boot automation.  This codes works fine in version 1.15.3

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: moon_lamp
  platform: ESP8266
  board: esp01_1m
  on_boot:
    then:
      - light.turn_on:
          id: moon_lamp
          red: 0
          green: 0
          blue: 0
          cold_white: 1

```

**Logs (if applicable):**

```
[cold_white] is an invalid option for [light: turn_on]. Did you mean [white]?

```

**Additional information and things you've tried:**
The reason I am using cold_white is because this is defined under the light: section in the code.  As mentioned, this works currently, just does not pass validation for an upgrade.

Full device code:
```
esphome:
  name: moon_lamp
  platform: ESP8266
  board: esp01_1m
  on_boot:
    then:
      - light.turn_on:
          id: moon_lamp
          red: 0
          green: 0
          blue: 0
          cold_white: 1

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.0.98
    gateway: 192.168.0.1
    subnet: 255.255.255.0
  ap:
    ssid: "Moon Lamp Fallback Hotspot"
    password: !secret wifi_fallback_ap_password
    ap_timeout: 3min

captive_portal:

web_server:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

output:
  - platform: esp8266_pwm
    pin: 14
    id: output_blue
  - platform: esp8266_pwm
    pin: 12
    id: output_green
  - platform: esp8266_pwm
    pin: 4
    id: output_red
  - platform: esp8266_pwm
    pin: 5
    id: output_coldwhite
  - platform: esp8266_pwm
    pin: 13
    id: output_warmwhite

light:
  - platform: rgbww
    name: Moon Lamp
    id: moon_lamp
    default_transition_length: 0s
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_coldwhite
    warm_white: output_warmwhite
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    effects:
      - lambda:
          name: Slow Rainbow
          update_interval: 11s
          lambda: |-
            static int state = 0;
            auto call = id(moon_lamp).turn_on();
            call.set_white(0.0);
            call.set_transition_length(10000);
            if (state == 0) {
              call.set_rgb(1.0, 0.0, 0.0);
            } else if (state == 1) {
              call.set_rgb(1.0, 0.5, 0.0);
            } else if (state == 2) {
              call.set_rgb(1.0, 1.0, 0.0);
            } else if (state == 3) {
              call.set_rgb(0.5, 1.0, 0.0);
            } else if (state == 4) {
              call.set_rgb(0.0, 1.0, 0.0);
            } else if (state == 5) {
              call.set_rgb(0.0, 1.0, 0.5);
            } else if (state == 6) {
              call.set_rgb(0.0, 1.0, 1.0);
            } else if (state == 7) {
              call.set_rgb(0.0, 0.5, 1.0);
            } else if (state == 8) {
              call.set_rgb(0.0, 0.0, 1.0);
            } else if (state == 9) {
              call.set_rgb(0.5, 0.0, 1.0);
            } else if (state == 10) {
              call.set_rgb(1.0, 0.0, 1.0);
            } else if (state == 11) {
              call.set_rgb(1.0, 0.0, 0.5);
            }
            call.perform();
            state++;
            if (state == 12)
              state = 0;

time:
  - platform: homeassistant
    id: homeassistant_time
```

ssieb, 2021-05-13:
As far as I can tell from git history, `light.turn_on:` has never supported `cold_white:` as a parameter.  The action appears to have been added about 2 years ago in 1.13.

closed 2021-05-13 by ssieb

sparkydave1981, 2021-05-13:
So how is it then that this currently works?  ie: it WAS supported.  How else can I get teh light to turn on with these parameters in the newer version of ESPhome?

ssieb, 2021-05-13:
Show me a config that works and what version it works with.  One that has the `light.turn_on:` action with `cold_white:` as a parameter.

sparkydave1981, 2021-05-13:
> Show me a config that works and what version it works with. One that has the `light.turn_on:` action with `cold_white:` as a parameter.

It's posted above.  It's all there.

ssieb, 2021-05-13:
You must have changed something in your config.  I just tested it and it doesn't work.
```
$ esphome version
Version: 1.14.5
$ esphome moon.yaml compile
INFO Reading configuration moon.yaml...
Failed config

esphome: [source moon.yaml:2]
  name: moon
  platform: ESP8266
  board: esp01_1m
  on_boot:  [source moon.yaml:6]
    then: 
      - [source moon.yaml:7]
        light.turn_on:  [source moon.yaml:8]
          id: moon_lamp
          red: 0
          green: 0
          blue: 0
          
          [cold_white] is an invalid option for [light.turn_on]. Did you mean [white]?
          cold_white: 1 [source moon.yaml:12]
```
Any esphome versions before that just don't work.

sparkydave1981, 2021-05-14:
I definitely have not changed the code at all. It works for me with the code provided above

ssieb, 2021-05-14:
I tested it with 1.14.5 and it definitely did not work.  There has never been code in esphome that would let that work.

sparkydave1981, 2021-05-14:
I'm not sure what else to say, the code above IS what is running in that lamp. When I power cycle the lamp it turns on to cold white, irrespective of what colour it was previously, which is what that code is for.

ssieb, 2021-05-14:
What version of esphome did you compile it with?  Can you do it again?

glmnet, 2021-05-15:
You might be running another version of ESPHome, i.e. another branch

sparkydave1981, 2021-05-15:
Guys, as per the first post, I have used this code with version 1.15.3 and it works.  That is what is currently running on this light.  I have never changed the code since the first time I ever flashed it.  I've updated it OTA when new versions of ESPhome have been released but never modified the code.
As I ssaid above, I think this code SHOULD be allowed so we can turn a light on with the desired settings when using a RGBW / RGBWW / RGBCW setup.

ssieb, 2021-05-15:
```
$ esphome version
Version: 1.15.3
$ esphome moon.yaml compile
INFO Reading configuration moon.yaml...
Failed config

esphome: [source moon.yaml:2]
  name: moon
  platform: ESP8266
  board: esp01_1m
  on_boot:  [source moon.yaml:6]
    then:  [source moon.yaml:7]
      - [source moon.yaml:7]
        light.turn_on:  [source moon.yaml:8]
          id: moon_lamp
          red: 0
          green: 0
          blue: 0
          
          [cold_white] is an invalid option for [light.turn_on]. Did you mean [white]?
          cold_white: 1 [source moon.yaml:12]
```
You really can't have used this.  Some RGBWW lights don't even have separate cold and warm controls, so this is not likely to change.

sparkydave1981, 2021-05-15:
Mate, we keep going around in circles, but all I can say is that I coded that light, with that code about a year ago, have not changed it whatsoever and it works.  If I set the light to purple, power cycle it, it turns on to full cold white, as per the above code.  ie: it works as expected from that code.

In HA, we can use similar code to turn on a light to a desired set of parameters, so why not implement that functionality in ESPhome?

ssieb, 2021-05-15:
You can, but you have to use these parameters: https://esphome.io/components/light/index.html#light-turn-on-action

sparkydave1981, 2021-05-15:
which don't allow for controlling the cold white LED's rather than warm white LED's