esphome/issues #2013: Access-Control-Allow-Origin not in /events or /domain/entity

meijerwynand, 2021-05-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

(linux) pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

Lolin ESP8266
**ESPHome version (latest production, beta, dev branch)**

v1.16.2

**Affected component:**

https://esphome.io/web-api/index.html

**Description of problem:**
Greetings, 

I noticed the HTTP header `Access-Control-Allow-Origin` is only passed when accessing the root page of the web server component. It is not passed when accessing the "REST API" (`/domain/entity`) or the EventSource (`/events`)

**Problem-relevant YAML-configuration entries:**
```yaml
...
web_server:
  port: 80
...
```

**Logs (if applicable):**

#### Root level
Accessing the web server on "root" level, you will notice the  `Access-Control-Allow-Origin: *` header is present close to the bottom of the log below
```
$ curl -v http://192.168.13.48/
*   Trying 192.168.13.48:80...
* TCP_NODELAY set
* Connected to 192.168.13.48 (192.168.13.48) port 80 (#0)
> GET / HTTP/1.1
> Host: 192.168.13.48
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Length: 1458
< Content-Type: text/html
< Access-Control-Allow-Origin: *
< Connection: close
< Accept-Ranges: none
...
```

#### REST API
Accessing the web server on "REST API" level, you will notice the  `Access-Control-Allow-Origin: *` header is not present
```
$ curl -v http://192.168.13.48/text_sensor/esp_tester_system_version
*   Trying 192.168.13.48:80...
* TCP_NODELAY set
* Connected to 192.168.13.48 (192.168.13.48) port 80 (#0)
> GET /text_sensor/esp_tester_system_version HTTP/1.1
> Host: 192.168.13.48
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Length: 124
< Content-Type: text/json
< Connection: close
< Accept-Ranges: none
< 
* Closing connection 0
{"id":"text_sensor-esp_tester_system_version","state":"1.16.2 Apr 30 2021, 13:15:46","value":"1.16.2 Apr 30 2021, 13:15:46"}
```

#### EventSource
Accessing the web server on "EventSource" level, you will notice the  `Access-Control-Allow-Origin: *` header is not present
```
$ curl -v http://192.168.13.48/events                               
*   Trying 192.168.13.48:80...
* TCP_NODELAY set
* Connected to 192.168.13.48 (192.168.13.48) port 80 (#0)
> GET /events HTTP/1.1
> Host: 192.168.13.48
> User-Agent: curl/7.68.0
> Accept: */*
> 
* Mark bundle as not supporting multiuse
< HTTP/1.1 200 OK
< Content-Type: text/event-stream
< Cache-Control: no-cache
< Connection: keep-alive
< Accept-Ranges: none
* no chunk, no close, no size. Assume close to signal end
< 
retry: 30000
id: 13166352
event: ping
data: 
...
```

**Additional information and things you've tried:**
With no option to set the `Access-Control-Allow-Origin: *` or it being default on, it does complicate things to develop or interface with the information via JS due to CORS protection in the browser. 

With the root page `Access-Control-Allow-Origin: *` being CORS- compatible/available it feels like either the REST API and EventSource should also conform to this. If it is a security concern I would like to suggest adding the "CORS/Access-Control-Allow-Origin"-option in the `yaml` config, e.g.

```yaml
...
web_server:
  port: 80
  cors: '*'
...
```

Your thoughts on this is appreciated.

Thank you.

moroshko, 2021-06-12:
Any update on this?
Would be really good to be able to make cross-origin requests to `/events`.

meijerwynand, 2021-08-02:
Greetings, 

# The short:

I was revisiting this issue and did some investigation and did manage to get the `"Access-Control-Allow-Origin"` working.

However the change I found appears to be implemented and **all** HTTP pages, even the captive portal / access point mode as it is done on the `web_server_base.h` file.

The alternative to make changes to each and every `handle_<some>_request` in the `web_server.cpp` file does seem to be a bit too excessive and quickly gets messy to implement in the nested / multi if-statement domains like light, switch, fan  etc. 

# The long

## Code 
In the file `components/web_server_base/web_server_base.h` just before running the `.begin()` line, default headers can be set to `DefaultHeaders::Instance().addHeader("Access-Control-Allow-Origin", "*");` according to the [ESPAsyncWebServe](https://github.com/me-no-dev/ESPAsyncWebServer#adding-default-headers) documentation. 

[image]

## Results
This does allow for all the pages previously reported to have the cross-origin header to be passed

### Captive portal / Access point mode
[image]

### Events
[image]

### API 
[image]

## Considerations
I did not create a PR for this. As this proposed change forms part of a very important file `components/web_server_base/web_server_base.h`. Your guidance on the better approach within the ESPHome ecosystem is valued. I would still like to suggest a config (yaml) based option for your consideration. 

Looking forward hearing your thoughts on this.

Thanks

meijerwynand, 2022-04-10:
Greetings, 

Looking at the improvements made on  https://github.com/esphome/esphome/pull/2688 I would like to know if the https://github.com/esphome/issues/issues/2013 Access-Control-Allow-Origin could form part of this.

Thanks