esphome/issues #2017: binary_sensor stops checking state?

jordansissel, 2021-05-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU ESP-32S

**ESPHome version (latest production, beta, dev branch)**

1.16.2

**Affected component:**

https://esphome.io/components/binary_sensor/index.html?highlight=device_class#binary-sensor-filters

**Description of problem:**

When using a gpio binary_sensor, I am finding that sometimes my device stops registering state changes entirely for this sensor. I haven't found any correlated settings involved in this problem.

**Problem-relevant YAML-configuration entries:**
```yaml
output:
  - platform: gpio
    pin: GPIO2
    id: blue_led

binary_sensor:
  - platform: gpio
    name: "power"
    device_class: power # https://www.home-assistant.io/integrations/binary_sensor
    pin: 
      number: GPIO23
      mode: INPUT_PULLDOWN

    # Debounce any light flickering.
    filters:
    - delayed_on_off: 100ms

    on_press:
      then:
        - output.turn_on: blue_led
    on_release:
      then:
        - output.turn_off: blue_led

  - platform: gpio
    device_class: problem # https://www.home-assistant.io/integrations/binary_sensor
    name: "fault"
    pin: 
      number: GPIO22
      mode: INPUT_PULLDOWN
```

**Logs (if applicable):**

```
[22:01:45][D][binary_sensor:036]: 'power': Sending state OFF
[22:01:45][D][binary_sensor:036]: 'power': Sending state ON
[22:01:45][D][binary_sensor:036]: 'fault': Sending state ON
[22:01:45][D][binary_sensor:036]: 'fault': Sending state OFF

```

**Additional information and things you've tried:**

Overall, I observe that the log output "Sending state ..." for each sensor will stop appearing the logs for unknown reasons.

I wondered if this was some serial port delay, so I setup `on_press`/`on_release` to toggle the blue led on the nodemcu board to represent the state of one of binary_sensor entries.

I then had a hypothesis that `delayed_on_off` filter was caused, so I configured two phototransistors (via binary_sensor gpio), one with and without that filter. Same rough effect.

I've tested a few different hypothesis and still haven't come up with a solution.

Any ideas of what I can do next? Is this a bug? It's hard to tell if the esp32s is stuck/stalled or if the binary_sensor has stopped reporting?

jordansissel, 2021-05-03:
I built a similar circuit for ESP8266 (a d1 mini clone) to try and test this on a different board. So far, I haven't seen this same behavior --binary_sensor appears to be working correctly without stalling on a d1 mini as compared to the nodemcu esp32s.

dylan09, 2021-05-27:
I have a similar problem.

The following code is working with es-home 1.16.2.

`packages:
   wifi: !include wifi.yaml
   mqtt: !include mqtt.yaml
   time: !include time.yaml
   sensor: !include def_sensor.yaml
   text_sensor: !include def_text_sensor.yaml
 
 esphome:
   name: ${node_name}
   platform: ESP8266
   board: esp01_1m
 
 captive_portal:
 
 logger:
 
 api:
   password: !secret api_passwd
 
 ota:
   password: !secret ota_passwd
 
 binary_sensor:
   - platform: gpio
     name: ${verbose_name}_is_online
     id: ${node_name}_is_online
     pin: 
       number: GPIO0
       mode: INPUT_PULLUP
       inverted: true
     
 switch:
   - platform: gpio
     name: ${verbose_name}_relay_0
     id: ${node_name}_relay_0
     restore_mode: ALWAYS_OFF
     pin: GPIO3
     icon: "mdi:speaker-wireless"
     on_turn_on:
       - switch.turn_on: led
       - delay: 500ms
       - switch.turn_off: ${node_name}_relay_0
       - switch.turn_off: led
 
   - platform: gpio
     # Verwenden wir nur zur Kontrolle. Ist per 10k Pullup auf High
     pin: GPIO2
     id: led
     inverted: true
     internal: true
 `

Same code compiled with es-home 1.18.0 binary sensor `id: ${node_name}_is_online` isn't working. I get no on / off signals. Not via API nor via MQTT. Also the logs didn't show any reaction if the GPIO is set or unset.

closed 2021-10-09 by stale[bot]