esphome/issues #2019: Graphic on OLED, which blocks wifi adapter on ESP8266

JaroslawPrzybyl, 2021-05-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HASS.IO, core-2021.4.1

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini

**ESPHome version (latest production, beta, dev branch)**

1.16.2

**Affected component:**

OLED 0.96" SSD1306

**Description of problem:**
I found interesting kind of bug: particular graphic on 0.96" oled display, which causing loss of wifi connection. In my code here is a part, which displays something:

**Problem-relevant YAML-configuration entries:**
```yaml
display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    id: my_display
    # reset_pin: D0
    address: 0x3C
    brightness: 40%
    lambda: |-
      it.filled_rectangle(0, 0, 127, 11);
      it.print(0, 0, id(MSS_10), COLOR_OFF, "Nawadnianie");
      it.printf(70, 0, id(MSS_10), COLOR_OFF, "WIFI:%.0fdbm", id(wifi_level).state);
      it.line(0, 50, 127, 50);
      it.line(40, 12, 40, 50);
      it.line(83, 12, 83, 50);
      
      it.print(2, 14, id(MSS_10), "Pompa");
      if(id(zompa).state) {
       it.print(10, 25, id(MSS_10), "ON");
       it.print(2, 35, id(MSS_10), "00m45s");
      } else it.print(5,25, id(MSS_20), "Off");
      
      it.print(43, 14, id(MSS_10), "Zawor 1");
      if(id(zawor1).state) {
       it.print(53, 25, id(MSS_10), "ON");
       it.print(45, 35, id(MSS_10), "00m45s");
      } else it.print(49,25, id(MSS_20), "Off");
      
      it.print(87, 14, id(MSS_10), "Zawor 2");
      if(id(zawor2).state) {
       it.print(97, 25, id(MSS_10), "ON");
       it.print(89, 35, id(MSS_10), "00m45s");
      } else it.print(95,25, id(MSS_20), "Off");
      
      //it.filled_rectangle(0, 51, 127, 63);
      it.printf(0, 54, id(MSS_10), "%s", id(uptime_human).state.c_str());
```
In this form it works perfect, you can see wifi signal -66dBm

[image]

Then, I only remove comments from penultimate line (it.filled.rectangle), no other changes, ESP is still on the same place, but after reloading it can't connect to wifi at all. Display looks as below (here was also changed color of "uptime" text , but it doesn't matter, only rectangle causes problems, uptime value is not shown here, because photo was taken short time after start, time below 1min is not shown):

[image]

As you see, wifi not connected. Log (level increased to verbose) shows that wifi adapter can't connect to wifi and wifi adapter is restarted in a loop:
```
[10:36:44][W][wifi:485]: Error while connecting to network.
[10:36:44][W][wifi:522]: Restarting WiFi adapter...
[10:36:44][V][wifi_esp8266:046]: Disabling STA.
[10:36:44]del if0
[10:36:44]usl
[10:36:44]mode : null
[10:36:44][V][wifi_esp8266:492]: Event: Changed Mode old=STA new=OFF
[10:36:44]wifi evt: 8
[10:36:49][D][wifi:324]: Starting scan...
[10:36:49][V][wifi_esp8266:044]: Enabling STA.
[10:36:49]mode : sta(40:f5:20:29:92:2f)
[10:36:49]add if0
[10:36:49][V][wifi_esp8266:492]: Event: Changed Mode old=OFF new=STA
[10:36:49]wifi evt: 8
[10:36:56]scandone
[10:36:56][D][wifi:339]: Found networks:
[10:36:56][I][wifi:385]: - 'Automatyka' [redacted]▂▄▆█
[10:36:56][D][wifi:386]:     Channel: 11
[10:36:56][D][wifi:387]:     RSSI: -69 dB
[10:36:56][I][wifi:385]: - 'Automatyka' [redacted]▂▄▆█
[10:36:56][D][wifi:386]:     Channel: 6
[10:36:56][D][wifi:387]:     RSSI: -81 dB
[10:36:56][D][wifi:389]: - [redacted] [redacted]▂▄▆█
[10:36:56][D][wifi:389]: - [redacted] [redacted]▂▄▆█
[10:36:56][D][wifi:389]: - [redacted] [redacted]▂▄▆█
[10:36:56][D][wifi:389]: - [redacted] [redacted]▂▄▆█
[10:36:56][D][wifi:389]: - [redacted] [redacted]▂▄▆█
[10:36:56][D][wifi:389]: - [redacted] [redacted]▂▄▆█
[10:36:56][I][wifi:194]: WiFi Connecting to 'Automatyka'...
[10:36:56][V][wifi:196]: Connection Params:
[10:36:56][V][wifi:197]:   SSID: 'Automatyka'
[10:36:56][V][wifi:200]:   BSSID: EA:63:DA:34:85:7E
[10:36:56][V][wifi:220]:   Password: [redacted]
[10:36:56][V][wifi:225]:   Channel: 11
[10:36:56][V][wifi:235]:   Using DHCP IP
[10:36:56][V][wifi:237]:   Hidden: NO
[10:36:57]scandone
[10:36:57]state: 0 -> 2 (b0)
[10:36:59]state: 2 -> 0 (2)
[10:36:59][W][wifi_esp8266:444]: Event: Disconnected ssid='Automatyka' bssid=[redacted] reason='Auth Expired'
[10:36:59]wifi evt: 1
[10:36:59]STA disconnect: 2
[10:36:59][W][wifi:485]: Error while connecting to network.
[10:36:59][W][wifi:522]: Restarting WiFi adapter...
```

Other side, my wifi AP shows in its logs, that this device is connecting properly, but after short time connection is breaken:

[image]

I already verified hardware, changed ESP, display, tried to power it from other power supply (large, workshop power supply, for sure enough), tried to move ESP close to AP. On PCB I have ceramic capacitors close to ESP and other one close to OLED. 
One interesting fact, which may indicate, where is a problem source: I also tried to switch on device without connected OLED display. Of course it doesn't work without display, lack of it is indicated in log, and it stops on this phase, but wifi adapter is enabled before this step and it looks unambiguous:
- without rectangle wifi is stable connected, device get IP and I can ping it.
- with rectangle, wifi is not connected, IP is not getting from DHCP, on wifi AP I see the same connection and disconnection loop.

**Logs (if applicable):**

given in description

**Additional information and things you've tried:**

given in description

JaroslawPrzybyl, 2021-05-03:
Additional info: before creating this ticket, I described problem on any local ESPHome group in my country. Before moment somebody wrote to me, that he reproduced this problem on m5stack with connected oled display. Same issue: filled rectangle at the bottom of screen --> wifi is dead.

3ative, 2021-05-08:
For my situation, wi-fi fails to connect if I have **both** an SSD1306 128x64 (With icons/graphics displayed) **AND** a Dallas temperature sensor connected. 
_- It's fine when either one is used, just not both at the same time._

3ative, 2021-05-08:
Additional:
Having both an OLED and a Dallas sensor running together on Boot didn't cause this Wi-Fi issue previously.
 - This problem started after v1.17

closed 2021-09-20 by stale[bot]