esphome/issues #2020: Output.turn_on do not work on_boot with high priority

kses, 2021-05-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 (Sparkfun Thing and Olimex Lipo dev board)
**ESPHome version (latest production, beta, dev branch)**

1.16.2 (as stated by the dashboard right corner)
**Affected component:**

https://esphome.io/components/esphome.html#esphome-on-boot
https://esphome.io/components/output/gpio.html

**Description of problem:**
I for a battery powered project I wanted the esphome to only power the sensor I am using (MPU5060 connected via i2c) when it is awake.

However if it use the on_boot automation to turn the gpio on to power the sensor, it either powers on to late (default priority) and i2c fails to setup or not at all (high priority). The output.turn_on do not work with on_boot on high priority.

**Problem-relevant YAML-configuration entries:**
```yaml
  on_boot:
    priority: 900
    then:
     - output.turn_on: gpio_15
  on_shutdown:
    then:
     - output.turn_off: gpio_15    
     
output:
  - platform: gpio
    pin: 15
    id: gpio_15
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```
No warning/log message related to the GPIO not turning on, more than i2c failing and later the mpu6050 component.
Multimeter to verify that the GPIO pin did not turn on. Also verified by that the LED on the senor board did not turn on.

**Additional information and things you've tried:**
Workaround:
Using arduino c++ instead works with a high priority on_boot.

  ```
on_boot:
    priority: 1900
    then:
      - lambda: |-
          pinMode(15, OUTPUT);
          digitalWrite(15, HIGH);
          delay(100);
```

ssieb, 2021-09-02:
You can change the sensor setup priority so that it starts after the GPIO has been turned on.

g5pw, 2026-03-17:
Huh, I actually noticed the same thing! `output.turn_on` on `priority: 800` did not actually turn on the GPIO.

g5pw, 2026-03-20:
Oh well, that would be a solid reason 😅 maybe a lint could be made, or priorities documented in a way that makes it clear that setting outputs at 800 (or earlier) would not work?