esphome/issues #2022: pulse_meter seems to be doubling the amount of pulses counted

pedrolamas, 2021-05-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):** ESP8266

ESP8266
**ESPHome version (latest production, beta, dev branch)**

v1.17.0
**Affected component:**

https://esphome.io/components/sensor/pulse_meter.html

**Description of problem:**

For a while now I've been using the `pulse_counter` to count the light pulses on my power meter and it has worked very well in that!

I see that the latest version of ESPHome is recommending to use `pulse_meter` so I replaced it today and it seems to be counting double the amount of pulses than the `pulse_counter`

**Problem-relevant YAML-configuration entries:**

old yaml:

```yaml
sensor:
  - platform: pulse_counter
    id: pulse_counter_d1
    name: "${esphome_name} Power Meter"
    pin: D1
    unit_of_measurement: "kW"
    accuracy_decimals: 3
    count_mode:
      rising_edge: DISABLE
      falling_edge: INCREMENT
    internal_filter: 10ms
    filters:
      - multiply: 0.075
    total:
      name: "${esphome_name} Total Pulses"
```

new yaml:

```yaml
sensor:
  - platform: pulse_meter
    id: pulse_counter_d1
    name: "${esphome_name} Power Meter"
    pin: D1
    unit_of_measurement: "kW"
    accuracy_decimals: 3
    internal_filter: 10ms
    filters:
      - multiply: 0.075
    total:
      name: "${esphome_name} Total Pulses"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

olicooper, 2021-05-07:
I noticed that `App.register_component` is called twice in my code. I wonder if this is causing the issue?

```c
power_kw = new pulse_meter::PulseMeterSensor();
App.register_component(power_kw);
// ...
App.register_sensor(power_kw);
power_kw->set_unit_of_measurement("kW");
power_kw->set_force_update(false);
// ...
```

I also have an issue where the 'total daily energy' keeps resetting itself. Might open a separate issue.

pedrolamas, 2021-05-07:
Thanks for your comments @olicooper, but I don't think that is a problem as I also see that with `pulse_counter`

pedrolamas, 2021-05-07:
Hi @stevebaxter, as you implemented the `pulse_meter` in https://github.com/esphome/esphome/commit/a77784a6da8ef60e065db96d769092c759faf6e4, I was hoping you might have some insight on this matter?

stevebaxter, 2021-05-07:
Couple of things to check:

1. Make sure you have the right pull-up/down setting for the pin your sensor is on. Photodiodes in particular are often a bit noisy when switching, this could be counted as a second pulse. Pulling to the right level should help with that.
2. Set a longer value for `internal_filter`. This is the length of time after a pulse that we will ignore any further pulses, again this will help you ignore any noise.

This is my setup for my electricity meter (1000 pulses/kWh). I'm using a photodiode with some electronics that's "active low", so I pull up the pin it's on (I'm pretty sure that's right, and it seems to work):
```
sensor:
  - platform: pulse_meter
    name: "Electricity"
    pin:
      number: D5
      mode: INPUT_PULLUP
    unit_of_measurement: "kW"
    internal_filter: 100ms
    accuracy_decimals: 3
    timeout: 2 min
    filters:
      # Convert pulses/min (Wh/min) to kW
      - multiply: 0.06
    total:
      name: "Electricity Meter"
      unit_of_measurement: "kWh"
      accuracy_decimals: 3
      filters:
        - multiply: 0.001
```
It looks like your meter uses 800 pulses/kWh? In this case, even if your house is using 20 kW (a lot) you should only get 4.5 pulses/second, so 100ms as the filter should be fine.

stevebaxter, 2021-05-07:
No need to invert - we detect a rising edge, it doesn't really matter if that edge is at the start or end of the pulse as it's the distance between pulses we're looking for. 10ms is very short, you want this to be as long as possible while still allowing you to record the maximum likely level from the meter.

stevebaxter, 2021-05-07:
What exactly is your sensor?

pedrolamas, 2021-05-07:
Right now I'm using a simple LDR but I do have a TEPT4400 transistor that I need to assemble!

With 100ms seems to be correctly measuring, here's my current config:

```yaml
sensor:
  - platform: pulse_meter
    id: pulse_meter_d1
    name: "${esphome_name} Power Meter"
    pin: D1
    unit_of_measurement: "kW"
    accuracy_decimals: 3
    internal_filter: 100ms
    filters:
      - multiply: 0.075
    total:
      name: "${esphome_name} Total Energy"
      unit_of_measurement: "kWh"
      accuracy_decimals: 3
      filters:
        - multiply: 0.001

  - platform: total_daily_energy
    name: "${esphome_name} Total Daily Energy"
    power_id: pulse_meter_d1
```

This seems to be updating home assistant every ~0.5s... is this expected?

[image]

stevebaxter, 2021-05-07:
I'm a software guy rather than a hardware one, but I'm not sure a bare LDR will give great results (how is it connected up?). You want something that switches cleanly between on and off to give you a good signal, with an LDR I'm guessing you may not get a clean "switch" as the voltage drops or rises past the "on" level for the pin. This noisy switch may produce a lot of spurious readings.

I'm using a very cheap LDR module like this one:

https://www.instructables.com/LDR-Sensor-Module-Users-Manual-V10/

This has a digital output that can be connected to a digital pin, plus a pot that allows you to adjust the level.

`pulse_meter` will send an update every time we detect a pulse, this is a key advantage over `pulse_counter`. You can use a filter if you want to throttle that.

pedrolamas, 2021-05-07:
I have a binary sensor on the same pin that logs all detected true/false changes (and controls a LED on a different pin that allows for visual confirmation) and I can see it is working fine with the current LDR.

The change to 100ms seems to have fixed the issue, but I will have to make some more tests to be sure and if everything is ok, I will close this issue!

pedrolamas, 2021-05-09:
Looking at the current results, I think it is working correctly with the 100ms value, so this was completely my mistake - sorry about that!

Closing this issue down now.

closed 2021-05-09 by pedrolamas