esphome/issues #2023: Device class attribute is missing in MQTT discovery payload

definitio, 2021-05-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Arch Linux, Python 3.9, pip.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.17.0

**Affected component:**

https://esphome.io/components/sensor/index.html

**Description of problem:**
Setting `device_class` doesn't work.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: template
    name: $room illuminance
    state_topic: ${room}/sensor/illuminance
    device_class: illuminance
    lambda: 'return id(indoor_illuminance_raw).state;'
    update_interval: 10s
    filters:
      - sliding_window_moving_average:
          window_size: 6
          send_every: 6
          send_first_at: 6
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2021-05-24:
Yeah the `mqtt` integration just hasn't added that feature yet. PRs to add this to `MQTTSensorComponent` are welcome!

definitio, 2021-07-26:
Fixed

closed 2021-07-26 by definitio

krconv, 2021-07-29:
The fix for sensors is working great 🥳 ; just posted a fix for covers in https://github.com/esphome/esphome/pull/2092