esphome/issues #2024: Rest Api Not working

sat2madgo, 2021-05-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.): HA OS on Raspberry Pi**

**ESP (ESP8266): ESP8266**

**ESPHome version (1.17)**

**Affected component: Web Server**

**Description of problem: I am getting error 404 not found when I try to get anything using rest APi's. There is nothing I see in logs.**

**Problem-relevant YAML-configuration entries:**
```yaml

web_server:
  port: 80
binary_sensor:
  - platform: gpio
    device_class: opening
    id: floatswitch
    name: "Test Float Switch"
    pin:
      number: D2
      mode: INPUT_PULLUP
    filters:
      - delayed_on_off: 100ms

```

**Logs (if applicable):**

```
Done! Flashing is complete!

Showing logs:
[12:13:13]bcn 0
[12:13:13]del if1
[12:13:13]usl
[12:13:13]mode : null
[12:13:13]mode : sta(50:02:91:d5:e4:03)
[12:13:13]add if0
[12:13:13]sleep disable
[12:13:19]scandone
[12:13:19]scandone
[12:13:20]state: 0 -> 2 (b0)
[12:13:20]state: 2 -> 3 (0)
[12:13:20]state: 3 -> 5 (10)
[12:13:20]add 0
[12:13:20]aid 1
[12:13:20]cnt 
[12:13:20]
[12:13:20]connected with X, channel 12
[12:13:20]dhcp client start...
[12:13:21]ip:192.168.0.109,mask:255.255.0.0,gw:192.168.5.1
[12:13:30]pm open,type:0 0

```

**Additional information and things you've tried: I have tried it with lights as well, its same.**

sat2madgo, 2021-05-08:
> Greetings,
> 
> The [API](https://esphome.io/web-api/index.html#rest-api) clearly states to call `/<domain>/<id>`.
> 
> Enabling debug in your yaml logs does help to gain more insights.
> 
> ```yaml
> # Enable logging
> logger:
>   level: DEBUG
> ```
> 
> What I did notice is the use of the `name:` and `id:` for the `binary_sensor` in your example
> 
> For my configs I use only name and not id
> 
> ```yaml
> binary_sensor:
> ...
>   - platform: gpio
>     name: "esp_tester_d1"
>     device_class: garage_door
>     pin:
>       number: D1
>       inverted: true
> ...
> ```
> 
> The logs show:
> 
> ```
> [17:01:24][D][binary_sensor:036]: 'esp_tester_d1': Sending state OFF
> [17:01:24][D][binary_sensor:036]: 'esp_tester_d1': Sending state ON
> ```
> 
> And I can access it via HTTP
> 
> ```
> $ curl http://tester.home/binary_sensor/esp_tester_d1
> {"id":"binary_sensor-esp_tester_d1","state":"ON","value":true}
> ```
> 
> However, when I change the details to represent configs similar to yours I get the same 404 errors.
> 
> ```yaml
> ...
>   - platform: gpio
>     name: "Fluffy bunny feet"
>     id: "esp_tester_d1"
>     device_class: garage_door
>     pin:
>       number: D1
>       inverted: true
> ...
> ```
> 
> Logs:
> 
> ```
> [17:18:21][D][binary_sensor:036]: 'Fluffy bunny feet': Sending state OFF
> [17:18:21][D][binary_sensor:036]: 'Fluffy bunny feet': Sending state ON
> ```
> 
> But the curl calls based on the `name:` or the `id:` all fail
> 
> ```
> $ curl http://192.168.13.48/binary_sensor/Fluffy%20bunny%20feet
> $ curl http://192.168.13.48/binary_sensor/Fluffy+bunny+feet
> $ curl http://192.168.13.48/binary_sensor/fluffy%20bunny%20feet
> $ curl http://192.168.13.48/binary_sensor/fluffy+bunny+feet
> $ curl -v http://192.168.13.48/binary_sensor/esp_tester_d1
> ```
> 
> Removing the spaces from the name
> 
> ```yaml
>   ## tester
>   - platform: gpio
>     name: "FluffyBunnyFeet"
>     id: "esp_tester_d1"
>     device_class: garage_door
>     pin:
>       number: D1
>       inverted: true
> ```
> 
> Logs
> 
> ```
> [17:41:28][D][binary_sensor:036]: 'FluffyBunnyFeet': Sending state OFF
> [17:41:28][D][binary_sensor:036]: 'FluffyBunnyFeet': Sending state ON
> ```
> 
> The curl calls does work only if the name is all lower case, using the id does not work and also produces a 404 error
> 
> ```
> curl -v http://192.168.13.48/binary_sensor/fluffybunnyfeet
> {"id":"binary_sensor-fluffybunnyfeet","state":"ON","value":true}
> 
> curl -v http://192.168.13.48/binary_sensor/FluffyBunnyFeet
> 404
> 
> curl -v http://192.168.13.48/binary_sensor/esp_tester_d1
> 404
> ```
> 
> Based on these finding you should adapt your `yaml` specifically the `name:` option to suite your immediate needs.
> 
> I would suggest you log an issue and report this finding, it does appear to be an matter to be addressed.
> 
> I trust this will help.

This explains everything. for some reason instead of ID its using names. Works once I remove ID's and keep names without spaces.

closed 2021-05-08 by sat2madgo

ssieb, 2021-05-13:
You need to read the docs carefully.  It is explained at https://esphome.io/web-api/index.html#rest-api
> this ID is created by taking the name of the component, stripping out all non-alphanumeric characters, making everything lowercase and replacing all spaces by underscores.

meijerwynand, 2021-05-17:
> You need to read the docs carefully. It is explained at https://esphome.io/web-api/index.html#rest-api
> 
> > this ID is created by taking the name of the component, stripping out all non-alphanumeric characters, making everything lowercase and replacing all spaces by underscores.

Thanks for the feedback, 

I have added to the REST API documentation how to find the `<id>` using `VERY_VERBOSE` in the logs.