esphome/issues #2026: fan.speed and fan.state have no (float) value after updating to 1.17.0

Bart-1992, 2021-05-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio
**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**
 it was working on 1.15.3 
-->

**Affected component:**

**Description of problem:**

I've always had a working solution in 1.15.3, however after updating today to 1.17 my automatic fan controller become unresponsive.

The issue is that i'm not receiving any 'status' at fan.state or fan.speed? Probably there is something wrong in my coding? How can I get fan status ( eg 100% or 33%) and use it in a statement ? 

When manually setting the fan speed, to for example 33%, the logger shows : [D][speed.fan:035]: Setting speed: 0.33

But when trying to get the speed value from the fan :
- 'id(itho_fan).state']
- 'id(itho_fan).speed']

they always appear to be 0 -->           

- logger.log:
                      format: "ITHO speed %.1f"
                      args: [ 'id(itho_fan).state']

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: centrale_afzuiging
  platform: ESP8266
  board: nodemcuv2
  includes:
    - custom_fan.h
  
## PULL UP ALL THE PINS ON BOOT   
  on_boot:
   then:
   - servo.write:
      id: my_servo
      level: 40.0% 
   - delay: 2000ms     
   - fan.turn_on: 
      id: itho_fan
      speed: 100
   - delay: 1000ms   
   - fan.turn_on: 
      id: itho_fan
      speed: 67
   - delay: 1000ms    
   - fan.turn_on: 
      id: itho_fan
      speed: 1 
   - switch.turn_on: auto_humidity
   - lambda: 'return id(dht2).publish_state(id(badkamer_luchtvochtigheid).state);'    
     
   

wifi:
  ssid: "Ziggo6665315"
  password: "dv6Vbtsqfqny"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Centrale Afzuiging"
    password: "ue95AYc0VhYC"

captive_portal:

# Enable logging
logger:
  level: VERBOSE 

# Enable Home Assistant API
api:

ota:

switch:
  - platform: template
    name: "Automatic humidity controller"
    id: auto_humidity
    optimistic: true

sensor:
  - platform: dht
    pin: D2
    temperature:
      name: "Badkamer Temperatuur"
      id: badkamer_temp
      accuracy_decimals: 4
    humidity:
      name: "Badkamer Luchtvochtigheid"
      id: badkamer_luchtvochtigheid
    model: DHT22
    update_interval: 30s
  - platform: template
    name: "Previous DHT reading"
    id: dht2
    update_interval: 10 s
    internal: true    
    
globals:
 - id: statusflag
   type: bool
   restore_value: no
   initial_value: 'false'  
   
    
time:
  - platform: sntp
    on_time:
      - seconds: /30
        #minutes: /1
        then:
            - logger.log: "getting in trigger loop "
            - logger.log:
                      format: "ITHO speed %.1f"
                      args: [ 'id(itho_fan).speed']
            - logger.log:
                      format: "dht2 value =  %.1f"
                      args: [ 'id(dht2).state']   
            - logger.log:
                      format: "calculation value badkamerluchtvochtigheid - dht2 =  %.1f"
                      args: [ '(id(badkamer_luchtvochtigheid).state - id(dht2).state)']    

            - if:
               condition: 
                 for:
                   time: 5 min
                   condition:               
               #- lambda: 'return (id(badkamer_luchtvochtigheid).state > 0.0  && id(badkamer_luchtvochtigheid).state < 82.5) && (id(auto_humidity).state == true);' # Absolute controlling
                    - lambda: 'return (((id(badkamer_luchtvochtigheid).state - id(dht2).state ) <= 15.0 ) && (id(auto_humidity).state == true)&& id(statusflag) == true);'
                    - lambda: 'return (id(itho_fan).speed != 0.33);'
               then:
               - globals.set:
                  id: statusflag
                  value: 'false'
               - servo.write:
                  id: my_servo
                  level: 40.0%
               - delay: 1 s                   
               - cover.template.publish:
                  id: living_valve
                  state: OPEN               
               - fan.turn_on: 
                  id: itho_fan
                  speed: 33
               - delay: 5 s
               - fan.turn_on: # Cross check that message is received by fan 
                  id: itho_fan
                  speed: 33               
               
               - logger.log: "automated humidity controller : LOW"               

            - if:
               condition: 
               # - lambda: 'return (id(badkamer_luchtvochtigheid).state > 85.0 && id(badkamer_luchtvochtigheid).state < 100.0) && (id(auto_humidity).state == true);'
                - lambda: 'return ((((id(badkamer_luchtvochtigheid).state - id(dht2).state ) > 15.0 ) && (id(auto_humidity).state == true))&& id(statusflag) == false);'               
                - lambda: 'return (id(itho_fan).speed != 1);'
               then:
               - globals.set:
                  id: statusflag
                  value: 'true'               
               - servo.write:
                  id: my_servo
                  level: -100.0%  
               - delay: 1 s 
               - cover.template.publish:
                  id: living_valve
                  state: CLOSED                 
               - fan.turn_on: 
                  id: itho_fan
                  speed: 100
               - delay: 5 s
               - fan.turn_on: 
                  id: itho_fan
                  speed: 100               
               - logger.log: "automated humidity controller : HIGH!"
        
               
            - if:
               condition:
                 for:
                   time: 20min
                   condition:
                    - lambda: 'return ((id(itho_fan).speed != 0.33)&& id(statusflag) == true);'
               then:
               - globals.set:
                  id: statusflag
                  value: 'false'               
               - servo.write:
                  id: my_servo
                  level: 40.0%
               - delay: 1 s                   
               - cover.template.publish:
                  id: living_valve
                  state: OPEN               
               - fan.turn_on: 
                  id: itho_fan
                  speed: 33
               - delay: 5 s
               - fan.turn_on: # Cross check that message is received by fan 
                  id: itho_fan
                  speed: 33               
               - logger.log: "automated humidity controller : LOW after 30 minutes timeout"                      
               
           
 
  - platform: sntp
    on_time:
      - seconds: 0
        minutes: /30
        then:               
            - if:
               condition: 
               - lambda: 'return ((float)id(itho_fan).speed == 0.33);'
               then:
               - logger.log: "We are at 0 speed, so publish servo to OPEN, and update DT value of luchtvochtigheid "               
               - servo.write:
                  id: my_servo
                  level: 40.0%
               - delay: 1 s  
               - cover.template.publish:
                  id: living_valve
                  state: OPEN                    
               - lambda: 'return id(dht2).publish_state(id(badkamer_luchtvochtigheid).state);'                
       
           
output:
  - platform: custom
    type: float 
    lambda: |-
      auto my_custom_fan_output = new MyCustomFanoutput();
      App.register_component(my_custom_fan_output);
      return {my_custom_fan_output};

    outputs:
      id: my_custom_fan
      
  - platform: esp8266_pwm
    id: pwm_output
    pin: GPIO02
    frequency: 50 Hz

fan:
  - platform: speed
    output: my_custom_fan
    id: itho_fan
    name: "Mechanische Ventilatie"
    speed_count: 3
servo:
  - id: my_servo
    output: pwm_output  
cover:
  - platform: template
    device_class: damper
    optimistic: true
    name: "Livingroom Valve"
    id: living_valve
    open_action:
     - servo.write:
        id: my_servo
        level: 40.0%
    close_action:
     - servo.write:
        id: my_servo
        level: -100.0%
 
    

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Misiu, 2021-05-06:
@Bart-1992 according to the docs (https://github.com/esphome/esphome-docs/pull/1056/files) fan speed is an int value, default from 1 to 100

Bart-1992, 2021-05-06:
@Misiu Thankyou for your reply. 

Now I see what i'm doing wrong. I assumed the setting given in % ( default 0.33/0.66/1.0 ) were also corresponding to the speed setting in ESPhome, but they are transposed to the corresponding 'int' setting you provide in the speed_count setting. 

So for example :

Speed_count = 3 
0.33 ( 33% ) = (fan.speed) = 1
0.66 ( 66%) = (fan.speed) = 2
1.0 ( 100% ) = (fan.speed) = 3

closed 2021-05-06 by Bart-1992