esphome/issues #2028: DHT sensor error when NAME in non-English characters

IvanGavrilenko, 2021-05-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

core-2021.4.6
AND
ESPHOME web server

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Wemos D1 mini
**ESPHome version (latest production, beta, dev branch)**

C:\Users\ivan>esphome version
Version: 1.16.2
**Affected component:**

https://esphome.io/components/sensor/dht.html

**Description of problem:**
When attached DHT-11 sensor and if name section in sensor block consists of only English cahracters - then web server and HomeAssistant receives both values - temperature and humidity.
But if the name section consists of only Russian characters then both web server and HomeAssistant show the single sensor (and web server shows name from temperature and value - from humidity - it looks like the temperature sensor obtains the value from the temperature and shows the temperature value and then after a part of second the humidity value is written to the same sensor and overwrites it's value).

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
#WORKING CONFIG
sensor:
  - platform: dht
    pin: D3
    temperature:
      name: Temperature #pay attantion to this name
      filters:
        - filter_out: nan
    humidity:
      name: Humidity #pay attantion to this name
      filters:
        - filter_out: nan
    update_interval: 20s

#CONFIG PRODUCING ERROR
sensor:
  - platform: dht
    pin: D3
    temperature:
      name: Температура #The error goes here 
      filters:
        - filter_out: nan
    humidity:
      name: Влажность #The error goes here 
      filters:
        - filter_out: nan
    update_interval: 20s
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
dht Web Server
States
Name	State	Actions
Температура	26 %	
Влажность		
See ESPHome Web API for REST API documentation.

OTA Update
Файл не выбран
Debug Log
[0;38m[VV][scheduler:152]: Running interval 'update' with interval=20000 last_execution=36085 (now=56085)[0m
[VV][scheduler:152]: Running interval 'update' with interval=20000 last_execution=36085 (now=56085)
[0;38m[VV][dht:176]: Data: Hum=0b0001100100000000, Temp=0b0010000000001000, Checksum=0b01000001[0m
[VV][dht:176]: Data: Hum=0b0001100100000000, Temp=0b0010000000001000, Checksum=0b01000001
[D][dht:048]: Got Temperature=32.8°C Humidity=25.0%
[V][sensor:013]: 'Температура': Received new state 32.799999
[0;38m[VV][sensor.filter:013]: Filter(0x3fff27bc)::input(32.799999)[0m
[VV][sensor.filter:013]: Filter(0x3fff27bc)::input(32.799999)
[0;38m[VV][sensor.filter:020]: Filter(0x3fff27bc)::output(32.799999) -> SENSOR[0m
[VV][sensor.filter:020]: Filter(0x3fff27bc)::output(32.799999) -> SENSOR
[D][sensor:092]: 'Температура': Sending state 32.80000 °C with 1 decimals of accuracy
[V][sensor:013]: 'Влажность': Received new state 25.000000
[0;38m[VV][sensor.filter:013]: Filter(0x3fff291c)::input(25.000000)[0m
[VV][sensor.filter:013]: Filter(0x3fff291c)::input(25.000000)
[0;38m[VV][sensor.filter:020]: Filter(0x3fff291c)::output(25.000000) -> SENSOR[0m
[VV][sensor.filter:020]: Filter(0x3fff291c)::output(25.000000) -> SENSOR
[D][sensor:092]: 'Влажность': Sending state 25.00000 % with 0 decimals of accuracy
[0;38m[VV][scheduler:152]: Running interval '' with interval=10000 last_execution=46688 (now=56688)[0m
[VV][scheduler:152]: Running interval '' with interval=10000 last_execution=46688 (now=56688)
[0;38m[VV][scheduler:152]: Running interval '' with interval=10000 last_execution=56688 (now=66691)[0m
[VV][scheduler:152]: Running interval '' with interval=10000 last_execution=56688 (now=66691)
[0;38m[VV][scheduler:152]: Running interval 'update' with interval=20000 last_execution=56085 (now=76085)[0m
[VV][scheduler:152]: Running interval 'update' with interval=20000 last_execution=56085 (now=76085)
[0;38m[VV][dht:176]: Data: Hum=0b0001101000000000, Temp=0b0010000000001000, Checksum=0b01000010[0m
[VV][dht:176]: Data: Hum=0b0001101000000000, Temp=0b0010000000001000, Checksum=0b01000010
[D][dht:048]: Got Temperature=32.8°C Humidity=26.0%
[V][sensor:013]: 'Температура': Received new state 32.799999
[0;38m[VV][sensor.filter:013]: Filter(0x3fff27bc)::input(32.799999)[0m
[VV][sensor.filter:013]: Filter(0x3fff27bc)::input(32.799999)
[0;38m[VV][sensor.filter:020]: Filter(0x3fff27bc)::output(32.799999) -> SENSOR[0m
[VV][sensor.filter:020]: Filter(0x3fff27bc)::output(32.799999) -> SENSOR
[D][sensor:092]: 'Температура': Sending state 32.80000 °C with 1 decimals of accuracy
[V][sensor:013]: 'Влажность': Received new state 26.000000
[0;38m[VV][sensor.filter:013]: Filter(0x3fff291c)::input(26.000000)[0m
[VV][sensor.filter:013]: Filter(0x3fff291c)::input(26.000000)
[0;38m[VV][sensor.filter:020]: Filter(0x3fff291c)::output(26.000000) -> SENSOR[0m
[VV][sensor.filter:020]: Filter(0x3fff291c)::output(26.000000) -> SENSOR
[D][sensor:092]: 'Влажность': Sending state 26.00000 % with 0 decimals of accuracy
[0;38m[VV][scheduler:152]: Running interval '' with interval=10000 last_execution=66688 (now=76688)[0m
[VV][scheduler:152]: Running interval '' with interval=10000 last_execution=66688 (now=76688)
```

**Additional information and things you've tried:**
The screenshots of working config:
[image]
[image]

The screenshots of errous config:
[image]
[image]

Complete YAMLs:
[YAML not working.txt](https://github.com/esphome/issues/files/6434613/YAML.not.working.txt)
[YAML working.txt](https://github.com/esphome/issues/files/6434614/YAML.working.txt)

Parts of VERY_VERBOSE logs containig sensor values obtainig:
[not working log.txt](https://github.com/esphome/issues/files/6434608/not.working.log.txt)
[working log.txt](https://github.com/esphome/issues/files/6434609/working.log.txt)

IvanGavrilenko, 2021-09-03:
Any comments to the issue?

oxan, 2021-09-19:
The problem here is that entities are internally identified by their "object id", which is the name with all non-Latin-alphanumericals stripped. Thus, the two sensors with only Russian characters in their name get the same object id.

IvanGavrilenko, 2021-09-19:
> The problem here is that entities are internally identified by their "object id", which is the name with all non-Latin-alphanumericals stripped. Thus, the two sensors with only Russian characters in their name get the same object id.

So may be adding transliteration to object id would be a good idea (as hame assistant does)?

oxan, 2021-09-19:
> So may be not stripping non-english characters but adding transliteration to object id would be a good idea (as hame assistant does)?

The problem with Unicode transliteration and things like that is that they require _huge_ datatables, for which an ESP doesn't have enough space.

IvanGavrilenko, 2021-09-19:
Well may be I do not know the full procedure of firmware compilation, but may be just add to the computer ESPhome a sort of pre-compilation - before compiling *.yaml transliterate fields that form "object id" all strings "name: " and so on?