esphome/issues #2031: Home Assistant never asks me for API password when I add it to my device

TheLastGimbus, 2021-05-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Installed in venv from pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

Tried this on ESP32 and 8266

**ESPHome version (latest production, beta, dev branch)**

v1.17.1

**Affected component:**

https://esphome.io/components/api.html

**Description of problem:**
I wanted to add a little bit of security to my devices by adding a password to API

Suprisingly, this doesn't change anything at all :thinking: I added it to my config like this:
```diff
# Enable Home Assistant API
api:
+  password: !secret api_pass
```
But my device is fully working in Home Assistant, and didn't ask me for any password. I also removed it and tried adding it again, and no such thing! Is the password feature disabled or my Home Assistant magically knows the password?

**Problem-relevant YAML-configuration entries:**
```yaml
api:
  password: !secret api_pass
```

OttoWinter, 2021-05-24:
That's odd, just tested and the behavior is correct here.

Are you sure you deployed the updated configuration? (By clicking upload?) Otherwise it's only locally in your config, but never deployed to the device.

glmnet, 2021-05-24:
I tested it and worked too

closed 2021-10-06 by stale[bot]