esphome/issues #2032: DF-plyer mini stopped working with HA

wasyl00, 2021-05-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Version | core-2021.5.0
Installation Type | Home Assistant OS
Development | false
Supervisor | true
Docker | true
Virtual Environment | false
Python Version | 3.8.9
Operating System Family | Linux
Operating System Version | 5.4.83-v7
CPU Architecture | armv7l

**ESP (ESP32/ESP8266, Board/Sonoff):**

esp8285 on Sonoff SV

**ESPHome version (latest production, beta, dev branch)**

Current version: 1.17.1

**Affected component:**

DF-Player mini
https://esphome.io/components/dfplayer.html

**Description of problem:**

I was previously successfully using the sample code from the website below but sadly after yesterday upgrade I'm no longer able to trigger from HA any of the services related to the DF  mini player. I have a button on my device which is able to run the player from the esp itself but all service commands coming from HA are not working and I also get 2 errors in HA (below).  I have also configured light in the in the same script which is working fine from HA only DF mini player part seem to be not working.

DF mini config on my esp
```
api:
  services:
  - service: dfplayer_next
    then:
      - dfplayer.play_next:
  - service: dfplayer_previous
    then:
      - dfplayer.play_previous:
  - service: dfplayer_play
    variables:
      file: int
    then:
      - dfplayer.play: !lambda 'return file;'
  - service: dfplayer_play_loop
    variables:
      file: int
      loop_: bool
    then:
      - dfplayer.play:
          file: !lambda 'return file;'
          loop: !lambda 'return loop_;'
  - service: dfplayer_play_folder
    variables:
      folder: int
      file: int
    then:
      - dfplayer.play_folder:
          folder: !lambda 'return folder;'
          file: !lambda 'return file;'

  - service: dfplayer_play_loop_folder
    variables:
      folder: int
    then:
      - dfplayer.play_folder:
          folder: !lambda 'return folder;'
          loop: True

  - service: dfplayer_set_device_tf
    then:
      - dfplayer.set_device: TF_CARD

  - service: dfplayer_set_device_usb
    then:
      - dfplayer.set_device: USB

  - service: dfplayer_set_volume
    variables:
      volume: int
    then:
      - dfplayer.set_volume: !lambda 'return volume;'
  - service: dfplayer_set_eq
    variables:
      preset: int
    then:
      - dfplayer.set_eq: !lambda 'return static_cast<dfplayer::EqPreset>(preset);'

  - service: dfplayer_sleep
    then:
      - dfplayer.sleep

  - service: dfplayer_reset
    then:
      - dfplayer.reset

  - service: dfplayer_start
    then:
      - dfplayer.start

  - service: dfplayer_pause
    then:
      - dfplayer.pause

  - service: dfplayer_stop
    then:
      - dfplayer.stop

  - service: dfplayer_random
    then:
      - dfplayer.random

```

**Logs (if applicable):**

First error log from HA
```
Logger: homeassistant.helpers.script.websocket_api_script
Source: components/esphome/__init__.py:536
First occurred: 8:49:36 AM (4 occurrences)
Last logged: 8:55:27 AM

websocket_api script: Error executing script. Unexpected error for call_service at pos 1: Not connected!
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 363, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 563, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1481, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1516, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 536, in execute_service
    await entry_data.client.execute_service(service, call.data)
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 373, in execute_service
    self._check_authenticated()
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 73, in _check_authenticated
    self._check_connected()
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 68, in _check_connected
    raise APIConnectionError("Not connected!")
aioesphomeapi.core.APIConnectionError: Not connected!

```
Second error log from HA
```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/esphome/__init__.py:536
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 8:49:36 AM (4 occurrences)
Last logged: 8:55:27 AM

[1777813392] Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 21, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 482, in handle_execute_script
    await script_obj.async_run(msg.get("variables"), context=context)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 1209, in async_run
    await asyncio.shield(run.async_run())
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 345, in async_run
    await self._async_step(log_exceptions=False)
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 363, in _async_step
    await getattr(self, handler)()
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 563, in _async_call_service_step
    await service_task
  File "/usr/src/homeassistant/homeassistant/core.py", line 1481, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1516, in _execute_service
    await handler.job.target(service_call)
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 536, in execute_service
    await entry_data.client.execute_service(service, call.data)
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 373, in execute_service
    self._check_authenticated()
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 73, in _check_authenticated
    self._check_connected()
  File "/usr/local/lib/python3.8/site-packages/aioesphomeapi/client.py", line 68, in _check_connected
    raise APIConnectionError("Not connected!")
aioesphomeapi.core.APIConnectionError: Not connected!
```

**Additional information and things you've tried:**

-
- I tried re uploading script to esp no issues, no errors. 
- Restarting whole thing couple of times.

glmnet, 2021-05-07:
did you try already removing the integration and adding it back?

wasyl00, 2021-05-08:
Ok I managed to sort it by completely removing the integration but I still dont know why, for some of my devices, strange grouping happen in the devices menu like below:

https://i.imgur.com/Kyt3O5P.png

closed 2021-05-08 by wasyl00