esphome/issues #2034: pulse_meter 'total' not working as expected

olicooper, 2021-05-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU v2 (ESP8266)

**ESPHome version (latest production, beta, dev branch)**

2021.4.6

**Affected component:**

https://esphome.io/components/sensor/pulse_meter.html

**Description of problem:**
Couple of issues...
1. Using the `total` variable, I want to output the total daily energy. With the original [total_daily_energy](https://esphome.io/components/sensor/total_daily_energy.html) component it used to reset at midnight but this one uses the generic 'sensor' component which doesn't do this. This new component is advertised as a drop-in for the original so I assumed it would work the same. Is this a design decision or an oversight?
2. The `total` that is output also keeps resetting randomly. My original configuration worked flawlessly. I am in the process of testing if it is due to hard reset from memory overflow or something. My initial thoughts are that it might be the speed that updated values are published?

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: pulse_meter
    pin: D5 #14
    unit_of_measurement: 'kW'
    icon: "mdi:flash"
    name: ${node_name} Power kW
    id: power_kw
    internal_filter: 50ms
    filters:
      - multiply: 0.06
    total:
      name: ${node_name} Total Daily Energy
      unit_of_measurement: "kWh"
      accuracy_decimals: 3
      filters:
        - multiply: 0.001
```

**Logs (if applicable):**

**Additional information and things you've tried:**

olicooper, 2021-05-11:
I have since updated the configuration to remove the 'total' from `pulse_meter` and add a seperate `total_daily_energy` configuration which is has a throttle filter set at 10 seconds. I have also added `esp8266_store_log_strings_in_flash: false` and the node is no-longer randomly rebooting, so I think there might be an issue with the `pulse_meter`.

TommySharpNZ, 2021-05-11:
Can you share what your YAML looks like now? I would also like a daily total that resets at midnight.

millerkyle72, 2021-05-22:
Seems like it would be good to have some reset option built into this pulse counter alternative. I'm sure the daily midnight reset does not work for everyone so maybe an internal service call to reset the total would be great. In my set-up I am using the pulse_counter and the integration sensor. I have the integration reset at midnight with the following built-in call:

`on_time:
      - hours: 0
        minutes: 0
        seconds: 0
        then:
          - sensor.integration.reset:  water_total
        `

millerkyle72, 2021-05-24:
A service for setting the total has been merged into dev branch. I think this [PR](https://github.com/esphome/esphome/pull/1757) solves this issue if you setup the time based automation to reset the total at midnight.

closed 2021-10-06 by stale[bot]