esphome/issues #2037: REST API and YAML/device id

meijerwynand, 2021-05-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Linux, pip3 install 
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 NodeMCU
**ESPHome version (latest production, beta, dev branch)**

1.17
**Affected component:**

https://esphome.io/web-api/index.html#rest-api

**Description of problem:**
The `id:` value in the `yaml` is not used when attempting an HTTP call using the REST API

**Problem-relevant YAML-configuration entries:**
```yaml
...
  - platform: gpio
    name: "Fluffy bunny feet"
    id: "esp_tester_d1"
    device_class: garage_door
    pin:
      number: D1
      inverted: true
...
```

**Logs (if applicable):**

```
All these calls return a 404 error

$ curl http://192.168.13.48/binary_sensor/Fluffy%20bunny%20feet
$ curl http://192.168.13.48/binary_sensor/Fluffy+bunny+feet
$ curl http://192.168.13.48/binary_sensor/fluffy%20bunny%20feet
$ curl http://192.168.13.48/binary_sensor/fluffy+bunny+feet
$ curl -v http://192.168.13.48/binary_sensor/esp_tester_d1
```

**Additional information and things you've tried:**

This was initially asked and addressed in https://github.com/esphome/issues/issues/2024

According to the [API](https://esphome.io/web-api/index.html#rest-api) calls must be made to `/<domain>/<id>`. 

Doing various tests does indicate the `id:` value in the `yaml` is not honoured when attempting an HTTP REST call.

## Working example
### Config
For this config only `name:` is used and  not `id:` 
```yaml
binary_sensor:
...
  - platform: gpio
    name: "esp_tester_d1"
    device_class: garage_door
    pin:
      number: D1
      inverted: true
...
```

### Logs
```
[17:01:24][D][binary_sensor:036]: 'esp_tester_d1': Sending state OFF
[17:01:24][D][binary_sensor:036]: 'esp_tester_d1': Sending state ON
```

### HTTP
```
$ curl http://tester.home/binary_sensor/esp_tester_d1
{"id":"binary_sensor-esp_tester_d1","state":"ON","value":true}
```

## Non-working example
Note: `name:` and `id:` is now supplied in the config

### Config
```yaml
...
  - platform: gpio
    name: "Fluffy bunny feet"
    id: "esp_tester_d1"
    device_class: garage_door
    pin:
      number: D1
      inverted: true
...
```

### Logs
```
[17:18:21][D][binary_sensor:036]: 'Fluffy bunny feet': Sending state OFF
[17:18:21][D][binary_sensor:036]: 'Fluffy bunny feet': Sending state ON
```
### HTTP
But the curl calls based on the `name:` or the `id:` all fail

```
$ curl http://192.168.13.48/binary_sensor/Fluffy%20bunny%20feet
$ curl http://192.168.13.48/binary_sensor/Fluffy+bunny+feet
$ curl http://192.168.13.48/binary_sensor/fluffy%20bunny%20feet
$ curl http://192.168.13.48/binary_sensor/fluffy+bunny+feet
$ curl -v http://192.168.13.48/binary_sensor/esp_tester_d1
```

Removing the spaces from the name
### Config
```yaml
  ## tester
  - platform: gpio
    name: "FluffyBunnyFeet"
    id: "esp_tester_d1"
    device_class: garage_door
    pin:
      number: D1
      inverted: true
```

### Logs
```
[17:41:28][D][binary_sensor:036]: 'FluffyBunnyFeet': Sending state OFF
[17:41:28][D][binary_sensor:036]: 'FluffyBunnyFeet': Sending state ON
```

### HTTP
The curl calls does work only if the name is all lower case, using the id does not work and also produces a 404 error

```
curl -v http://192.168.13.48/binary_sensor/fluffybunnyfeet
{"id":"binary_sensor-fluffybunnyfeet","state":"ON","value":true}

curl -v http://192.168.13.48/binary_sensor/FluffyBunnyFeet
404

curl -v http://192.168.13.48/binary_sensor/esp_tester_d1
404
```

ssieb, 2021-05-13:
You need to read the docs carefully.  It is explained at https://esphome.io/web-api/index.html#rest-api
> this ID is created by taking the name of the component, stripping out all non-alphanumeric characters, making everything lowercase and replacing all spaces by underscores.

closed 2021-05-13 by ssieb

meijerwynand, 2021-05-17:
> You need to read the docs carefully. It is explained at https://esphome.io/web-api/index.html#rest-api
> 
> > this ID is created by taking the name of the component, stripping out all non-alphanumeric characters, making everything lowercase and replacing all spaces by underscores.

Thanks for the feedback, 

I have added to the REST API documentation how to find the `<id>` using `VERY_VERBOSE` in the logs with this PR - https://github.com/esphome/esphome-docs/pull/1168.

[image]